espressif/arduino-esp32 #8918: Space key key code does not exist in USBHIDKeyboard.h

HSBL-ko-gyo, 2023-11-25:
### Board

all

### Device Description

-

### Hardware Configuration

-

### Version

v2.0.14

### IDE Name

arduinoIDE

### Operating System

all

### Flash frequency

all

### PSRAM enabled

no

### Upload speed

11520

### Description

There is no key code corresponding to the space key in USBHIDKeyboard.h. To use the space key, you need to use the ASCII value 32, which is not intuitive. I suggest adding a key code that corresponds to the space key.

For example, you can define:
#define KEY_SPACE 0x20
This change allows you to write code for pressing the space bar just like any other special key:
Keyboard.press(KEY_SPACE);
We would appreciate it if you would consider fixing this issue. Thank you.

### Sketch

```cpp
Keyboard.press(KEY_SPACE);
```

### Debug Message

```plain
note: suggested alternative: 'KEY_HOME'

exit status 1

Compilation error: 'KEY_SPACE' was not declared in this scope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-25:
You are welcome to add something like this yourself.  Just navigate to the file in github, and click the pencil icon to edit.

HSBL-ko-gyo, 2023-11-25:
@lbernstone,

Thank you for your response. Although I am still a beginner who is not familiar with using GitHub, I created a pull request regarding the addition of `KEY_SPACE`. Here is my suggestion: https://github.com/espressif/arduino-esp32/pull/8921

If there are any mistakes, I would appreciate it if you could point them out. I would like to continue learning and moving forward so that I can contribute to this project.

thank you very much.

HSBL-ko-gyo, 2023-11-25:
@lbernstone,

Thank you for your prompt response. I didn't realize that you were already working on the improvements I was about to suggest. Thank you very much for this wonderful contribution.

We look forward to your continued cooperation in the development of this project. I'm really thankful to you.

closed 2023-11-25 by HSBL-ko-gyo