espressif/arduino-esp32 #8924: Still true: esp-idf 4.4 ?

diplfranzhoepfinger, 2023-11-26:
### Related area

v

### Hardware specification

v

### Is your feature request related to a problem?

v

### Describe the solution you'd like

v

### Describe alternatives you've considered

v

### Additional context

v

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2023-11-26:
Yes, there is no released Arduino version based on IDF 5.1. It is coming

SuGlider, 2023-11-26:
@diplfranzhoepfinger - the documentation is still valid. The stable Arduino version is the Core 2.0.14 based on IDF 4.4.6.
Next Ardino Core (still in Alpha version) is 3.0.0 and it will be based on IDF 5.1, as @Jason2866 commented.

There is a way to build Arduino as IDF component, just using Managed Components.
This is demonstrated in #8878 
For that, all you need is to clone IDF 4.4 (because the Arduino Managed Component is still 2.0.14) and then run `idf.py build` after running `install and export IDF scripts`.

peteDDD, 2023-11-27:
So there is this announcement: [Announcing the Arduino ESP32 Core version 3.0.0](https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4)

And there are these instructions: [Announcing the Arduino ESP32 Core version 3.0.0](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#how-to-update-to-the-latest-code)

But has anyone gotten it working on VSC/PlatformIO?

Jason2866, 2023-11-27:
@peteDDD Yes. Made a fork we use for Tasmota to build Tasmota for the C2 and C6.

peteDDD, 2023-11-28:
@Jason2866 Thank you.  Is there a straightforward way to install that fork or should I continue to wait for the "official" release.

diplfranzhoepfinger, 2023-11-28:
> @diplfranzhoepfinger - the documentation is still valid.

but then it is nonsense. 

you should make a 2.0.0 Version Doc with that hint, 

and "latest" should refer to what is bleeding Edgle. like in ESP-IDF Documentaiton.

Jason2866, 2023-11-28:
> @Jason2866 Thank you. Is there a straightforward way to install that fork or should I continue to wait for the "official" release.

Use this
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.11.11/platform-espressif32.zip
```

SuGlider, 2023-11-28:
> > @diplfranzhoepfinger - the documentation is still valid.
> 
> but then it is nonsense. 
> 
> you should make a 2.0.0 Version Doc with that hint, 
> 
> and "latest" should refer to what is bleeding Edgle. like in ESP-IDF Documentaiton.

A new documentation with 2 version, like in ESP-IDF, will be published when the release version of Arduino Core 3.0.0 is out.
So far we are in Alpha and Beta versions, although it is already in master.

peteDDD, 2023-11-28:
> > @Jason2866 Thank you. Is there a straightforward way to install that fork or should I continue to wait for the "official" release.
> 
> @peteDDD Use this
> 
> ```
> platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.11.11/platform-espressif32.zip
> ```

@Jason2866 THANK YOU!  Brilliant.   You probably know that great feeling of getting productive again... and I'm feeling it now.

```
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0
 - framework-espidf @ 3.50102.0 (5.1.2)
```

Finding a lot of interesting small issues, mostly having to do with less permissive field types for some esp_log statements.
Also, RTC_GPIO_IN_REG and some others are no where to be found, but I worked around that using higher level statements than now work in the rtc code, for example:
this...
```
#define PULSE_CNT_IS_LOW() \
  ((REG_GET_FIELD(RTC_GPIO_IN_REG, RTC_GPIO_IN_NEXT) & BIT(RAIN_CNT_RTC_GPIO_NUM)) == 0)
```
became this...
```
#define PULSE_CNT_IS_LOW() \
  (rtc_gpio_get_level(RAIN_CNT_GPIO_NUM) == 0)
```

Thanks again!

peteDDD, 2023-11-29:
@Jason2866, I just hit a road block.  One of the libraries I use, OTAdrive, depends on SPIFFS.H and that can no longer be found.  Any suggestions before I pass this over to the OTAdrive folks?  There appears to be some issue with file name lengths per this [](posting: https://community.platformio.org/t/unable-to-build-image-by-using-spiffs-h/32900)  THANKS!

Jason2866, 2023-11-29:
@peteDDD Not a bug, it is removed. The fork is made and optimized for project Tasmota.
Since SPIFFS is not maintained anymore and slow, we replaced with LittleFS.
There will no build with SPIFFS from me.

peteDDD, 2023-11-30:
> OTADRIVE.setFileSystem

@Jason2866.  Thanks again however in the context of OTAdrive, I believe OTADRIVE.setFileSystem() only applies to "resources" which is one of the three classes of things OTAdrive can manage (firmware, json config files, and "resources" such as jpg files).  I do not believe this setting will change the build dependencies for SPIFFS.H     and so far my testing agrees.   Though it works well, the doc for OTAdrive is quite lacking still.  We'll see what I learn from them tomorrow.  Thanks again.

VojtechBartoska, 2023-12-12:
Closing this as answered.

closed 2023-12-12 by VojtechBartoska