espressif/arduino-esp32 #8925: Erro de conexão TCP/IP esp32

AwanneZanca, 2023-11-26:
### Board

ESP32

### Device Description

ESP32 DEVKITV1

### Hardware Configuration

GPIO 27 CONNECTED TO A RELAY, GPIO 26 TO AN LED AND GIPO 36 TO A SOIL MOISTURE SENSOR

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windowns

### Flash frequency

 80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

This problem appears on my serial monitor after trying to make the Wi-Fi connection, I saw that it is possibly a TCP/IP error.

### Sketch

```cpp
#define pinoAnalog 36 // Define o pino A0 como "pinoAnalog"
#define pinoRele 27 // Define o pino 27 como "pinoRele"
#define PIN_LED 26 // Define o pino 26 como "PIN_LED"
#include <WiFi.h>

int ValAnalogIn; // Introduz o valor analógico ao código
 
// Configurações Wi-Fi
const char* ssid = "Irrigador-Grupo2";
const char* pass = "x";
bool ligar = false;

WiFiServer sv(80);

void setup() {

  Serial.begin(115200); 
  pinMode(pinoRele, OUTPUT); // Declara o pinoRele como Saída
  pinMode(PIN_LED, OUTPUT); // Declara o PIN_LED como Saída

  Serial.println("\n");
  WiFi.softAP(ssid, pass);
  Serial.print("Se conectando a:");
  Serial.println(ssid);

  IPAddress ip = WiFi.softAPIP();

  Serial.print("Endereço de IP:");
  Serial.println(ip);

  sv.begin();
  Serial.println("Servidor online");

}
 
void loop() {

  WiFiClient client = sv.available(); //Cria o objeto cliente

  if (client) { //Se este objeto estiver disponível
    String line = ""; //Variável para armazenar os dados recebidos

    while (client.connected()) { //Enquanto estiver conectado
      if (client.available()) { //Se estiver disponível
        char c = client.read(); //Lê os caracteres recebidos
        if (c == '\n') { //Se houver uma quebra de linha
          if (line.length() == 0) { //Se a nova linha tiver 0 de tamanho
            client.println("HTTP/1.1 200 OK"); //Envio padrão de início de comunicação
            client.println("Content-type:text/html");
            client.println();

            ValAnalogIn = analogRead(pinoAnalog); // Leitura analógica no início do bloco HTML
            int Porcento = map(ValAnalogIn, 0, 1023, 0, 100); // Relaciona o valor analógico à porcentagem

            client.println("<h1>Controle do Sistema de Irrigação</h1>");
            client.print("Porcentagem atual:"); 
            client.print(Porcento); 
            client.println();
            client.print("Ligue o irrigador clicando <a href=\"/ligar\">aqui</a><br>"); //Linha para o sensor
            client.print("Desligue o irrigador clicando <a href=\"/desligar\">aqui</a><br>"); //Linha para desligar o sensor
            client.println();
            break;
          } else {   
            line = "";
          }
        } else if (c != '\r') { 
          line += c; //Adiciona o caractere recebido à linha de leitura
        }
        if (line.endsWith("GET /ligar")) { //Se a linha terminar com "/ligar", liga o led

          digitalWrite(pinoAnalog, HIGH);               
        }
        if (line.endsWith("GET /desligar")) { //Se a linha terminar com "/desligar", desliga o led
          digitalWrite(pinoAnalog, LOW);              
        }
      }
      client.stop();
    }
  }

  ValAnalogIn = analogRead(pinoAnalog); // Relaciona o valor analógico com o recebido do sensor
  int Porcento = (100 - ValAnalogIn/100); // Relaciona o valor analógico à porcentagem
  Serial.print(Porcento); // Imprime o valor em Porcento no monitor Serial
  Serial.println("%"); // Imprime o símbolo junto ao valor encontrado
  
  if (Porcento < 70) { // Se a porcentagem for menor ou igual à
    //Serial.println("Irrigando a planta ..."); // Imprime a frase no monitor serial
    digitalWrite(pinoRele, HIGH); // Altera o estado do pinoRele para nível Alto
    digitalWrite(PIN_LED, HIGH); // Altera o estado do PIN_LED para nível Alto
  }
  
  else { // Se não ...
    //Serial.println("Planta Irrigada ..."); // Imprime a frase no monitor serial
    digitalWrite(pinoRele, LOW); // Altera o estado do pinoRele para nível Baixo
    digitalWrite(PIN_LED, LOW); // Altera o estado do PIN_LED para nível Baixo
  }
  delay (1000); // Estabelece o tempo de 1s para reinicializar a leitura
   //Para o cliente
      }
```

### Debug Message

```plain
assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x40083669:0x3ffb1fb0 0x4008bbd9:0x3ffb1fd0 0x40091059:0x3ffb1ff0 0x400eafee:0x3ffb2120 0x400fb421:0x3ffb2150 0x400fb481:0x3ffb2170 0x400ea7dd:0x3ffb21c0 0x400d46d2:0x3ffb21e0 0x400d474c:0x3ffb2230 0x400d2901:0x3ffb2250 0x400d5dd2:0x3ffb2290

ELF file SHA256: 4a2e6cd472256400

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-26:
You will need to [decode your backtrace](https://github.com/me-no-dev/EspExceptionDecoder) to see where the code is crashing. You probably need a delay in your client.read loop.

Jason2866, 2023-11-27:
GPIO 36 is input only. This may end in unexpected behaviour `digitalWrite(pinoAnalog, HIGH);`

SuGlider, 2023-12-07:
Closing this issue. Feel free to open it again.

closed 2023-12-07 by SuGlider

Fernando-Martins-Leal, 2024-12-18:
I'd experienced similar issue here after updating board on Arduino IDE to ESP32S3 version 3.1.0.
Error:  "assert failed: tcp_alloc".
Two days later burning my brain with AI helps, modifying codes and failed tests, I just rollback boards version to 3.0.0 and bingo!