espressif/arduino-esp32 #8931: WiFi completely stopped working on one of my ESP32C3 dev boards after upgrading from 2.0.0 to 2.0.14

happyme531, 2023-11-28:
### Board

NodeMCU ESP32-C3F-Kit

### Device Description

NodeMCU ESP32-C3F-Kit from AI-Thinker. 

### Hardware Configuration

None

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Manjaro

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Originally I am using this board as a ESPHome node, and it is working perfectly.
Yesterday I added a SenseAir S8 sensor (UART interface) to this node, but the node can't access the sensor properly. After a lot of debugging effort, I think the SDK version is probably too low (2.0.0 specified).
After upgrading to 2.0.14, the sensor reading seems okay, but as long as it tries to connect to WiFi, the code crashed.

Initially it is considered a ESPHome issue, but to better locate the issue, I did a minimal test using Arduino v2.0.14 as well, and the issue persists. This time not crashing, but code stucking at ` WiFi.mode(WIFI_STA);`
Switch back to v2.0.0 solves the issue, as what is happening on ESPHome.

### Sketch

```cpp
#include "WiFi.h"
void setup()
{
    Serial.begin(115200);
    Serial.println("WiFi init");
    WiFi.mode(WIFI_STA);
    Serial.println("WiFi init okay");
}

void loop()
{
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6a0
load:0x403ce710,len:0x25f4
entry 0x403cc710
WiFi init
[   168][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
```

### Other Steps to Reproduce

This problem only happened on one of my two dev boards. 
Another board, `luatos.com/t/esp32c3` works properly on SDK 2.0.14 as well.

ESPHome with ESP-IDF 4.4.5 crash at the same point as ESPHome with Arduino-ESP32 2.0.14, indicating this could be ESP-IDF's problem. Here are these two logs for reference.
[logs.txt](https://github.com/espressif/arduino-esp32/files/13487645/logs.txt)
[logs (1).txt](https://github.com/espressif/arduino-esp32/files/13487644/logs.1.txt)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

happyme531, 2023-11-28:
related: https://github.com/esphome/issues/issues/3826

Jason2866, 2023-11-28:
The issue you linked is unrelated. It is an old IDF issue. Try an official example to test.
For example this one https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino

VojtechBartoska, 2023-11-28:
Hello, there is nothing we are aware of which is not working.

lbernstone, 2023-11-28:
Erase flash before uploading

happyme531, 2023-11-28:
> The issue you linked is unrelated. It is an old IDF issue. Try an official example to test. For example this one https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino

SDK 2.0.0:

```log
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x12f4
load:0x403ce000,len:0x93c
load:0x403d0000,len:0x292c
entry 0x403ce000
I (32) boot: ESP-IDF v4.4-dev-2313-gc69f0ec32 2nd stage bootloader
I (32) boot: compile time 12:10:29
I (32) boot: chip revision: 2
I (32) boot_comm: chip revision: 2, min. bootloader chip revision: 0
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (42) boot.esp32c3: SPI Mode       : DIO
I (46) boot.esp32c3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (76) boot:  2 app0             OTA app          00 10 00010000 00140000
I (83) boot:  3 spiffs           Unknown data     01 82 00150000 000b0000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 2, min. application chip revision: 0
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=14dc8h ( 85448) map
I (120) esp_image: segment 1: paddr=00024df0 vaddr=3fc8de00 size=032e8h ( 13032) load
I (122) esp_image: segment 2: paddr=000280e0 vaddr=40380000 size=07f38h ( 32568) load
I (130) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=8d384h (578436) map
I (221) esp_image: segment 4: paddr=000bd3ac vaddr=40387f38 size=05dach ( 23980) load
I (226) esp_image: segment 5: paddr=000c3160 vaddr=50000000 size=00010h (    16) load
I (231) boot: Loaded app from partition at offset 0x10000
I (231) boot: Disabling RNG early entropy source...
[   268][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: user-hotspot

Waiting for WiFi... [   276][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   421][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[   422][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
[  2649][V][WiFiGeneric.cpp:306] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 3
[  2649][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  2653][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  2658][I][WiFiMulti.cpp:119] run(): [WIFI] 3 networks found
[  2664][D][WiFiMulti.cpp:149] run():  --->   0: [6][24:41:8C:A3:7D:CF] user-hotspot (-29)  
[  2672][D][WiFiMulti.cpp:151] run():        1: [1][30:B0:37:A0:49:10] some-wlan (-59)  
[  2679][D][WiFiMulti.cpp:151] run():        2: [6][30:B0:37:9F:F1:10] some-wlan (-60)  
[  2687][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 24:41:8C:A3:7D:CF SSID: user-hotspot Channel: 6 (-29)
[  2698][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  4949][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: user-hotspot, BSSID: 24:41:8c:a3:7d:cf, Channel: 6, Auth: OPEN
[  4951][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  7719][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (0).
.[  8768][V][WiFiGeneric.cpp:294] _arduino_event_cb(): STA Got New IP:10.42.0.194
[  8769][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  8772][D][WiFiGeneric.cpp:857] _eventCallback(): STA IP: 10.42.0.194, MASK: 255.255.255.0, GW: 10.42.0.1
[ 13121][V][WiFiGeneric.cpp:306] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 3
[ 13122][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 13127][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 13131][I][WiFiMulti.cpp:119] run(): [WIFI] 3 networks found
[ 13137][D][WiFiMulti.cpp:149] run():  --->   0: [6][24:41:8C:A3:7D:CF] user-hotspot (-29)  
[ 13145][D][WiFiMulti.cpp:151] run():        1: [1][30:B0:37:A0:49:10] some-wlan (-58)  
[ 13153][D][WiFiMulti.cpp:151] run():        2: [6][30:B0:37:9F:F1:10] some-wlan (-59)  
[ 13160][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 24:41:8C:A3:7D:CF SSID: user-hotspot Channel: 6 (-29)
[ 13172][V][WiFiGeneric.cpp:289] _arduino_event_cb(): STA Disconnected: SSID: user-hotspot, BSSID: 24:41:8c:a3:7d:cf, Reason: 8
[ 13182][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 13189][W][WiFiGeneric.cpp:829] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 13198][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[... 15 lines cut ...]
HTTP/1.1 400 Bad Request
Closing connection.
Waiting 5 seconds before restarting...
```

SDK 2.0.14: (code hangs)

```log
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6a0
load:0x403ce710,len:0x25f4
entry 0x403cc710
[   176][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: user-hotspot

Waiting for WiFi... [   182][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
```

Erased flash (`esptool.py -b 921600 erase_flash`) before each test.

The issue is very stray. Potential hardware fault in this specfic chip probably caused the problem?

Jason2866, 2023-11-28:
I think it is a hardware fault. Do you have erased the whole flash before flashing the sketch?
This is mandatory since it triggers the wifi calibration process. Without it uses the old settings.

SuGlider, 2023-11-28:
It is not a hardware fault. 
The WiFi NVS information already stored in the ESP32 has its  structure changed from IDF 4.4.5 (Arduino 2.0.13) to IDF 4,4,6 (Arduino 2.0.14).

One solution is to just update to 2.0.13.
Another is to erase flash before uploading as @lbernstone mentioned.

This is also related to https://github.com/espressif/arduino-esp32/issues/8828

Jason2866, 2023-11-28:
@SuGlider Just read now. He did erase flash test in last tests. So there cant be left overs.

SuGlider, 2023-11-28:
> @SuGlider Just read now his last post fully. He did erase flash in last tests. So there cant be left overs.

Thanks @Jason2866 for the note. I'll test it here...

SuGlider, 2023-11-29:
@happyme531 - I tried to reproduce such issue using a C3 devkit, but I couln't. It works everytime I try.

1- Erase flash
2- Flash the `WiFiClientBasic.ino` example with 2 valid SSIDs + Arduino Core 2.0.0
3- Run it (fine).
4- Flash (not erasing flash) the same example, changing the Debug Level to force it to rebuild the Core.lib with 2.0.14
5- Run it (fine).

I can't make it fail.

Looking into the logs (2 files from this issue), it seems that there are different causes for the Mediation Guru Error.
It may indicate some trouble with the reading the Flash of the C3... not sure.

Anyway, I can't reproduce this issue. Please let me know if there is something else I can try.

lbernstone, 2023-11-29:
Your code doesn't match the output.  Please actually show us how you are calling WiFiMulti.

SuGlider, 2023-11-29:
I also tested the same process with the initial sketch and I got no issues moving from 2.0.0 to 2.0.14.
I still can't reproduce the issue.

```cpp
#include "WiFi.h"
void setup()
{
    Serial.begin(115200);
    Serial.println("WiFi init");
    WiFi.mode(WIFI_STA);
    Serial.println("WiFi init okay");
}

void loop()
{
}
```

#### Output with 2.0.0:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1428
load:0x403ce000,len:0xc04
load:0x403d0000,len:0x292c
SHA-256 comparison failed:
Calculated: 9f7363434bc7a1a2434ba3062500fa10b9fce5bb859899ee0424321b4ddaf742
Expected: 9b18b42e3e8e407f5e7b13f26c80172eda36d674c584e818f50843c766ebde69
Attempting to boot anyway...
entry 0x403ce000
I (48) boot: ESP-IDF v4.4-dev-2313-gc69f0ec32 2nd stage bootloader
I (49) boot: compile time 12:10:14
I (49) boot: chip revision: 3
I (49) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (55) qio_mode: Enabling default flash chip QIO
I (59) boot.esp32c3: SPI Speed      : 80MHz
I (63) boot.esp32c3: SPI Mode       : QIO
I (67) boot.esp32c3: SPI Flash Size : 4MB
I (71) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (91) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (97) boot:  2 app0             OTA app          00 10 00010000 00140000
I (104) boot:  3 app1             OTA app          00 11 00150000 00140000
I (110) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (117) boot: End of partition table
I (120) boot_comm: chip revision: 3, min. application chip revision: 0
I (126) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=12e20h ( 77344) map
I (145) esp_image: segment 1: paddr=00022e48 vaddr=3fc8de00 size=03210h ( 12816) load
I (147) esp_image: segment 2: paddr=00026060 vaddr=40380000 size=09fb8h ( 40888) load
I (156) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=87804h (555012) map
I (234) esp_image: segment 4: paddr=000b782c vaddr=40389fb8 size=03d2ch ( 15660) load
I (237) esp_image: segment 5: paddr=000bb560 vaddr=50000000 size=00010h (    16) load
I (243) boot: Loaded app from partition at offset 0x10000
I (243) boot: Disabling RNG early entropy source...
WiFi init
WiFi init okay
```

#### Output with 2.0.14 flashed over previous 2.0.0:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
WiFi init
[   152][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   183][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   184][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
WiFi init okay

```

happyme531, 2023-11-29:
From all tests, investigations and the fact that the issue is not reproducible on any other devices, it is very likely that the issue is caused by hardware problem in this chip, which is "lucky" enough to not causing problem with old SDK (and by this mean passed factory QA tests). 
I will close this issue now. Huge thanks for all your effort!