espressif/arduino-esp32 #8934: HardwareSerial setMode fails

econeale, 2023-11-28:
### Board

Adafruit QT Py ESP32-S3

### Device Description

Adafruit QT Py ESP32-S3 with 4MB flash and 2MB PSRAM

### Hardware Configuration

ESP32-S3 using hardware UART to interface with a MAX33046E UART / RS485 transceiver. Board is configured as described [in documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#circuit-a-collision-detection-circuit).

GPIO 16 (UART RX) -> MAX3304 RO
GPIO 5 (UART TX) -> MAX3304 DI
GPIO 37 (UART RTS) -> MAX3304 DE 

### Version

v2.0.14

### IDE Name

Arduino IDE 2

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When attempting to initialize the HardwareSerial object and set the mode to RS485_HALF_DUPLEX, the `setMode` function returns false. The pin selection looks valid to me which leads me to believe one of the internal pointers is invalid and the underlying HAL implementation is producing an error but I'm not sure how to go about debugging that. 

### Sketch

```cpp
#include <HardwareSerial.h>

#define RS485_RX_PIN 16
#define RS485_TX_PIN 5
#define RS485_RTS_PIN 37

HardwareSerial RS485(1);

void setup() {
  Serial.begin(9600);
  delay(2000);

  RS485.begin(9600, SERIAL_8N1, RS485_RX_PIN, RS485_TX_PIN);
  if(!RS485.setPins(-1, -1, -1, RS485_RTS_PIN)){
    Serial.print("Failed to set RS485 pins");
  }

  if(!RS485.setHwFlowCtrlMode(HW_FLOWCTRL_RTS)) {
    Serial.print("Failed to set flow control");
  }

  if(!RS485.setMode(MODE_RS485_HALF_DUPLEX)) {
    Serial.print("Failed to set RS485 mode");
  }
}

void loop() {
  if (RS485.available()) {
    Serial.write(RS485.read());
  }
  if (Serial.available()) {
    RS485.write(Serial.read());
  }
}
```

### Debug Message

```plain
No debug messages produced; the sketch output is:

16:08:06.938 -> Failed to set RS485 pinsFailed to set RS485 mode
```

### Other Steps to Reproduce

I thought it might be related to #8755 but I'm not using the `Serial` objects provided by Arduino for the RS485 device. Switching the RS485 object to from UART 1 to UART 2 also had no impact.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-28:
Set verbose debug level in the tools menu and see if you get a better error from that.

SuGlider, 2023-11-29:
@econeale - Try it again by using just `setMode()`, with no `setHWFlowCtrlMode()`

``` cpp
void setup() {
  Serial.begin(9600);
  delay(2000);

  RS485.begin(9600, SERIAL_8N1, RS485_RX_PIN, RS485_TX_PIN);
  if(!RS485.setPins(-1, -1, -1, RS485_RTS_PIN)){
    Serial.print("Failed to set RS485 pins");
  }

//  if(!RS485.setHwFlowCtrlMode(HW_FLOWCTRL_RTS)) {
//    Serial.print("Failed to set flow control");
//  }

  if(!RS485.setMode(MODE_RS485_HALF_DUPLEX)) {
    Serial.print("Failed to set RS485 mode");
  }
}
````

You can also check the information in the ESP-IDF example at 
https://github.com/espressif/esp-idf/blob/ab03c2ea13ecaac1510b75e93b32cf0c472640fb/examples/peripherals/uart/uart_echo_rs485/

econeale, 2023-11-29:
Nothing gets reported when I increase the Core Debug level to Verbose; I would have expected _something_ which is kind of suspicious. 

But, running without setting `setHwFlowCtrlMode` seems to work. @SuGlider Is there someplace that I could add this as an example?

lbernstone, 2023-11-29:
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Serial

econeale, 2023-11-29:
Greate, thanks!

closed 2023-11-29 by econeale