espressif/arduino-esp32 #8935: OTA+UART+ESP32+arduino

wasure6, 2023-11-29:
### Related area

board  support

### Hardware specification

ESP32-D0WD-V3

### Is your feature request related to a problem?

Hello, at present, my program is implemented based on arduino software, because arduino has libraries that I rely on, then there is a problem of late maintenance: OTA.  

### Describe the solution you'd like

I want to ask, can do a serial communication level OTA?  I could customize the protocol.For example, the total flash size of the chip is 30M,0~10M is used to store boot code, 10M~20M is used to store APP code, and 20M~30M is used as a backup area to store the code to be updated. In this way, not only the BOOT can be updated, but also the APP can be updated. At this time, I need to use the address to read and write the FLASH code (library function), and the address jump to the corresponding code (or library function). How do you splice two pieces of code together and burn them to the chip? During debugging, how to view the entire flash tool?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-11-29:
You could have a bootloader that moves files around, but that is not going to be trivial. See [UF2](https://github.com/adafruit/tinyuf2).

VojtechBartoska, 2024-09-10:
Hello, this issue is quite old and there is no answer for some time.

To keep the issue tracker of this project manageable, I am closing this ticket as expired. 

If needed, you can reopen it. Thanks for understanding.

closed 2024-09-10 by VojtechBartoska