espressif/arduino-esp32 #8945: Update.setMD5() should downcase the MD5 string

jksemple, 2023-11-30:
### Board

ESP32 ESP32S3 ESP8266 etc

### Device Description

Seeedstudio ESP32S3 Sense but any Espressif device

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When requesting a firmware update from a remote OTA server using HttpUpdate the remote server returns an expected MD5 in header "x-MD5" as a 32 char hex string. If received HttpUpdate passes this to Update for comparison with an internally computed MD5 calculated using MD5Builder.
Update stores the expected MD5 using Update::setMD5(). OTA servers built on .Net will send MD5 as uppercase hex strings by default. In Update::end() the actual MD5 of the received firmware is compared with the expected MD5 and it fails because of case difference.

Update::setMD5() should not assume lowercase hex characters and should downcase the expected MD5 before storing to the _target_md5 member variable.

This was raised earlier as Issue #4641 Uppercase MD5 gets rejected but was closed due to inaction.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
Done 856064 of 856656
Done 856656 of 856656
Done 856656 of 856656
[205243][E][HttpUpdate.cpp:445] runUpdate(): Update.end failed! (MD5 Check Failed)

[205246][E][HttpUpdate.cpp:359] handleUpdate(): Update failed

[205247][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
[205256][E][HTTPClient.cpp:408] disconnect(): tcp is closed

HTTP_UPDATE_FAILED Error (7): Update error: MD5 Check Failed
```

### Other Steps to Reproduce

The problem goes away if the remote server downcases the value in the "x-MD5" header before returning it.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jksemple, 2023-11-30:
As a more general point it is a bit inconsistent to render the device's MAC address as upper case hex and MD5 checksums as lower case hex

P-R-O-C-H-Y, 2023-12-04:
Hi @jksemple, thank you for opening this issue once again. Feel free to open a PR with the proposed fix:
_"`UpdateClass::setMD5()` should not assume lowercase hex characters and should downcase the expected MD5 before storing to the `_target_md5` member variable."_

jksemple, 2023-12-05:
Ha. You beat me to it.
Thanks for fixing

PR #8959 (2023-12-05): UpdateClass::setMD5 store expected md5 in lowercase
## Description of Change
As calculated MD5 checksum is always lowercase, force `UpdateClass::setMD5` to store the expected MD5 in lowercase as some servers may return the MD5 in uppercase.

## Tests scenarios
Tested forcing Arduino OTA to receive Uppercase MD5 by espota.py. Working as expected with MD5 lowercase or uppercase.

## Related links
Closes #8945