espressif/arduino-esp32 #8949: M1pro macOS  Ventura 13.6.2  Install Error !

mu3517, 2023-12-01:
### Board

esp32

### Device Description

nothing

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

macOS  Ventura 13.6.2

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

your code https://github.com/espressif/arduino-esp32/blob/master/tools/get.py

Your installation code written here to obtain the system version has an error in performing system version identification when Python 2 and 3 are installed at the same time. The Python2 version you use by default "sys_platform = platform.platform()"

`def identify_platform():
    arduino_platform_names = {'Darwin'   : {32 : 'i386-apple-darwin',   64 : 'x86_64-apple-darwin'},
                              'DarwinARM': {32 : 'arm64-apple-darwin',  64 : 'arm64-apple-darwin'},
                              'Linux'    : {32 : 'i686-pc-linux-gnu',   64 : 'x86_64-pc-linux-gnu'},
                              'LinuxARM' : {32 : 'arm-linux-gnueabihf', 64 : 'aarch64-linux-gnu'},
                              'Windows'  : {32 : 'i686-mingw32',        64 : 'x86_64-mingw32'}}
    bits = 32
    if sys.maxsize > 2**32:
        bits = 64
    sys_name = platform.system()
    sys_platform = platform.platform()
    if 'Darwin' in sys_name and (sys_platform.find('arm') > 0 or sys_platform.find('arm64') > 0):
        sys_name = 'DarwinARM'
    if 'Linux' in sys_name and (sys_platform.find('arm') > 0 or sys_platform.find('aarch64') > 0):
        sys_name = 'LinuxARM'
    if 'CYGWIN_NT' in sys_name:
        sys_name = 'Windows'
    print('System: %s, Bits: %d, Info: %s' % (sys_name, bits, sys_platform))
    return arduino_platform_names[sys_name][bits]
`

The following content is the result of execution in my macos environment.

`gxl@gg ~ % python                                              
Python 2.7.18 (v2.7.18:8d21aa21f2, Apr 19 2020, 20:48:48) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.platform()
'Darwin-22.6.0-x86_64-i386-64bit'
>>> exit()
gxl@gg ~ % python3
Python 3.11.6 (main, Oct  2 2023, 20:46:14) [Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.platform()
'macOS-13.6.2-arm64-arm-64bit'
>>> 
`

### Sketch

```cpp
no
```

### Debug Message

```plain
no
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-12-05:
So you are saying Python2 identifies your computer wrong? What fix does make sense to you? Check and force Python3?
Python3 is the default on Mac (on 13.x as well). The worst that seems to happen is you will get the x86 toolchains (and you will run them through Rosetta2, which will result in the same binary, maybe slightly slower.

> The Python2 version you use by default

What do you mean by that? We do not use any python version by default.

mu3517, 2023-12-05:
[image]

Different versions of Python obtain different system identifiers, resulting in different downloaded installation files, which may lead to installation errors.

(https://github.com/espressif/arduino-esp32/blob/master/tools/get.py) (platform.platform()) in the installation code obtains the system identification differently. My computer is not an Intel chip , an Apple chip.

me-no-dev, 2023-12-05:
I understand. We do not force a python version in order to support older platforms. Python2 is no longer supported by it's maintainers, I believe, so I am not really sure what action should we, the esp32-arduino team, do about it. In your case, you will get the intel binaries and you will run the compilation through rosetta2, which will work just fine. Also as I said, your version of Mac OS does come with python3, so you must have installed Python2 yourself and must have made it the default Python in order to run the script with it.

If we force Python3, then we will break it for everyone still on Python2. If we leave it as-is, you will run the script with `python3 get.py` and it will work for you, else you will get the intel binaries, which will also work.

Am I missing something here?

VojtechBartoska, 2024-01-31:
Closing this as answered, there is no action needed. Thanks

closed 2024-01-31 by VojtechBartoska