espressif/arduino-esp32 #8964: Strange effect of I_ST ULP macro to jumps

BojanJurca, 2023-12-05:
### Board

ESP32 Dev Module

### Device Description

Connect GPIO 25 (used as analog output) and GPIO 34 (used as analog input) with a wire.

### Hardware Configuration

I'm ADC reading GPIO 34 but I don't think this may be the problem.

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

ULP C macro program doesn't jump properly if I_ST macro is being used before the jump. The following ULP program works as expected - it takes 31 samples, but it doesn't store them to RTC_SLOW_MEM:

            I_MOVI  (R1, 0),        // R1 = RTC_SLOW_MEM;                 // R1 is pointer to RTC_SLOW_MEM array
            I_MOVI  (R2, 0),        // R2 = 0;                            // R2 is sample counter

            M_LABEL (1),
                I_DELAY (40000),        // delay (5);                         // since ULP runs at 8 MHz, 40000 cycles correspond to 5 ms (max possible delay is 65535 cycles or 8.19 ms)
                I_DELAY (40000),
                I_ADC   (R3, 0, 6),     // R3 ≈ adc1_get_raw (ADC1_CHANNEL_6);
                I_ADDI  (R2, R2, 1),    // R2 ++
                // I_ST    (R3, R2, 0),    // RTC_SLOW_MEM [R2 + 0] = R3;        // store measurement result:  STORE 1
                // I_ST    (R2, R1, 0),    // RTC_SLOW_MEM [R1 + 0] = R2;        // store sample count         STORE 2
            I_MOVR (R0, R2),
            M_BL (1, 32),

            I_ST  (R2, R1, 0),          // RTC_SLOW_MEM [R1 + 0] = R2;        // store sample count (again) STORE 3
            I_HALT ()

Now uncomment I_ST macros. This should have absolutely no effect to the program execution. But it does. The program only takes 2 samples now:

            I_MOVI  (R1, 0),        // R1 = RTC_SLOW_MEM;                 // R1 is pointer to RTC_SLOW_MEM array
            I_MOVI  (R2, 0),        // R2 = 0;                            // R2 is sample counter

            M_LABEL (1),
                I_DELAY (40000),        // delay (5);                         // since ULP runs at 8 MHz, 40000 cycles correspond to 5 ms (max possible delay is 65535 cycles or 8.19 ms)
                I_DELAY (40000),
                I_ADC   (R3, 0, 6),     // R3 ≈ adc1_get_raw (ADC1_CHANNEL_6);
                I_ADDI  (R2, R2, 1),    // R2 ++
                I_ST    (R3, R2, 0),    // RTC_SLOW_MEM [R2 + 0] = R3;        // store measurement result:  STORE 1
                I_ST    (R2, R1, 0),    // RTC_SLOW_MEM [R1 + 0] = R2;        // store sample count         STORE 2
            I_MOVR (R0, R2),
            M_BL (1, 32),

            I_ST  (R2, R1, 0),          // RTC_SLOW_MEM [R1 + 0] = R2;        // store sample count (again) STORE 3
            I_HALT ()

### Sketch

```cpp
// This is just a test ULP C macro code. It reads 31 analog values and store them in
// RTC_SLOW_MEM [1 .. 32]. It also stores the number of samples taken into 
// RTC_SLOW_MEM [0]
//
// Connect GPIO 25, which will serve as output of analog voltage with a wire to 
// GPIO 34 where the voltage will be mesaured.

#include "driver/adc.h"   

#include "esp32/ulp.h"

        // ULP program
        const ulp_insn_t program [] = {
            I_MOVI  (R1, 0),        // R1 = RTC_SLOW_MEM;                 // R1 is pointer to RTC_SLOW_MEM array
            I_MOVI  (R2, 0),        // R2 = 0;                            // R2 is sample counter

            M_LABEL (1),
                I_DELAY (40000),        // delay (5);                         // since ULP runs at 8 MHz, 40000 cycles correspond to 5 ms (max possible delay is 65535 cycles or 8.19 ms)
                I_DELAY (40000),
                I_ADC   (R3, 0, 6),     // R3 ≈ adc1_get_raw (ADC1_CHANNEL_6);
                I_ADDI  (R2, R2, 1),    // R2 ++
                // I_ST    (R3, R2, 0),    // RTC_SLOW_MEM [R2 + 0] = R3;        // store measurement result:  STORE 1
                // I_ST    (R2, R1, 0),    // RTC_SLOW_MEM [R1 + 0] = R2;        // store sample count         STORE 2
            I_MOVR (R0, R2),
            M_BL (1, 32),

            I_ST  (R2, R1, 0),          // RTC_SLOW_MEM [R1 + 0] = R2;        // store sample count (again) STORE 3
            I_HALT ()
        };
    size_t load_addr = 0;
    size_t size = sizeof (program)/sizeof (ulp_insn_t);

void setup () {
    Serial.begin (115200);

    int voltage = 100; // 0 = 0 V .. 255 = 3.3V
    dacWrite (25, voltage);

    memset (RTC_SLOW_MEM, 0, CONFIG_ULP_COPROC_RESERVE_MEM);

    // load ULP program
    ESP_ERROR_CHECK (ulp_process_macros_and_load (load_addr, program, &size));

    // run ULP program 
    ESP_ERROR_CHECK (ulp_run (load_addr));

    // wait until ULP finishes
    while ((RTC_SLOW_MEM [0] & 0xFFFF) < 15) { // only 16 LS bits of RTC_SLOW_MEM [0] are useful
        delay (10);
        Serial.printf ("measurements: %i\n", (RTC_SLOW_MEM [0] & 0xFFFF));
    }

    // output the measurements
    for (int i = 1; i <= (RTC_SLOW_MEM [0] & 0xFFFF); i++) 
        Serial.printf ("  %4i", ~RTC_SLOW_MEM [i] & 0xFFF); // only 12 LS bits are used for ADC conversion and (at least with some ESP32 boards) the bits are inverted
    Serial.printf ("\n");
}

void loop () {

}
```

### Debug Message

```plain
No error message.
```

### Other Steps to Reproduce

Upload the sketch to ESP32 and run it.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-05:
You should post at https://github.com/espressif/esp-idf/issues. This has nothing to do with arduino-esp32.

me-no-dev, 2023-12-05:
Yup. ULP instruction debugging is outside of Arduino context.
However:
You did not specify what version Arduino you have installed, therefore we do not know which version of ESP-IDF you are running on.
Could you please specify that?

closed 2023-12-06 by BojanJurca