espressif/arduino-esp32 #8975: IDF >v5.1 required for ardunio as a component?

tongclement, 2023-12-09:
### Board

ESP32 Dev Module

### Device Description

N.A.

### Hardware Configuration

N.A.

### Version

other

### IDE Name

VSCode (ESP IDF)

### Operating System

macOS 

### Flash frequency

N.A.

### PSRAM enabled

yes

### Upload speed

N.A.

### Description

When installing Arduino as a component following these instructions: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

The error message was that idf >=5.1 is required.
However, I read everywhere that currently v4.4 should be used instead. 
Did I accidently download a development branch?
Many thanks

Full error message: 
```
...
Executing "ninja version"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
Skipping optional dependency: espressif/esp-sr
Skipping optional dependency: espressif/esp-sr
.......................................................................Skipping optional dependency: espressif/esp-sr
CMake Error at /Users/username/esp/esp-idf/tools/cmake/build.cmake:452 (message):
  HINT: Please check manifest file of the following component(s): arduino

  ERROR: Because project depends on idf (>=5.1) which doesn't match any
  versions, version solving failed.

Call Stack (most recent call first):
  /Users/usernamel/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:8 (project)
...
```

### Sketch

```cpp
N.A.
```

### Debug Message

```plain
N.A.
```

### Other Steps to Reproduce

Run 
```
mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
idf.py menuconfig
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tongclement, 2023-12-09:
currently on ESP-IDF v4.4.3-dirty

Jason2866, 2023-12-09:
Arduino branch master needs IDF 5.1 to compile.

SuGlider, 2023-12-10:
Try this sequence using idf v4.4:

1- Go to you IDF 4.4.3-dirty and update it to IDF 4.4.6:
git fetch
git checkout release/v4.4
git pull
git submodule update
./install.sh
. ./export.sh
idf.py --version 

this last step shall display v.4.4.6 as the current version, which is the same from Arduino Core 2.0.14.

2- copy the all files from our Arduino as Component Example:
 https://github.com/espressif/arduino-esp32/tree/master/idf_component_examples/Hello_world
into any folder, for instance, `/myProj`

Default target is ESP32.

3- This example will use Arduino from Managed Components. There is no need to clone it.
But there are few changes to be done to the `/myProj/main/idf_omponent.yml` file:
Comment out the last 2 line by adding a `#` symbol:
```
#    override_path: '../../../'
#    pre_release: true
```

4- Just build the example that will print `Hello world!` every second in the UART0 (console):
(from `/myProj` folder, just build it and wait it to start the serial monitor from the idf.py tool)

idf.py -p /dev/ttyUSB0 flash monitor 

5- Change `myProj/main/main.cpp` to the sketch you need to run. The example is this:
``` cpp
#include "Arduino.h"

void setup(){
  Serial.begin(115200);
}

void loop(){
    Serial.println("Hello world!");
    delay(1000);
}
```

me-no-dev, 2023-12-11:
master branch of this repo is to be used with ESP-IDF v5.1, while the `release/v2.x` branch is to work with ESP-IDF v4.4

4444monkey, 2023-12-14:
Hi, I got same questiion here.
The installation steps per https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html are still baffling me. My cuurent version is ESP-IDF 4.4.6 wht Windows system 11 based.  I followed the same steps and got the error message was that" idf >=5.1 is required" as well. That means the installation guide stated in this website has not supported the version before 5.1, correct?

Is there any clear procedure to integrate the Arduino as component in such old version like 4.4.6?
I've tried another method by directly download the zip file of the corresponding version 2.0.14. and added it into project components folder running idf.py build. Somehow I cannot make it.

me-no-dev, 2023-12-14:
Difference is exactly what I said above @4444monkey . Instead of master branch of this repo, you use the `release/v2.x` branch. Master is for IDF 5.1 and release/v2.x is for IDF 4.4

4444monkey, 2023-12-14:
Thanks for your clarification but where I can I find the guide to configure the previous release/v2.x is for IDF 4.4?
Seems it is not quite handy to dealt with it.

me-no-dev, 2023-12-14:
@4444monkey instead of `git clone https://github.com/espressif/arduino-esp32.git arduino`, you do `git clone https://github.com/espressif/arduino-esp32.git -b release/v2.x arduino`, or since you have already cloned it, you cd into the folder and `git checkout release/v2.x`

4444monkey, 2023-12-15:
Thanks again! I suddenly see the light! 
If I use "release/v2.x arduino", should I need to run this code "git submodule update --init --recursive"?
Then, I don't need to use this part https://github.com/espressif/esp32-arduino-lib-builder, right?

left-win, 2023-12-29:
> @4444monkey instead of `git clone https://github.com/espressif/arduino-esp32.git arduino`, you do `git clone https://github.com/espressif/arduino-esp32.git -b release/v2.x arduino`, or since you have already cloned it, you cd into the folder and `git checkout release/v2.x`

Thank you so much, Sir. You've been very helpful

VojtechBartoska, 2024-01-31:
Closing this as answered.

closed 2024-01-31 by VojtechBartoska