espressif/arduino-esp32 #8976: WiFiProv (SofAP mode) not working if service_key (Password used for SofAP method) is set

dsilletti, 2023-12-09:
### Board

ESP32-S3-DevKitC

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

macOS 14.0

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

WiFiProv (SofAP mode) it's not working if service_key (Password used for SofAP method) is set.

The generated QR code does not include the password to connect to the SoftAP.

No issues setting service_key to NULL (SofAP with no Password)

Additional issue: bool reset_provisioned = true; does not delete previosly provisioned data on boot, I had to add  these 2 lines in the setup too get the WiFi config cleared on boot:
```
  WiFi.mode(WIFI_MODE_NULL);
  WiFi.eraseAP();
```

### Sketch

```cpp
/*
Please read README.md file in this folder, or on the web:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFiProv/examples/WiFiProv

Note: This sketch takes up a lot of space for the app and may not be able to flash with default setting on some chips.
  If you see Error like this: "Sketch too big"
  In Arduino IDE go to: Tools > Partition scheme > chose anything that has more than 1.4MB APP
   - for example "No OTA (2MB APP/2MB SPIFFS)"
*/

#include "WiFiProv.h"
#include "WiFi.h"

#define USE_SOFT_AP // Uncomment if you want to enforce using the Soft AP method instead of BLE
const char * pop = "abcd1234"; // Proof of possession - otherwise called a PIN - string provided by the device, entered by the user in the phone app
const char * service_name = "PROV_test"; // Name of your device (the Espressif apps expects by default device name starting with "Prov_")
const char * service_key = "testpassword"; // Password used for SofAP method (NULL = no password needed)
bool reset_provisioned = true; // When true the library will automatically delete previously provisioned data.

// WARNING: SysProvEvent is called from a separate FreeRTOS task (thread)!
void SysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        Serial.print("\nConnected IP address : ");
        Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
        break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        Serial.println("\nDisconnected. Connecting to the AP again... ");
        break;
    case ARDUINO_EVENT_PROV_START:
        Serial.println("\nProvisioning started\nGive Credentials of your access point using smartphone app");
        break;
    case ARDUINO_EVENT_PROV_CRED_RECV: {
        Serial.println("\nReceived Wi-Fi credentials");
        Serial.print("\tSSID : ");
        Serial.println((const char *) sys_event->event_info.prov_cred_recv.ssid);
        Serial.print("\tPassword : ");
        Serial.println((char const *) sys_event->event_info.prov_cred_recv.password);
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL: {
        Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
        if(sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR)
            Serial.println("\nWi-Fi AP password incorrect");
        else
            Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
        Serial.println("\nProvisioning Successful");
        break;
    case ARDUINO_EVENT_PROV_END:
        Serial.println("\nProvisioning Ends");
        break;
    default:
        break;
    }
}

void setup() {
  Serial.begin(115200);

  WiFi.onEvent(SysProvEvent);

// BLE Provisioning using the ESP SoftAP Prov works fine for any BLE SoC, incuding ESP32, ESP32S3 and ESP32C3.
#if CONFIG_BLUEDROID_ENABLED && !defined(USE_SOFT_AP)
    Serial.println("Begin Provisioning using BLE");
    // Sample uuid that user can pass during provisioning using BLE
    uint8_t uuid[16] = {0xb4, 0xdf, 0x5a, 0x1c, 0x3f, 0x6b, 0xf4, 0xbf,
                        0xea, 0x4a, 0x82, 0x03, 0x04, 0x90, 0x1a, 0x02 };
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name, service_key, uuid, reset_provisioned);
    log_d("ble qr");
    WiFiProv.printQR(service_name, pop, "ble");
#else
    Serial.println("Begin Provisioning using Soft AP");
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name, service_key);
    log_d("wifi qr");
    WiFiProv.printQR(service_name, pop, "softap");
#endif
}

void loop() {
}
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Use ESP SoftAP Prov app from Espressif, scan the qr code provided in console

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-09:
I haven't looked into the QR issue, but the reset_provisioned isn't happening because the beginProvision with SoftAP does not include include it as a parameter.  Try putting `, reset_provisioned = reset_provisioned` at the end of that line.

dsilletti, 2023-12-09:
thank you @lbernstone , correct, the example was missing the reset_provisioned parameter for the SoftAp case.
The SoftAP example should be fixed to:
```WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP,` WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name, service_key, NULL, reset_provisioned);
```

for the main issue, the SoftAP with password, I don't see any option  in the library to generate a QR code that includes the password:

```void printQR(const char *name, const char *pop, const char *transport);```

so I don't see how the mobile can connect to the SoftAP if the password is not provided via QR code.

dsilletti, 2023-12-09:
OK found the solution, need to add the "password" key value to the qr code json.

This works perfectly:

https://espressif.github.io/esp-jumpstart/qrcode.html?data={%22ver%22:%22v1%22,%22name%22:%22PROV_test%22,%22pop%22:%22abcd1234%22,%22transport%22:%22softap%22,%22password%22:%22test1234%22}

doudar, 2024-03-15:
I'm running into this issue as well with Esp32's that had previous versions of this library installed and subsequently upgraded firmware to the the new library. Calling WiFi.eraseAP() once after the firmware upgrade solves the issue permanently (no need to call it on subsequent boots).

closed 2025-01-16 by Parsaabasi