espressif/arduino-esp32 #8977: ANY Arduino build for ESP32-s2 for board = lolin_s2_mini fails with USBSerial not declared in scope of HardwareSerial.h (IDFGH-11650)

peteDDD, 2023-12-09:
### Board

ESP32-S2  lolin_s2_mini

### Device Description

(Wemos) lolin_s2_mini

### Hardware Configuration

Not relevant to the issue.

### Version

v2.0.14

### IDE Name

Visual Studio Code with Platformio

### Operating System

Windows 10

### Flash frequency

does not matter

### PSRAM enabled

no

### Upload speed

does not matter

### Description

ANY CODE built with espidf @ 3.50102.0 (5.1.2)/arduinoespressif32 @ 3.0.0 specifying  lolin_s2_mini as the board and with CONFIG_AUTOSTART_ARDUINO=y fails in
**HardwareSerial.h:360:16:  with error: 'USBSerial' was not declared in this scope**

### Sketch

```cpp
#include <Arduino.h>

void setup()
{}

void loop()
{}
```

### Debug Message

```plain
In file included from C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/Arduino.h:196,
                 from C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/chip-debug-report.cpp:17:
C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/chip-debug-report.cpp: In function 'void printBeforeSetupInfo()':     
C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/HardwareSerial.h:360:16: error: 'USBSerial' was not declared in this scope; did you mean 'Serial'?
  360 | #define Serial USBSerial
      |                ^~~~~~~~~
C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/chip-debug-report.cpp:260:3: note: in expansion of macro 'Serial'
```

### Other Steps to Reproduce

Using 
 - framework-arduinoespressif32 @ 3.0.0
 - framework-espidf @ 3.50102.0 (5.1.2)
 which was downloaded clean on 9 December 2023 at 13:00 UTC

Platformio.ini:
```
[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = espidf, arduino
```
sdkconfig includes:
`CONFIG_AUTOSTART_ARDUINO=y`

Proceed to build the most simple of sketches using platformio build button or command

(For the record, no issue doing the same with esp32-s3-devkitc-1 including using printf and ESP_LOG).

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-10:
Are you using the [sdkconfig](https://github.com/espressif/arduino-esp32/blob/2.0.14/tools/sdk/esp32s2/sdkconfig) from arduino-esp32?
Note that in >3.0 it will be located at https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32s2/sdkconfig

peteDDD, 2023-12-10:
> Are you using the [sdkconfig](https://github.com/espressif/arduino-esp32/blob/2.0.14/tools/sdk/esp32s2/sdkconfig) from arduino-esp32?
> Note that in >3.0 it will be located at https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32s2/sdkconfig

@lbernstone Thank you for this information.  

As you likely know, Platformio is in charge of loading the "correct" sdkconfig file for each project.   And in fact "forces" the one it thinks is correct for the framework and board specified.

I did a diff on the one from the link you provided and it is quite different, with about 340 more lines,  compared to the one that platformio is loading.   

**I tried:**
1: Copying the "new" sdkconfig file to sdkconfig_lolin_S2_mini in my project directory.  As soon as the build starts, platformio replaces the file with the same one it was loading before.
2:  Copying the "new" sdkconfig file to sdkconfig in my project directory.  As soon as the build starts, platformio replaces sdkconfig_lolin_S2_mini file with the same one it was loading before.

So it appears the problem may be that these settings in platformio.ini
 - framework-arduinoespressif32 @ 3.0.0
 - framework-espidf @ 3.50102.0 (5.1.2)
do not load the correct sdkconfig files.

**How to fix that?**   Is there a platformio.ini setting I can use for now to force the use of the esp32-arduino-libs?  Or is there an installation of esp32-arduino-libs that I need to do beyond just specifying the following in my platformio.ini?
```
[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = espidf, arduino
```

This is odd because, for example, my same projects build just fine when I specify
`board = esp32-s3-devkitc-1`

peteDDD, 2023-12-10:
further... I tried the following platformio.ini per the installation instructions
```
[env:lolin_s2_mini]
platform = https://github.com/platformio/platform-espressif32.git
board = lolin_s2_mini
framework = arduino
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

and got the following errors
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_s2_mini.html
PLATFORM: Espressif 32 (6.4.0+sha.ff6ec88) > WEMOS LOLIN S2 Mini
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.51cb927 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
KeyError: 'framework-arduinoespressif32-libs':
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\peted\.platformio\platforms\espressif32@src-5f117260f75b328038ec9d3fd0e14a68\builder\main.py", line 346:
    target_elf = env.BuildProgram()
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\peted\.platformio\platforms\espressif32@src-5f117260f75b328038ec9d3fd0e14a68\builder\frameworks\arduino.py", line 41:
    SConscript(
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\peted\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\peted\.platformio\packages\framework-arduinoespressif32\tools\platformio-build.py", line 40:
    FRAMEWORK_LIBS_DIR = platform.get_package_dir("framework-arduinoespressif32-libs")
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 32:
    pkg = self.get_package(name)
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\peted\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 21:
    owner=self.packages[name].get("owner"),
========================== [FAILED] Took 1.22 seconds ==========================
```

peteDDD, 2023-12-10:
UPDATE: 

1- I uninstalled ALL espressif platforms from Platformio/Visual Code Studio. 
2- Set platformio.ini as follows
```
[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = arduino
```
3- When I saved the platformio, tool manager took over, as usual, and indicated: 
`Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20014.0`
4- Build now works!

BUT NOW... my ESP32-S3 projects won't build...
stdini.h, stdio.h, string.h, stdlib.h,  and others not found...

lbernstone, 2023-12-10:
All the platformio glue is maintained at https://github.com/platformio/platform-espressif32. They will be able to address your problem better than we can.

SuGlider, 2023-12-10:
@Jason2866 is one of the Tasmota maintainers. 
He has tons of experience with PlatformIO.
I think that he can guide you.

peteDDD, 2023-12-10:
Yes, indeed, @Jason2866 is brilliant and generous with his advice.

VojtechBartoska, 2023-12-11:
Hello @peteDDD, probably we will be not able to help you. 

Please ask in PIO issue tracker, you can also take a look on PIO docs [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html?highlight=platformio#installing-using-platformio:~:text=Installing%20using%20PlatformIO-,%EF%83%81,-PlatformIO%20is%20a)

Jason2866, 2023-12-11:
@peteDDD Can you upload a complete failing setup to your github in a new repo?
Will have a look, it is better to have everything. Probably just a small detail.

peteDDD, 2023-12-11:
> @peteDDD Can you upload a complete failing setup to your github in a new repo?
> Will have a look, it is better to have everything together, than talking about pieces. Probably just a small detail.

a little too big to upload here, so saved to GoogleDrive:   [hello_world_wemos_S2_mini](https://drive.google.com/file/d/1mUPO2k_TRod8eveX6OtmdLD7ykbiZuaw/view?usp=sharing)

In the project  you can see that it will build and run fine for ESP32-S3-devkitc-1 but for ESP32-S2 lolin-S2-mini.

@Jason2866 Thanks for your help here.
Pete

SuGlider, 2023-12-11:
@peteDDD - I have taken a look at your project. It is a mixing of PlatformIO and Arduino as Component.
How are you building it?

PIO doesn't use `sdkconfig` file at all. It does the same as Arduino IDE and uses precompiled IDF libraries.

SuGlider, 2023-12-11:
>In the project you can see that it will build and run fine for ESP32-S3-devkitc-1 but for ESP32-S2 lolin-S2-mini.

S3-devkitc-1 has 2 USB ports. One is UART, the other one is USB OTG or CDC JTAG. Which one do you use?
ESP32-S2 lolin-S2-Mini has only USB OTG port. It needs the firmware to enable 2 flags in order to make it build and work correclty:

1) ARDUINO_USB_CDC_ON_BOOT=1 for turning ON USB port
2) ARDUINO_USB_MODE=0  for selecting USB OTG mode

If you are build it using VSCode IDF tools (not PIO), you must add it to the CMAKE file.
if you are building it using VSCode PIO tools, you must add it to the `platformio.ini` file.

SuGlider, 2023-12-11:
By the way... I have just created a PIO project with your code. I also have a Lolin S2 board and it works perfectly fine using default setting.

My `platformio.ini` file is this:
```
[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = arduino
build_flags = 
	-D ARDUINO_USB_MODE=0
	-D ARDUINO_USB_CDC_ON_BOOT=1
```

I see no issue at all with PlatformIO and Lolin S2 Mini.

peteDDD, 2023-12-11:
Thank you.  Where can one read about those flags?  Did not have to set
these in the past (that is prior to esp-idf 5).  This changed overnight
when moving from 4.4 to 5.0/5.1

SuGlider, 2023-12-11:
These flags are there since Arduino Core 2.0.1 (2021).
It can be found searching [PIO Community](https://community.platformio.org/tag/espressif32) for `CDC USB`.

SuGlider, 2023-12-11:
This is the PIO project I have created, if necessary. It prints and blinks the S2 Lolin Mini LED from GPIO15.

[PIO_S2_wemos_Blink.zip](https://github.com/espressif/arduino-esp32/files/13639923/PIO_S2_wemos_Blink.zip)

SuGlider, 2023-12-11:
@peteDDD - Can this issue be considered solved and then I can close it? Thanks.

Jason2866, 2023-12-11:
Some questions. Do you want to build an Arduino project or Arduino as a component of IDF?
Arduino does compile fine with the S2. Arduino as component does not compile, probably missing a setting in sdkconfig.

Platformio has the possibility to compile Arduino as component without manualy installing IDF.
Platformio does install everything needed.`menuconfig` is called from Platformio with `pio run -t menuconfig` to do the sdkconfig settings. This is the point here. This settings needs to be done! That it compiles with the S3 is just a luck punch.

If there is no need, do not use the setup Arduino as component from IDF.

Jason2866, 2023-12-11:
@peteDDD If you want to use latest Alpha3 build, there is no official one.
You can use my fork with:
```
[env:lolin_s2_mini]
platform = https://github.com/Jason2866/platform-espressif32.git
board = lolin_s2_mini
framework = arduino
```
The build flags are not needed, since the are already set in the boards manifest.

Jason2866, 2023-12-11:
@SuGlider The issue can be closed. S2_mini does compile with current official Platformio Arduino core 2.0.14 and with my fork with Arduino 3.0.0-alpha3.

SuGlider, 2023-12-11:
Thanks @Jason2866 for the help and investigation. 
@peteDDD - Fell free to open this issue again, whenever necessary.

closed 2023-12-11 by SuGlider

peteDDD, 2023-12-11:
> The build flags are not needed, since the are already set in the boards manifest.

@Jason2866... that is exactly what I expected would be when I started the project from Platformio...

Is it that your "custom" version of platform = https://github.com/Jason2866/platform-espressif32.git has different board manifest files?

peteDDD, 2023-12-11:
Thank you @SuGlider.  

> These flags are there since Arduino Core 2.0.1 (2021).
> It can be found searching [PIO Community](https://community.platformio.org/tag/espressif32) for CDC USB.

That's nice to know but how would one know to even look for those flags (amongst the many, many platformio flags)?  The platformio page on the board does not make mention of these.   Why is this not simply in the manifest files for the board.  Perhaps that is the real point...  I could build with this board without those flags one day and then the next... moving to IDF 5.1, I could not.   This, to me, seems like someone dropped a ball somewhere (and that is why I filed an issue).  Do you disagree with that perspective?

Thanks again!

Jason2866, 2023-12-11:
>  when I invoke "framework = arduino, espidf" is this what is being referred to a "arduino as a component"?

Yes. When doing this you have to do settings in sdkconfig and in Platformio since technically it is an IDF project. Platformio provides "menuconfig" for sdkconfig settings.

With Platformio you can add IDF components and Arduino Librarys.
Be aware such a setup has a learning curve ;-).  You can ask for help here https://community.platformio.org/tag/espressif32

SuGlider, 2023-12-11:
@peteDDD 
>That's nice to know but how would one know to even look for those flags (amongst the many, many platformio flags)?

It seems clearly to be a PlatformIO support's fault...

I have found out that the default "defines" from PlatformIO github for this board is incomplete.
https://github.com/platformio/platform-espressif32/blob/develop/boards/lolin_s2_mini.json

It is missing `"-DARDUINO_USB_MODE=0"` there i the `extra_flags`. 
You can open an issue to their github nad ask them to fix it. :-)

```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s2_out.ld"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_LOLIN_S2_MINI",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_CDC_ON_BOOT=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "hwids": [
      [
        "0X303A",
        "0x80C2"
      ]
    ],
    "mcu": "esp32s2",
    "variant": "lolin_s2_mini"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s2.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "WEMOS LOLIN S2 Mini",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.wemos.cc/en/latest/s2/s2_mini.html",
  "vendor": "WEMOS"
}
```

Jason2866, 2023-12-11:
> I understand this question. These flags are part of the ESP32 Arduino Core code.
PlatformIO uses exactly the same code from our Core (ESP32 Core Github).
I think that PIO support could add these options in the PIO menu...
We have added it to the Arduino IDE Menu.
But this is a question to the PIO developers, I guess.

Here are all boards supported from Platformio https://github.com/platformio/platform-espressif32/tree/develop/boards
In the json manifests the default options are set to get a booting board.

SuGlider, 2023-12-11:
> Here are all boards supported from Platformio https://github.com/platformio/platform-espressif32/tree/develop/boards In the json manifests the default options are set to get a booting board.

We re in sync, @Jason2866 !!! :-)

Jason2866, 2023-12-11:
The way via boards name is a dead end. There are many boards with the same name and different configuration sold.

SuGlider, 2023-12-11:
Is there any documentation/tutorial about this Tasmota feature within PIO? Or it is Tasmota building process only?

Jason2866, 2023-12-11:
Better it is in official docu https://docs.platformio.org/en/latest/projectconf/sections/platformio/options/directory/boards_dir.html

peteDDD, 2023-12-11:
> I think that PIO support could add these option in the PIO menu...

@Rodrigo Garcia

> This is the PIO project I have created, if necessary. It prints and blinks
> the S2 Lolin Mini LED from GPIO15.
> PIO_S2_wemos_Blink.zip
> <https://github.com/espressif/arduino-esp32/files/13639923/PIO_S2_wemos_Blink.zip>

I loaded, built, and ran that project.  No output from printf.  Did you get
printf output?
For good measure, I added:
[CODE]
 static const char *TAG = "Setup";
ESP_LOGE(TAG,"Hello from LOGE");
[/CODE]

No joy with either printf or ESP_LOGE

SuGlider, 2023-12-11:
@peteDDD - printf() and ESP_LOGE() are IDF API. Those functions will send the output to UART, not to Native USB CDC.
Use Arduino API instead: `log_printf()` in the same that you use `printf()` 
Example: `log_printf("Var i = %d", i);`

### Another important point:

In Arduino, `log_printf()` will also send its output to UART by default.
In order to redirect it to ESP32-S2 USB CDC, it is necessary to explicitaly declare it using `Serial.setDebugOutput(true);` in the sketch.

Due to the USB CDC settings, `Serial` will actually be just be `#define Serial USBCDC` which means that it uses all APIs from 
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/USBCDC.h

About Arduino general logging, you can read the code in 
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-log.h

Example:

``` cpp
void setup() {
  Serial.begin(); // no baudrate is only valid for USB CDC. Using Lolin S2 settings, USB CDC is enabled.
  Serial.setDebugOutput(true);
  while(!Serial) delay(100); // waits for USB CDC to detect that the Serial Monitor is active
}

void loop() {
  Serial.println("Loop - using Serial Print to USB CDC.");
  log_printf("Loop - using Logging");
  delay(1000);
}
```

SuGlider, 2023-12-11:
Another way would be to use Arduino `log_e();` like in `ESP_LOGE()`, but with no TAG.

For that you must add the LOG Level to the platformio.ini:
```
[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = arduino
build_flags = 
	-D ARDUINO_USB_MODE=0
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-D CORE_DEBUG_LEVEL=5
```

Pick the Level that you need from:
```
#define ARDUHAL_LOG_LEVEL_NONE       (0)
#define ARDUHAL_LOG_LEVEL_ERROR      (1)
#define ARDUHAL_LOG_LEVEL_WARN       (2)
#define ARDUHAL_LOG_LEVEL_INFO       (3)
#define ARDUHAL_LOG_LEVEL_DEBUG      (4)
#define ARDUHAL_LOG_LEVEL_VERBOSE    (5)
```

Use `log_e()`, `log_w()`, `log_i()`, `log_d()`, `log_v()`.

Example:

``` cpp
void setup() {
  Serial.begin(); // no baudrate is only valid for USB CDC. Using Lolin S2 settings, USB CDC is enabled.
  Serial.setDebugOutput(true);
  while(!Serial) delay(100); // waits for USB CDC to detect that the Serial Monitor is active
}

void loop() {
  static uint32_t c = 0;
  Serial.println("Loop - using Serial Print to USB CDC.");
  log_e("Loop - using Logging Count = %d", c++);
  delay(1000);
}
```

SuGlider, 2023-12-11:
@peteDDD - You shall now have all necessary tools.

peteDDD, 2023-12-11:
So I can't use ESP_LOG or printf with the board...   I have those used extensively in all of my code, across many ESP32 boards and projects.  Sigh.

So where does `framework = arduino, espidf` come in so I can use my usual esp tool set?

SuGlider, 2023-12-12:
How will you make ESP_LOG print to the USB CDC using IDF?
I think that IDF doesn't support USB CDC output for logging, therefore, `ramework = arduino, espidf` may not help.

My suggestion is to redefine all "ESP_LOGX()", like this:

``` cpp
#if 1
#undef ESP_LOGE
#undef ESP_LOGI
#undef ESP_LOGD
#undef ESP_LOGW
#undef ESP_LOGV
#define ESP_LOGE(tag, format, ...) log_e(format, ##__VA_ARGS__)
#define ESP_LOGI(tag, format, ...) log_i(format, ##__VA_ARGS__)
#define ESP_LOGD(tag, format, ...) log_d(format, ##__VA_ARGS__)
#define ESP_LOGW(tag, format, ...) log_w(format, ##__VA_ARGS__)
#define ESP_LOGV(tag, format, ...) log_v(format, ##__VA_ARGS__)
#endif
```

By adding this to some general `myProj.h` that is included in all places of you project,  you will get it working fine.

peteDDD, 2023-12-12:
Thank you.  That is a reasonable workaround.  I have to also test
ets_printf...might just have to just get away from this board 🤔

SuGlider, 2023-12-12:
This board is nice and it has a good price.
Take a look at the USBCDC.cpp implementation of setDebugOutput() and also in SerialHardware.cpp.

Both have the combination to deal with USB CDC and ets_xxxx implementation.

Maybe you may want to go ahead and create your own functions for the Lolin S2 Mini.

Good luck!

SuGlider, 2023-12-12:
> > @SuGlider Most boards Manifests are community provided. I did PRs to correct some. Even board manufacturers have already wrong provided...
> 
> Thank Jason! PR sent to PIO GH. [platformio/platform-espressif32#1251](https://github.com/platformio/platform-espressif32/pull/1251)

@peteDDD  - The fix (PR) was merged. 
Lolin S2 Mini shall work with no need to change anything in the platformio.ini whenever this is released to the public use.

ErikBasalt, 2023-12-16:
I'm having a similar issue with the serial USB too:
Using lolin_s2_mini board (ESP32-S2), connected with USB-C to my Windows 10 PC, running VSC/Platformio with Arduino framework. Using "Upload and Monitor" works fine, I see the upload and the VSC Terminal connects properly and I see all the logging.

However, when I reset the ESP device (or let it crash), the connection with the VSC Terminal gets lost, and I cannot see restart logging. VSC Terminal only gets active again when I press a button in the Terminal first. When I use Putty, I even have to start a new window. This hinders me in seeing any diagnostic info why the ESP device restarted (or where it crashed).

I tried "-DARDUINO_USB_CDC_ON_BOOT=1 -D ARDUINO_USB_MODE=0" in platformio.ini and/or lolin_s2_mini.json without success.

How can I keep the VSC Terminal (or Putty) connected during restart, and showing all the startup logging?

peteDDD, 2023-12-16:
Erik,
You may find CoolTerm more resilient than Putty or the VSC terminal.
Pete

ErikBasalt, 2023-12-17:
Thanks for the tip, I tried CoolTerm and indeed it reconnects automatically (after briefly showing Disconnected and all "LEDs" in the bottom right corner going off and on).

Now I can easily see all my logging before the restart (or crash) and my program's startup logging.

However, still I'm not getting any logging info about the reason for the restart (e.g. me pressing the Reset button, calling ESP.restart() or performing a deliberate divide-by-zero).

Shouldn't the ESP32-S2 show backtrace and generic boot  info to the USB serial? So in case of a bug I can find the erratic code line?

peteDDD, 2023-12-17:
Erik, 
Please post your full platformio.ini

ErikBasalt, 2023-12-17:
Hi Pete,

Thanks for taking interest in this problem. My platformio.ini (omitted some comment lines):
`[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
;some unsuccesful trials in comment below
;build_flags = 
;	-D ARDUINO_USB_MODE=0
;	-D ARDUINO_USB_CDC_ON_BOOT=1
;build_flags = -D ARDUINO_USB_CDC_ON_BOOT=1 -D ARDUINO_USB_MODE=0
;build_flags = -DARDUINO_USB_CDC_ON_BOOT=1
;build_unflags = ARDUINO_USB_MODE
framework = arduino
lib_deps = 
	SPI
	bblanchon/ArduinoJson@^6.21.4
	esphome/AsyncTCP-esphome@^2.0.1
	ottowinter/ESPAsyncWebServer-esphome@^3.1.0
	majicdesigns/MD_MAX72XX@^3.5.1
	majicdesigns/MD_Parola@^3.7.1
	makuna/NeoPixelBus@^2.7.7
	wnatth3/WiFiManager@^2.0.16-rc.2
	bertmelis/espMqttClient@^1.5.0
board_build.filesystem = littlefs
;monitor_speed = 115200
;monitor_echo = yes`

The first part of my (Arduino) setup function:
`void setup() {
    Serial.begin(115200);
    /*
        while (!Serial) { // don't wait for serial on ESP32-S2, otherwise it will require the serial monitor during runtime
            yield();
        } // wait for serial attach
    */
    delay(5000); // wait until Platformio monitor is active
    ...etc...`

ErikBasalt, 2023-12-17:
Hmmm, that <> Code option didn't work as expected, the platformio.ini file is not very readable like this. Instead, please find it attached in this post (had to add .txt extention)
[platformio.ini.txt](https://github.com/espressif/arduino-esp32/files/13697641/platformio.ini.txt)

peteDDD, 2023-12-17:
Erik, when you are entering your code, you can mouse over the code and then click on the "<>" icon for code.  That will format it. .. but you do need some carriage returns.  I think this is what you have...

`[env:lolin_s2_mini] 
platform = espressif32 
board = lolin_s2_mini
 ;some unsuccesful trials in comment below
 ;build_flags =  
 ;	-D ARDUINO_USB_MODE=0
 ;	-D ARDUINO_USB_CDC_ON_BOOT=1
 ;build_flags = -D ARDUINO_USB_CDC_ON_BOOT=1 -D ARDUINO_USB_MODE=0
 ;build_flags = -DARDUINO_USB_CDC_ON_BOOT=1
 ;build_unflags = ARDUINO_USB_MODE 
framework = arduino 
lib_deps =  
SPI bblanchon/ArduinoJson@^6.21.4 esphome/AsyncTCP-esphome@^2.0.1
 ottowinter/ESPAsyncWebServer-esphome@^3.1.0 majicdesigns/MD_MAX72XX@^3.5.1 
majicdesigns/MD_Parola@^3.7.1 
makuna/NeoPixelBus@^2.7.7 wnatth3/WiFiManager@^2.0.16-rc.2 
bertmelis/espMqttClient@^1.5.0 
board_build.filesystem = littlefs 
;monitor_speed = 115200
;monitor_echo = yes`

those semi-colons are commenting out those lines.  Is that what you intended?

`void setup()
 { 
   Serial.begin(115200);
 /* while (!Serial) { // don't wait for serial on ESP32-S2, otherwise it will require the serial monitor during runtime yield(); }
 // wait for serial attach */ 
delay(5000); // wait until Platformio monitor is active ...etc...`

ErikBasalt, 2023-12-17:
The semicolons are indeed intentional, just to keep track of my previous experiments.

`while(!Serial){}`
doesn't work for my ESP32-S2 (it used to with my ESP8266), it will loop forever when connected to a USB power adapter (i/o PC). I see the same issue now when using CoolTerm in Disconnected mode.

Note that apart from this logging error during restart, my device is working as expected (matrix LED display, led strip, WiFi, web server, web sockets, etc).

Now I briefly tried 
`framework = arduino, espidf `
...but ended up with strange build and compilation errors.
Coming days I will be offline due to a business trip, will get back online again later this week. Meanwhile, any hint or help is very much appreciated.

peteDDD, 2023-12-17:
I am back to working on this and continue to have no joy... Only output to the serial port that I have working is Serial.print and only with framework= Arduino.  Still same USBSerial not found error if I use framework = arduino, espidf which I need for stuff like using the ets_sys.h in wake stubs.  But here is my test code as it stands

platformio
```
[env:lolin_s2_mini]
;platform = espressif32
platform = https://github.com/Jason2866/platform-espressif32.git
board = lolin_s2_mini
framework = arduino
build_flags = 
	-D ARDUINO_USB_MODE=0
	-D ARDUINO_USB_CDC_ON_BOOT=1
    -D CORE_DEBUG_LEVEL=4
monitor_speed = 115200
lib_ldf_mode = deep
```
ESP_LOF_FIX.h
```
#ifndef _ESP_LOG_FIX_H_
#define _ESP_LOG_FIX_H_

#if 1
#undef ESP_LOGE
#undef ESP_LOGI
#undef ESP_LOGD
#undef ESP_LOGW
#undef ESP_LOGV
#define ESP_LOGE(tag, format, ...) log_e(format, ##__VA_ARGS__)
#define ESP_LOGI(tag, format, ...) log_i(format, ##__VA_ARGS__)
#define ESP_LOGD(tag, format, ...) log_d(format, ##__VA_ARGS__)
#define ESP_LOGW(tag, format, ...) log_w(format, ##__VA_ARGS__)
#define ESP_LOGV(tag, format, ...) log_v(format, ##__VA_ARGS__)
#endif

#if 1
#undef printf
#define printf(format, ...) log_printf(format, ##__VA_ARGS__)
#endif

#endif // _ESP_LOG_FIX_H_
```

main.cpp
```
#include <Arduino.h>
// #include <ets_sys.h> // ets_sys.h is not available in Arduino framework without espidf

#include "ESP_LOG_FIX.h"

#define ARDUHAL_LOG_LEVEL_DEBUG

void setup()
{
   static const char *TAG = "setup";
   esp_log_level_set("*", ESP_LOG_NONE);
   esp_log_level_set(TAG, ESP_LOG_DEBUG);

   Serial.begin(115200);
   Serial.setDebugOutput(true); // needed on lolin-S2-mini https://github.com/espressif/arduino-esp32/issues/8977#issuecomment-1850998077
   while (!Serial)
   {
      delay(100);
   }
}

void loop()
{
   static const char *TAG = "loop";
   esp_log_level_set("*", ESP_LOG_NONE);
   esp_log_level_set(TAG, ESP_LOG_DEBUG);
   printf("Hello World\n"); // DOES NOT WORK
   // ets_printf("est_printf");  // ets_sys.h is not available in Arduino framework without espidf
   ESP_LOGE(TAG, "Hello from ESP_LOGE"); // DOES NOT WORK
   log_printf("log_printf says Hello"); // DOES NOT WORK
   Serial.println("Serial.println Hello World"); // works!  

   delay(1000);
}
```

terminal output:
```
Serial.println Hello World
Serial.println Hello World
Serial.println Hello World
Serial.println Hello World
Serial.println Hello World
Serial.println Hello World
Serial.println Hello World
...
```

ErikBasalt, 2023-12-20:
I understand that `framework = arduino, espidf` is giving difficulties at @peteDDD as well, so I'm back to `framework = arduino` again. Not getting any further, next lines do'n make any difference:
```
build_flags = 
	-D ARDUINO_USB_MODE=0
	-D ARDUINO_USB_CDC_ON_BOOT=1
```

The main 2 problems I still have are:
1) In VSC Terminal, when I add `while(!Serial){}` to my setup(), the code will start after a fresh upload from VSC/Platformio, but it does NOT run automatically after a restart. Then it waits until I press a keyboard button in the Terminal and sets up the USB serial connection again. Directly after pressing the button I see this logging:
```
Disconnected (ClearCommError failed (PermissionError(13, 'Het apparaat herkent de opdracht niet.', None, 22)))
Reconnecting to COM8     Connected!
```
                 (Translated: The device does not recognize the command)

2) With or without `while(!Serial){}`, with our without CoolTerm, I cannot see any diagnostic info when I reboot the ESP32-S2. I can only see my own logging before the restart (e.g. saying I will trigger ESP.restart(), or will perform a deliberate device-by-zero) and my own startup logging. In between no backtrace or diagnostic info at all that shows to reason for the restart.

I'm using a genuine [Wemos Lolin S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) and tried a second board with same result. I'm wondering: I a the only one facing these 2 problems?

ErikBasalt, 2023-12-22:
(also posted on [esp32.com](https://www.esp32.com/viewtopic.php?f=19&t=37299))

Today I tried logging on an ESP32-S3 board (Lilygo T7 S3), and here I don't have an issue. Straightforward platformio.ini and main.cpp. Here restarting is shown properly, when crashing intentionally I get backtrace info that can be decoded and leads to the faulty source line. Just what I would expect.

The electronic circuit of the USB connection of this S3 board is very similar to the S2 that I actually want to use.

=> Why doesn't logging work properly on S2?
Anyone?

**S3 board - platformio.ini**
```
[platformio]
;boards_dir =  boards

[env:lilygo-t7-s3]
platform = espressif32
board = lilygo-t7-s3
framework = arduino

build_flags = 
	-D ARDUINO_USB_CDC_ON_BOOT=1

monitor_speed = 115200
monitor_filters = esp32_exception_decode
```

**S3-board - main.cpp**
```
#include <Arduino.h>

void setup()
{
    Serial.begin(115200);
    // Serial.setDebugOutput(true);
    while (!Serial)
    {
        yield();
    }
    // delay(5000);
    Serial.println("Setup done, start loop()");
}

void loop()
{
    static int i = 0;
    Serial.println(i++);
    delay(1000);
    if (Serial.available())
    {
        switch (Serial.read())
        {
        case '*':
            Serial.println("Do restart...");
            Serial.flush();
            ESP.restart();
            break;
        case '!':
        {
            Serial.println("Do crash...");
            Serial.flush();
            char *cptr = nullptr;
            *cptr = '!';
        }
        break;
        default:
            // ignore
            break;
        }
    }
}
```

**S3 board - lilygo-t7-s3.json** (source: https://github.com/Xinyuan-LilyGO/T7-S3)
```
{
    "build": {
        "arduino":{
            "ldscript": "esp32s3_out.ld",
            "memory_type": "qio_opi",
            "partitions": "default_16MB.csv"
        },
        "core": "esp32",
        "extra_flags": [
            "-DBOARD_HAS_PSRAM",
            "-DARDUINO_USB_MODE=1"
        ],
        "f_cpu": "240000000L",
        "f_flash": "80000000L",
        "flash_mode": "qio",
        "hwids": [
            [
                "0X303A",
                "0x1001"
            ]
        ],
        "mcu": "esp32s3",
        "variant": "esp32s3"
    },
    "connectivity": [
        "wifi",
        "bluetooth"
    ],
    "debug": {
        "openocd_target": "esp32s3.cfg"
    },
    "frameworks": [
        "arduino",
        "espidf"
    ],
    "name": "LILYGO T3-S3",
    "upload": {
        "flash_size": "16MB",
        "maximum_ram_size": 327680,
        "maximum_size": 16777216,
        "require_upload_port": true,
        "speed": 921600
    },
    "url": "https://www.aliexpress.us/item/3256804591247074.html",
    "vendor": "LILYGO"
}

```

peteDDD, 2023-12-23:
Erik,
I have no problems with any form of printing or logging on any other board except for the Lolin_S@_Mini.
Pete

ErikBasalt, 2023-12-23:
Same here. For now I'm giving up. The logging is limited but the Lolin_S2_mini board is doing it's job nicely and I do need its small form factor.