espressif/arduino-esp32 #8978: Board support for ESP32-BOX-3

AarooneyFarr, 2023-12-09:
### Related area

Board Support

### Hardware specification

Support for ESP32-BOX-3

### Is your feature request related to a problem?

No

### Describe the solution you'd like

I would like board support for ESP-BOX-3

### Describe alternatives you've considered

ESP-BOX doesn't seem to be working for my ESP-BOX-3

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-12-12:
@AarooneyFarr - There are a few differences from BOX3 to BOX:

- It has no Accelerometer
- GPIO47 is the Backlight, while BOX uses GPIO45
- GPIO45 is the I2S LCLK, while BOX uses GPIO47 
- 2 options for Display + Touch Screen => ILI9341+GT911 or ST7789+TT21100
- BOX3 has SD_MMC + SD Card Slot

Do you already have any project in a public repository?

VojtechBartoska, 2023-12-13:
Hello, this will be covered by BSP (Board support package) from @me-no-dev, adding to 3.1.0 milestone. The tasks is now in the progress but is blocked by 3.0.0 release.

SuGlider, 2023-12-27:
@AarooneyFarr - I have just added a Adafruit GFX based driver for the ESP-BOX-3 that you can test and use.
In order to use it, it is necessary to select the ESP32-S3-Box board and it will change the `TFT_BL` pin to 47, instead of 45 when `ESP32_S3_Box_3_TFT.h` is included.

If necessary, `I2S_LRCK` pin has to be changed to 45, instead of 47.

The example in the repository runs the Adafruit GFX demo and also prints the touch point in the Serial Monitor. It also allows "finger painting" (using touch) at any rotation mode, at the end of the GFX demo.

https://github.com/SuGlider/Adafruit_ESP32S3_BOX_3/

Hvae fun!

SuGlider, 2023-12-27:
I also just realised that there is an effect with `Serial.print()` in the ESP-BOX-3 when the GFX demo sketch is run with no open `Serial Monitor`. This is a sort of delay in every `Serial.print()` from loop, related to some issue with the USB CDC driver.

In order to avoid it, it is necessary to disable `Serial`, by commenting out the `Serial` initialization:

``` cpp
void setup() {
  //Serial.begin(115200);
  //Serial.setDebugOutput(true);
  //while (!Serial) delay(100);

```

weoiss1998, 2023-12-27:
Is it affecting the ESP32 too?

SuGlider, 2023-12-27:
> Is it affecting the ESP32 too?

No. Only ESP32-S3 with OTG CDC acting as `Serial`.
ESP32 has no USB port, just UART + UART-USB-CDC converter chip.
I got this issue using Arduino Core 2.0.13. Therefore, I'm not sure if this also happens with 3.0.0+

SuGlider, 2023-12-28:
@weoiss1998 - it only affects the S3, C3 and C6 when using Hardware CDC and JTAG as `Serial`. 
This demands the IDE to be set with `USB Mode: "Hardware CDC and JTAG"` and `USB CDC On Boot: "Enabled"`.

The workaround is to set the USB Writing Timeout to Zero, like this:

``` cpp
  Serial.begin(); // USB HW CDC doesn't need any baudrate
  Serial.setDebugOutput(true);  // sends all log_e(), log_i() messages to USB HW CDC
  Serial.setTxTimeoutMs(0);       // sets no timeout when trying to write to USB HW CDC
```

me-no-dev, 2023-12-28:
@SuGlider adding the SOF check might help reduce that time if USB is not plugged into a Host

SuGlider, 2023-12-29:
The issue is not if it is plugged to the Host, but if the serial monitor (or any other, like Putty) has opened a connection.
The S3-Box-3 is **always** connected to my Win11 PC, (host) and the COM port is always enumerated, but the issue is related to start or not the `Serial Monitor` (or `Putty` or any other).

Other potential problem is that whenever I close the `Serial Monitor`, the board is automaticaly reset.
There is a code line in  `setup()` like this: `while(!Serial) delay(100);` 
It never does anything when the board is first time turned on or when it is manually reset.

But, if I open the `serial monitor`, close it and, because of it, the board is reset, **that line of code takes effect**!
The sketch will only start after I open the `serial monitor`... and then I can close/open it and that line of code will always work, holding the sketch execution.

Something is not being done in the right order of events... This is not the expected behaviour.

SuGlider, 2023-12-29:
It seems that it has more to do with a IRQ related to the CDC USB End Point than to SOF.... 
At least it seems to be so.... I need to investigate it further.

me-no-dev, 2023-12-29:
if board gets reset, that means that the host is toggling the RTS/DTS pins on disconnect. Maybe Windows related?

SuGlider, 2023-12-29:
This is USB only, there are no RTS/DTS pins. send it Can it send RTS/DTS signal through CDC-ACM?
The board reset with this boot message: `rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)`
Therefore, the board recognizes that USB is the last reset cause.

closed 2026-04-17 by me-no-dev