espressif/arduino-esp32 #8979: BLE Provisioning crash with Bluetooth

garudaonekh, 2023-12-10:
### Board

ESP32-DEV

### Device Description

Hi,
This code work in espidf
https://github.com/espressif/esp-idf/tree/release/v4.4/examples/provisioning/wifi_prov_mgr

I try to get this work in espidf, arduino

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

MacOS

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

It crashed at this line

  `      ESP_ERROR_CHECK(wifi_prov_mgr_start_provisioning(security, pop, service_name, service_key));`

Based on my research, it maybe due to starting BLE from Arduino is different. 

### Sketch

```cpp
I add the app_main() function inside the setup()function
```

### Debug Message

```plain
D (2530) esp_netif_lwip: call api in lwip: ret=0x0, give sem␛[0m
D (ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40092cd8
file: "IDF/components/protocomm/src/transports/protocomm_nimble.c" line 486
func: simple_ble_start
expression: esp_nimble_hci_and_controller_init()

abort() was called at PC 0x40092cdb on core 1

---
DEBUG Stacktrace

0x40083022: panic_abort at /Users/x/.platformio/packages/framework-espidf/components/esp_system/panic.c line 408
0x40092ce5: esp_system_abort at /Users/x/.platformio/packages/framework-espidf/components/esp_system/esp_system.c line 137
0x4009999e: abort at /Users/x/.platformio/packages/framework-espidf/components/newlib/abort.c line 46
0x40092cdb: _esp_error_check_failed at /Users/x/.platformio/packages/framework-espidf/components/esp_system/esp_err.c line 43
0x4014b6ef: simple_ble_start at /Users/x/.platformio/packages/framework-espidf/components/protocomm/src/transports/protocomm_nimble.c line 486
0x4014bada: protocomm_ble_start at /Users/x/.platformio/packages/framework-espidf/components/protocomm/src/transports/protocomm_nimble.c line 945
0x400d5e95: prov_start at /Users/x/.platformio/packages/framework-espidf/components/wifi_provisioning/src/scheme_ble.c line 50
0x400d4521: wifi_prov_mgr_start_service at /Users/x/.platformio/packages/framework-espidf/components/wifi_provisioning/src/manager.c line 287
0x400d5806: wifi_prov_mgr_start_provisioning at /Users/x/.platformio/packages/framework-espidf/components/wifi_provisioning/src/manager.c line 1522
0x400d2fb5: app_main() at src/main.cpp line 310
0x400d301f: setup() at src/main.cpp line 351
0x400d828e: loopTask(void*) at /Users/x/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 42
0x40096005: vPortTaskWrapper at /Users/x/.platformio/packages/framework-espidf/components/freertos/port/xtensa/port.c line 142
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-12-10:
@garudaonekh - The Arduino example for WiFi provisioning with BLE works fine.
You can find it at https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino

garudaonekh, 2023-12-11:
> @garudaonekh - The Arduino example for WiFi provisioning with BLE works fine. You can find it at https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino

Thanks, sketch works in Arduino and the size sketch size is 1.5MB which is not feasible for my project as I need OTA and other log data.  And it works fine with ESPTOUCH BLE, however, it doesn't work with Flutter BLE(https://github.com/ogabrielinacio/esp_provisioning_ble), failed at the submission of wifi password stage.

Most importantly, I can't get it to work in my project which is Platformio(ESPIDF, Arduino). I try to set bluetooth mode to BR/EDR Only in menuconfig but still failed
`beginProvision(): wifi_prov_mgr_start_provisioning failed!`

I go another way:
I can get this legacy version to work with Security0 https://github.com/espressif/esp-idf/tree/release/v4.4/examples/provisioning/legacy/ble_prov

But with Security1, it failed to accept POP PIN from Mobile App.

me-no-dev, 2023-12-11:
> I add the `app_main()` function inside the `setup()` function

This is not correct. app_main is the function that get's called by the bootloader and then in that function the setup/loop task is created. Also BR/EDRis Classic BT, not BLE, so hgow would BLE work if you have turned it off?

Your best approach would be to make sure that your arduino-as-component project is properly defined and that you have used/started with sdkconfig that was used to build the IDF Libs. You probably have some option missing or something else is not OK.

latest sdkconfig for ESP32 and ESP-IDF v5.1.2 is [here](https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32/sdkconfig)

SuGlider, 2023-12-12:
@garudaonekh - A few questions:

1- What is your platformio.ini file? Please show it.
2- Do you have the `src/Kconfig.projbuild` correctly set? This may be the issue with `beginProvision(): wifi_prov_mgr_start_provisioning failed!`
3- Do you have a `/sdkconfig.defaults` correctly set?

SuGlider, 2023-12-12:
@garudaonekh - Is there any repository with the code platformio project that I could test?

SuGlider, 2023-12-12:
I have this PIO project that uses IDF and Arduino based on the code from https://github.com/espressif/esp-idf/tree/release/v4.4/examples/provisioning/wifi_prov_mgr. 
I builds and runs fine.
[WiFi_Prov_with_BLE_IDF.zip](https://github.com/espressif/arduino-esp32/files/13652086/WiFi_Prov_with_BLE_IDF.zip)

garudaonekh, 2023-12-13:
> @garudaonekh - The Arduino example for WiFi provisioning with BLE works fine. You can find it at https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino

I see some issues with this example which make it unstable.

1-Wifi list is empty. Still trying to figure out the solution.
https://github.com/espressif/arduino-esp32/issues/8305

2-Bluetooth release issue which I have to remove BLE in the esp32-hal-misc.c file 
  ```
 // if(!btInUse()){
     //   esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
   // }
```

garudaonekh, 2023-12-13:
> I have this PIO project that uses IDF and Arduino based on the code from https://github.com/espressif/esp-idf/tree/release/v4.4/examples/provisioning/wifi_prov_mgr.
> 
> It builds and runs fine. [WiFi_Prov_with_BLE_IDF.zip](https://github.com/espressif/arduino-esp32/files/13652086/WiFi_Prov_with_BLE_IDF.zip)

Thanks. This so far is the most stable. I am still trying to make my it work with my Flutter iOS version.
I will keep this issue on for a few more days.

Phetdanai, 2024-01-06:
shoot the touble now!!

VojtechBartoska, 2024-02-12:
@garudaonekh is this still valid? Thanks

closed 2024-02-17 by garudaonekh