espressif/arduino-esp32 #8988: ESP_SR Basic.ino sketch doesn't work.

RobertMorton56, 2023-12-12:
### Board

ESP32-S3-BOX-Lite

### Device Description

ESP32-S3-BOX-Lite stock, with no tweaks

### Hardware Configuration

As per ESP32-S3-BOX-Lite defaults. No other devices attached apart from UART for debugging. UART connected to G43 and G44 as per defaults.

### Version

latest master (checkout manually)

### IDE Name

arduino-cli

### Operating System

linux LTS kernel

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Everything compiles, and uploads fine. When viewing the debug serial output I keep getting:

``` E (73) AFE_SR: ERROR: Please select wake words! ```

I assumed the model would be uploaded to SPIFFS etc with this example. Is this not the case? My goal is to just get the Basic.ino sketch running.

Thanks

### Sketch

```cpp
As it is in master.
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379dc2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
E (73) AFE_SR: ERROR: Please select wake words!

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-12-12:
@me-no-dev PTAL. Thanks.

me-no-dev, 2023-12-12:
First please enable core debug from the board menu to level "debug" or "verbose". It will print a lot of things.
Then make sure that you have selected the appropriate partition scheme, else model would not be downloaded.

Also, even when you get all that to compile fine, the box-lite also requires yo to properly setup and init the audio DAC and ADC controllers. We will have all that covered soonish through a BSP Library, but we are not there yet.

RobertMorton56, 2023-12-12:
Ah OK. Got you. Is there an recommended partitions.csv I can start with?

Rather than waste people's time maybe I'll wait for the library release? Soonish like a month? Longer?

I reckon there's not much left to get this sketch up and running on the box-lite with the current code as is. Thanks for the reply, looking forward to trying this hardware out with this core.

me-no-dev, 2023-12-13:
Select the ESP32-S3 dev module, enable OPI PSRAM and select [ESP SR 16M (3MB APP/7MB SPIFFS/2.9MB MODEL)](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/esp_sr_16.csv) partition from the menu. This will set you up to get the proper partitions and model. Let me know if that does not work

RobertMorton56, 2023-12-13:
Yep, that partition scheme worked - however I stuck with the esp32s3box. The ESP32-S3 dev module just loops these logs:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdad5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
```

Compiling for the esp32s3box gives:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
[   156][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 1
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   386740 B ( 377.7 KB)
  Free Bytes        :   357944 B ( 349.6 KB)
  Allocated Bytes   :    23908 B (  23.3 KB)
  Minimum Free Bytes:   352896 B ( 344.6 KB)
  Largest Free Block:   319476 B ( 312.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8386296 B (8189.7 KB)
  Allocated Bytes   :        0 B (   0.0 KB)
  Minimum Free Bytes:  8386296 B (8189.7 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
[... 63 lines cut ...]
  --------------------------------------
     2 : I2S_STD_MCLK
    15 : I2S_STD_DIN
    16 : I2S_STD_DOUT
    17 : I2S_STD_BCLK
    19 : USB_DM
    20 : USB_DP
    40 : GPIO
    41 : GPIO
    43 : UART_TX[0]
    44 : UART_RX[0]
    47 : I2S_STD_WS
============ After Setup End =============
```

It doesn't respond to voice so I imagine the audio DAC and ADC controllers still need to be initialized as you suggested?

To note, I'm using arduino-cli so I used the following command to compile:

```
arduino-cli compile -v --warnings all --fqbn esp32:esp32:esp32s3box:PartitionScheme=esp_sr_16,DebugLevel=debug
```

I got the pin definitions from the schematic:
[image]

me-no-dev, 2023-12-13:
Yes. all started fine, but you do need to init the DAC and ADC. We will soon have support for all that.

@lucasssvaz if you get it going, please post the drivers here, so that @RobertMorton56 can continue testing :)

lucasssvaz, 2023-12-22:
I'm implementing some I2S examples in #9030. Maybe it will help

RobertMorton56, 2023-12-22:
Great, thanks for this. From a brief look at your example, the DAC is used. I assume I'll need to init the ADC too if I want to use the ESP32-box-lite's microphone?

Thanks again

lucasssvaz, 2023-12-23:
Yeah, you will need to configure the [ES7243E ADC](https://www.pawpaw.cn/media/documents/2022-04/ES7243E_DS_pawpaw%E6%9C%A8%E7%93%9C%E7%A7%91%E6%8A%80.pdf) to convert the analog signal from the Mic to I2S and, if you want to use the speaker, you'll need to also configure the [ES8156 DAC](http://www.everest-semi.com/pdf/ES8156%20PB.pdf) (don't forget to enable the PA chip by driving the GPIO 46 high).

RobertMorton56, 2023-12-24:
Great, I'll try that. Thanks!

RobertMorton56, 2023-12-28:
Tried that, it seems like the esp-adf is required. I did try and bring some of it into the Arduino space but it's getting messy. Might be better to wait till you guys have v3 ready. I assume you'll have es7234 and es8156 included in that release?

lucasssvaz, 2024-01-08:
@RobertMorton56 What I meant is using the ES8388 driver as a base and getting the registers addresses and values from the ADF. You won't be able to use the exact commands that are used there but the codec registers and values will be the same.

VojtechBartoska, 2024-01-31:
Hello @RobertMorton56, is this issue still valid or we can close it?

VojtechBartoska, 2024-02-12:
Closing, if needed you can reopen.

closed 2024-02-12 by VojtechBartoska

KoStard, 2025-12-04:
Useful conversation, thank you!