espressif/arduino-esp32 #8990: Can't find default sdkconfig

DavidSchinazi, 2023-12-13:
### Board

ESP32

### Device Description

m5stack ATOM Matrix (pico32)

### Hardware Configuration

m5stack ATOM Matrix

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

macOS 14.1.2

### Flash frequency

160MHz

### PSRAM enabled

yes

### Upload speed

1500000

### Description

Apologies for the strange question. My setup is that I build my code using PlatformIO with both `espidf` and `arduino` frameworks set. This allows me to compile my own ESP-IDF with a custom `sdkconfig` file. I used to periodically pull the default sdkconfig from [this link](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig). Unfortunately, that file was removed by commit 6f7a1ca76ada4d705d79ce863450c91349327a79, the latest version before removal was [here](https://github.com/espressif/arduino-esp32/blob/082a61ebe476384941bf4e38ce0363f928adba12/tools/sdk/esp32/sdkconfig). I searched in the repository and I can't find it. Do you no longer keep a copy of the default config for ESP32? If not, is there a way for me to generate that myself?

Here's the relevant parts of my `platformio.ini`:
```
[env:myenv]
framework = espidf, arduino
platform = espressif32
board = pico32
```
Thank you!

### Sketch

```cpp
NA
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-12-13:
Hi, sdk is still provided and it is renamed to `esp32-arduino-libs`.

You can also build your own libraries using lib-builder, take a look on this [docs page](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html?highlight=lib%20builder).

Jason2866, 2023-12-13:
In short you can't use actual Arduino branch master with official Platformio espressif32.
There is no setup, nor support from Platformio for the development alpha 3.0.0 core.
As @VojtechBartoska wrote you have to change Platformio to use the builded `esp32-arduino-libs`.
I have done a Platformio setup (modified fork) for project Tasmota with changes optimized for Tasmota with Core 3.0.0-alpha3. You can use with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.12.10/platform-espressif32.zip
```
The big difference is the removed SPIFFS and mbedtls support.

DavidSchinazi, 2023-12-13:
Thank you both! I was able to create the sdkconfig from tip-of-tree by running the following:
```
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh esp32
. ./export.sh
cd
git clone https://github.com/espressif/esp32-arduino-lib-builder
cd esp32-arduino-lib-builder
./build.sh -t esp32 -b idf_libs
less out/tools/esp32-arduino-libs/esp32/sdkconfig
```

And here are the commands for an older version like 4.4:

```
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout --recurse-submodules --force release/v4.4
./install.sh esp32
. ./export.sh
cd
git clone https://github.com/espressif/esp32-arduino-lib-builder
cd esp32-arduino-lib-builder
git checkout --recurse-submodules --force release/v4.4
./build.sh -t esp32 -b idf_libs -A idf-release/v4.4 -I release/v4.4
less out/tools/sdk/esp32/sdkconfig
```

Note that on macOS, this requires installing gawk from brew and putting it on PATH.
```
brew install gsed gawk jq
export PATH="/opt/homebrew/opt/gawk/libexec/gnubin:$PATH"
```

To the maintainers, feel free to close this issue - might be worth documenting these steps somewhere in case it helps others though.

VojtechBartoska, 2023-12-14:
Great, closing as solved.

closed 2023-12-14 by VojtechBartoska