espressif/arduino-esp32 #8992: SDMMC issue "send_op_cond (1) returned 0x107"

handmade0octopus, 2023-12-13:
### Board

Custom board

### Device Description

Custom board with 10k pullups for the channels. Pins are ok, some traffic happens on MISO and MOSI:
[image]
[image]

### Hardware Configuration

	#define SD_SCK 		39
	#define SD_MISO		38
	#define SD_MOSI		40

### Version

v2.0.14

### IDE Name

Platformio + VCS

### Operating System

W11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Just got new board (nearly same layout and components as my trusted ESP32 based board) with ESP32-S3-WROOM-1(N8R2) module with PSRAM. Only found one person with this problem online but was unrelated so I am asking here.

Here is my board config:
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "boards/partitions-4MB-tinieruf2.csv"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_USB_CDC_ON_BOOT=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "adafruit_feather_esp32s3"
  },
  "connectivity": [
    "wifi",
    "can"
  ],
  "debug": {
    "openocd_board": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Gauge.S V5 Module",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "http://gauge.sorek.uk/",
  "vendor": "sorek.uk"
}
```

And here is my custom partition layout (I skipped "factory" partition - not sure what it was for):
```
# ESP-IDF Partition Table,,,,,
# Name,   Type, SubType, Offset,  Size, Flags
# bootloader.bin,,          0x1000, 32K,,
# partition table,          0x8000, 4K,,,
nvs,      data, nvs,      0x9000,  20K,
otadata,  data, ota,      0xe000,  8K,
ota_0,    app,  ota_0,   0x10000, 0x1E0000,
ota_1,    app,  ota_1,   0x1F0000,0x1E0000,
spiffs,   data,spiffs,0x3F0000,0x10000,
```

### Sketch

```cpp
#define SD_SCK 		39
	#define SD_MISO		38
	#define SD_MOSI		40
(...)

SD_USE.setPins(SD_SCK, SD_MISO, SD_MOSI, -1, -1, -1);

mainConfig.sdReady = SD_USE.begin("/sdcard", true, false, BOARD_MAX_SDMMC_FREQ, 5);
if(!mainConfig.sdReady) {
  mainConfig.sdReady = SD_USE.begin("/sdcard", true, false, SDMMC_FREQ_DEFAULT, 5);
}
```

### Debug Message

```plain
E (287) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (287) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   621][E][SD_MMC.cpp:148] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
E (327) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (327) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   660][E][SD_MMC.cpp:148] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
```

### Other Steps to Reproduce

I tried few different arduino versions but didn't help.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-12-13:
@P-R-O-C-H-Y  PTAL

@handmade0octopus and you are positive that you have connected all pins properly? You show wiring for one-lane SDMMC, but used SPI naming. Also you should not have pull-up on the CLK line. Just on CMD and D0

handmade0octopus, 2023-12-13:
> @P-R-O-C-H-Y PTAL
> 
> @handmade0octopus and you are positive that you have connected all pins properly? You show wiring for one-lane SDMMC, but used SPI naming. Also you should not have pull-up on the CLK line. Just on CMD and D0

I wish there was a button "closed - OP is idiot".

Thank you, of course I swapped MISO and MOSI as I was 99% sure I am right but ESP32 doesn't let you chose which pins are which.

closed 2023-12-13 by handmade0octopus