espressif/arduino-esp32 #8996: Serial missing bytes

zekageri, 2023-12-14:
### Board

ESP32-Wrover

### Device Description

Nothing fancy, just an ESP32-Wrover with 4mb psram and 16mb flash

### Hardware Configuration

Nothing fancy, just an ESP32-Wrover with 4mb psram and 16mb flash

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I wrote a modbus application using my ESP32 as a master. Sometimes the response from the slave devices are cut off at the start.
I suspect that it is because `Serial1.flush();` is taking a long time and the esp excutes other tasks. When it comes back to pull the EN pin to LOW it is too late.

When this happens always the first bytes are cut off. That's why i think that the esp is too slow to process the data which coming back. ( the salve device is specifically made to answer back as soon as possible to speed up the bus )

I need some kind of a TX interrupt so i can set the `MBUS_EN` pin to LOW as soon as the uart sent the last byte from the TX buffer.

### Sketch

```cpp
#define MBUS_BAUD 115200
#define MBUS_RX 34
#define MBUS_TX 15
#define MBUS_EN 5

void setup(){
    Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
    pinMode(MBUS_EN, OUTPUT);
}

void Modbus::rawWrite(uint8_t * data, int length){
    int crc16 = getCRC_16(data, length);
    uint8_t *buffer;
    if( !alloc(buffer,length+2) ){ return; } // Just allocate some space in psram

    memcpy(buffer, data, length);
    buffer[length]   = lowByte(crc16);
    buffer[length+1] = highByte(crc16);
    
    digitalWrite(MBUS_EN, HIGH);
    Serial1.write(buffer, length+2);
    Serial1.flush(); // Here is the problem. Probably the RTOS is jumping to other tasks and does not come back immidiately.
    digitalWrite(MBUS_EN, LOW); // So MBUS_EN does not pulled to low immidiately.

    free(buffer);
}
```

### Debug Message

```plain
Missing my packet first 4 byte. Sometimes its 3 byte sometimes its 6.
Expecting 0x0A as address but got back 0x15 which is totally wrong.

[45](Modbus) - Invalid raw response from address: 0x0A. Got response: 0x15 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x39 0xC2
(Modbus) - Wrong address
```

### Other Steps to Reproduce

I need a way to attach an interrupt to UART TX pin so i can set my MBUS_EN pin to LOW immidiately when the data is sent!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-12-14:
@SuGlider Please help with triage, when you have some spare time. Thanks

zekageri, 2023-12-15:
Okay so i found out about the `MODE_RS485_HALF_DUPLEX`.
I can set my UART to RS485 hw controll mode so the RTS PIN is toggled by the hardware.
This is working and i can't see invalid messages now but i don't see any examples on how to adjust the timing of the pin.
It is much slower compared to my software solution.

```cpp
Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
Serial1.setPins(-1, -1, -1, MBUS_EN);
//Serial1.setHwFlowCtrlMode(MODE_RS485_HALF_DUPLEX, 10); // If i uncomment this line my bus just froze.
Serial1.setMode(MODE_RS485_HALF_DUPLEX);
```

rawWrite

```cpp
//digitalWrite(MBUS_EN, HIGH);
Serial1.write(buffer, length+2);
Serial1.flush();
//digitalWrite(MBUS_EN, LOW);
```

Right now i don't even initialise the RTS pin. It just works with HW FLOW CONTROLL but it is slower. Much slower.
How to set the timing? ( trashold )

closed 2023-12-15 by zekageri