espressif/arduino-esp32 #9004: ESP32-S3 hangs on Serial if no Serial Monitor is connected

mvermand22, 2023-12-15:
### Board

ESP32-S3

### Device Description

wt32-sc01-plus 3.5 inch display board with Wireless-Tag WT32-S3-WROVER chip

### Hardware Configuration

no

### Version

v2.0.11

### IDE Name

Arduino IDE 2.2.1

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I think the issue will happen on any esp32-S3:
When you use Serial.begin(...) & Serial.println(...) in a scetch, then the board will hang if no serial monitor is opened.
See discussion: https://github.com/espressif/esp-idf/issues/10324#issuecomment-1857488299
The arduino-esp32 in Arduino IDE 2.x should be upgraded to v5.1.2

### Sketch

```cpp
void setup() {
  Serial.begin(9600);
  Serial.println("Hello World");
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```

### Debug Message

```plain
NA, since it only happens when not connected to Arduino-IDE
```

### Other Steps to Reproduce

Open the Serial Monitor, reboot the board.
The led will blink.
Close the Serial Monitor, reboot the board.
The led will not blink.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-12-15:
The upcoming core 3.0.0 has the change included since it is based on IDF 5.1
Core 2.0. is based on IDF 4.4.

mvermand22, 2023-12-15:
Any idea when this will be available in Arduino-IDE 2.x?

Jason2866, 2023-12-15:
Arduino IDE has nothing to do which Arduino core version is used.

mvermand22, 2023-12-15:
Ok so when this core 3.0 version is released, I Will be able to download it in Arduino IDE?

lbernstone, 2023-12-15:
You can download the development version today in Arduino IDE.
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide

SuGlider, 2023-12-18:
@mvermand22 - ESP32-S3 has two modes for USB CDC (it seems to be the Serial interface for this board - Not UART).
By default, it uses USB HW CDC/JTAG mode. But you can select the USB OTG mode as well.
It can be selected in the Arduino IDE menu of the board.

Maybe this work around can solve this issue. 
You can also try Arduino Core Alpha-3 based on IDF 5.1 - not sure if this IDF branch fixes it.

pat1, 2023-12-18:
I have the same problem with ESP32-C3 with platformio.
compiling with:
```
  -D ARDUINO_USB_MODE=1    -D ARDUINO_USB_CDC_ON_BOOT=1 
```
and
```
Serial.setTxTimeoutMs(0);
```
do not help.
Only comment Serial.begin make the firmware running.

Arduino Core Alpha-3 based on IDF 5.1 do not work for me:
```
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32

```
return:
```
Converting stimawifi.ino
KeyError
```

mvermand22, 2023-12-18:
@SuGlider thanks. I'm a bit scared of using that option, as my bord has no boot-button, only a reset button.
I found an article (don't know which anymore) that stated that when using USB-OTG mode, you need to use both Boot and Reset buttons to get the board in flash mode.

closed 2024-02-28 by me-no-dev

SuGlider, 2024-02-28:
@mvermand22 @pat1 @Jason2866 @lbernstone 

The original issue is related to the JTAG/HW CDC Serial peripheral from ESP32-S3, ESP32-C3, ESP32-C6 and ESP32-H2.

One issue is that when USB cable is unplugged, `flush()` will block and `write()` will also block as soon as the HWCDC Ringbuffer gets full.

The second issue is that It won't work properly when the sketch waits for the CDC to be connected using a Serial Monitor, using a code like `while(!Serial) delay(100);` 

These and some other issues were fixed by the PR #9275 

When USB is unplugged, nothing will block any HW CDC writing or flushing.
When the buffer gets full, its content will be discarted and it will show the latest written data.
By this way, when the sketch connects to a serial monitor, it will display the latest data flow.

`bool HWCDC` will return true only when CDC is connected using some Serial Monitor Application.
It is possible plug, unplug, open and close the terminal that it will work correctly.
Closing the terminal while using Windows 10/11 will reset the board because for some reason Windows raises USB ACM RTS that makes the board consider that it shall reset. But it is possible to unplug the cable and the software will detect it and adjust its status to indicate that the CDC connection has been terminated. It is possbile to  plug back the USB cable and reconnect automatically by opening the serial monitor again.

It has been fixed for arduino core 3.0.0-RC1 and it is available in the master branch.

Check the PR examples.

pat1, 2024-02-28:
Thanks,
for me is not so simple to check it with platformio.

Jason2866, 2024-02-28:
@pat1  Did a Platformio build from latest master. You can test with this setup
```
platform = https://github.com/Jason2866/platform-espressif32.git
```

SuGlider, 2024-02-29:
@Jason2866 - I found a last potential delay issue with HW CDC not connected... PR #9307 shall be also mixed in a day or so.
One it is merged, please add that one to your PIO build.

pat1, 2024-02-29:
@Jason2866 work for me in Linux and ESP32-C3