espressif/arduino-esp32 #9007: WebServer::authenticate uses wrong braces in malloc

TD-er, 2023-12-15:
### Board

Any

### Device Description

---

### Hardware Configuration

---

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

```
C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp:138:54: warning: pointer to a function used in arithmetic [-Wpointer-arith]
  138 |       char *toencode = (char *)malloc[toencodeLen + 1];
      |                                                      ^
cc1plus.exe: warning: pointer to a function used in arithmetic [-Wpointer-arith]
```

Took a while to see what was wrong here, but the code uses `malloc[]` instead of `malloc()`

### Sketch

```cpp
---
```

### Debug Message

```plain
---
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2023-12-17:
Yep, totally agree that we should move away from doing the low-level memory allocation ourselves.

I always try to write code which has "no warnings", so I have all warnings enabled.
To be honest, I had never seen this warning, so had to look through the code several times to see what was actually wrong here.

SuGlider, 2023-12-18:
Thanks @TD-er !! Nice catch for sure!

PR #9012 (2023-12-18): Fixes Malloc call in WebServer code
## Description of Change
This fixes a wrong (but "valid") call for `malloc()`.

C:/Users/user/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.cpp:138:54: warning: pointer to a function used in arithmetic [-Wpointer-arith]
  138 |       char *toencode = (char *)malloc[toencodeLen + 1];
      |                                                      ^
cc1plus.exe: warning: pointer to a function used in arithmetic [-Wpointer-arith]

## Tests scenarios
Just CI

## Related links
Fixes #9007
closed 2023-12-18 by me-no-dev