espressif/arduino-esp32 #9010: adc oneshot read fail

josef2600, 2023-12-17:
### Board

ESP32 DEV

### Device Description

nothing is attached. only put 100nf caps on ADC pins.

### Hardware Configuration

analog pins, gpio 32-33-34-35

### Version

latest master (checkout manually)

### IDE Name

arduino

### Operating System

win 10

### Flash frequency

80 mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

since in board version 3.0 you have removed any checking attach pins or custom settings, i only have this in setup:
```
  analogReadResolution(12); //12 for all chips except ESP32S3 where default is 13 bits (range from 0 to 8192)
  analogSetAttenuation(ADC_11db); //for C3 => ADC_11db: 2770 mv - ADC_6db: 1509 mv - ADC_2_5db: 1083 mv - ADC_0db: 800 mv

```
i have created a task that only reads multi analog pins:

```
        my_static_readadc_cX += analogReadMilliVolts(32);
        my_static_readadc_cX += analogReadMilliVolts(33);
        my_static_readadc_cX += analogReadMilliVolts(34);
        my_static_readadc_cX += analogReadMilliVolts(35);
```
but i always get these warnings:

```
E (38271) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 37257][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (39327) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 38312][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (40381) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 39366][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (42487) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 41472][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (43541) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 42526][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (44592) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 43577][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (45646) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 44631][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!

```
in version 2.0.14 this code works just fine, but in version 3 it gets me these warnings and reads bad values. the real value is 0, but when i get these, the reading values become 250 or 300 or like that instead of 0. there are zero volt on them.
i am using debug (info). 
what is going on? why did you change everything so poorly?

### Sketch

```cpp
----
```

### Debug Message

```plain
E (38271) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 37257][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (39327) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 38312][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (40381) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 39366][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (42487) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 41472][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (43541) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 42526][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (44592) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 43577][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
E (45646) adc_oneshot: adc_oneshot_get_calibrated_result(250): adc oneshot read fail
[ 44631][E][esp32-hal-adc.c:333] __analogReadMilliVolts(): adc_oneshot_get_calibrated_result failed!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-12-18:
Hi @josef2600, can you please post whole Debug output from serial? In 3.0.0 we added "Before setup" and "After Setup" prints for debugging purposes.

josef2600, 2023-12-18:
thank you. i will.

josef2600, 2023-12-19:
sorry for the delay. my project is very large and investigating anything take a long time. the ADC reading is in a task.
anyway, i tested few things:
if i use my ADC code in main loop, it works. then, i thought to change few settings for task creating. and i got it!
if i do this:
`xTaskCreatePinnedToCore(analog_reading_all_channels_event_task, "Task_analog_reading", 4000, NULL, 2 , NULL,0);`
everything works. but if i change the core to 1:
`xTaskCreatePinnedToCore(analog_reading_all_channels_event_task, "Task_analog_reading", 4000, NULL, 2 , NULL,1);`
then i get errors for  adc oneshot read fail and so on. my Arduino has these settings:
Arduino runs on core 0
events run on core 1
my tasks run on core 0
my analog reading task run on core 1.
why is that? why analog readings don't work on core 1 ?

josef2600, 2023-12-19:
one thing i forgot, in board manager 2.0.14 my Task_analog_reading stack memory was 2000, and it was working great. but since  board manager v3.0.0 , it was panicking and was resetting. after i changed my stack size to 4000, it worked. 

update:
even when task is running on core 0, i still get `adc oneshot read fail` sometimes too. not many, but i get them. specially in start and creations of analog pins.

P-R-O-C-H-Y, 2023-12-19:
@josef2600 Thanks for good report. I will try to investigate more and test it why it's happening and what exactly the issue is.

josef2600, 2023-12-27:
update 3:
this is why i don't call programmers, programmers! who in the hell decided to use c++ 2020 for a microcontroller? IT IS NOT A PC.
any way, i found if i change my reading register to atomic, it will perform way better:
```
std::atomic<uint32_t> my_static_readadc_cX(0);
...
        my_static_readadc_cX.fetch_add(analogReadMilliVolts(pinXan));
        my_static_readadc_cX.fetch_add(analogReadMilliVolts(pinXan));
        my_static_readadc_cX.fetch_add(analogReadMilliVolts(pinXan));
        my_static_readadc_cX.fetch_add(analogReadMilliVolts(pinXan));

```
since volatile has been deprecated,i have read the why from the c++ community, but they are talking about PC not microcontrollers with 50 instructions. so volatile cant be used in it and i think there are a lot of changes needed in the source code. 
on another note, i have a project that uses w5500 for Ethernet and when i need to work on that, i need to downgrade to version 2.0.14 , but if i need to work on my main project, i have to install version 3.0.0 and they are absolutely incompatible! what a sideshow (you know it)!
if version 3 doesn't work, why esp has released it? 

this part is directed to esp company and has almost nothing to do with Arduino itself. but i love to know if any one has any answer to them or am i just an idiot! thank you.
i have few questions:
1. why i cant access the low level instructions? i need to control hardware in my own code. why am i blocked? but in IDF it is possible. 
2. in continues of the number one, why i cant use the controller as what it is, a microcontroller? why it has to have an operating system? because with rtos (or any os) i cant access the hardware directly and i am blocked and it has to do 10 other job before it does my job. for example, if i want to change a pin from 0 to 1, it runs a long code and doesn't do it immediately as it should? 

as i said, i might be wrong. then if it is possible to use this chip directly, please tell me how? because i checked, and it's core is designed with free rtos and without direct access to hardware.
thank you.

VojtechBartoska, 2024-02-20:
@P-R-O-C-H-Y Any action points regarding this issue for 3.0.0 release?

P-R-O-C-H-Y, 2024-02-21:
@josef2600 I am now testing the issue to see, if there is any bug in the code.
When I am taking a look of what you are trying to achieve, why not to use the ADC Continuous mode as you read from multiple pins and calculating an average? That's what ADC Continuous mode is for :)

P-R-O-C-H-Y, 2024-02-21:
@josef2600 Also would be great if you can provide a smallest test sketch that is ready to go with reproducible issue.

P-R-O-C-H-Y, 2024-02-21:
I have no issue with the ADC Oneshot. I have tested multiple channels, reading in a Task, all good to me.
Feel free to reopen the issue if you have any news. Closing it for now.

P-R-O-C-H-Y, 2024-02-21:
I have created this simple sketch which is running fine. Always got precise result of reading.
On 2 volts set on DC power supply I read 2015 mV.  

```
uint32_t adc_reading[4] = {0,0,0,0};
bool adc_read_done = false;

void analog_reading_all_channels_event_task(void *pvParameters)
{
  while (1)
  {
    if(!adc_read_done){
      adc_reading[0] = 0;
      adc_reading[1] = 0;
      adc_reading[2] = 0;
      adc_reading[3] = 0;
      for (int i = 0; i < 4; i++){
        adc_reading[0] += analogReadMilliVolts(32);
        adc_reading[1] += analogReadMilliVolts(33);
        adc_reading[2] += analogReadMilliVolts(34);
        adc_reading[3] += analogReadMilliVolts(35);
        vTaskDelay(1);
      }
      adc_read_done = true;
    }
    vTaskDelay(1000);
  }
}

void setup() {
  // initialize serial communication at 115200 bits per second:
  Serial.begin(115200);
  
  //set the resolution to 12 bits (0-4095)
  analogReadResolution(12);

  xTaskCreatePinnedToCore(analog_reading_all_channels_event_task, "Task_analog_reading", 4000, NULL, 2 , NULL,0);
}

void loop() {
  if(adc_read_done){
    Serial.printf("PIN 32 analog value = %d\n",(adc_reading[0]/4));
    Serial.printf("PIN 33 analog value = %d\n",(adc_reading[1]/4));
    Serial.printf("PIN 34 analog value = %d\n",(adc_reading[2]/4));
    Serial.printf("PIN 35 analog value = %d\n",(adc_reading[3]/4));
    adc_read_done = false;
  }
}
```

closed 2024-02-21 by P-R-O-C-H-Y