espressif/arduino-esp32 #9013: ESP32 BLE dual role: Callback error when connecting to Client

SAOKIM99, 2023-12-18:
### Board

ESP32 Dev Module

### Device Description

KIT ESP32 Dev Module
[image]

### Hardware Configuration

[env:esp32dev]
platform = espressif32 @ 6.4.0
board = esp32dev
framework = arduino
upload_port = COM19
monitor_port = COM19
monitor_speed = 115200

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

default 

### Description

I am developing a project using ESP32's BLE with 2 roles at the same time:
- Server: connects to the phone to send information
- Client: connects to another BLE device
Problem:
- When connecting and disconnecting from the phone:
  + Callback function: only onConnectServer & onDisconnectServer

- When connecting and disconnecting from a BLE device:
+ Callback functions: onConnectServer & onDisconnectServer and onConnectClient & onDisconnecrClient
(see message log)

This affects the connection between the ESP32 and the phone when the ESP32 connects or disconnects with another BLE device.
How can the ESP32 recall the correct device being connected and not cause confusion?

### Sketch

```cpp
#include <Arduino.h>
#include "BLEDevice.h"
//#include "BLEScan.h"
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

[... 206 lines cut ...]
on Connect Client
[  5940][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[  5941][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[  5950][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
on DisConnect Server
on Disconnect Client
ac:0b:fb:67:fd:6a
on Connect Server
on Connect Client
[1062633][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[1062633][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[1062642][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
on DisConnect Server
on Disconnect Client
ac:0b:fb:67:fd:6a
on Connect Server
on Connect Client
[1376076][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[1376077][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[1376086][E][BLERemoteCharacteristic.cpp:289] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dcsleep, 2024-01-21:
I ran into this issue with similar code using PlatformIO. That is, BLEServerCallbacks::onConnect is called when a connection is made using BLEClient::connect. Only BLEClientCallbacks::onConnect should be called when a connection is made.

platform = espressif32 @ 6.5.0
board = adafruit_feather_esp32s3_nopsram
framework = arduino

The problem is also documented in [#7115](https://github.com/espressif/arduino-esp32/issues/7115). [This comment](https://github.com/espressif/arduino-esp32/issues/7115#issuecomment-1411685543) offers a workaround. The [PR (#7851)](https://github.com/espressif/arduino-esp32/pull/7851) that closed [#7115](https://github.com/espressif/arduino-esp32/issues/7115) resolves an issue related to the workaround but does not address the underlying problem.

SAOKIM99, 2024-10-04:
Is there any solution to the problem?
Why does the Espressif framework not have this problem but the Arduino framework does?
@lucasssvaz

lucasssvaz, 2024-10-04:
cc @SuGlider

closed 2025-01-15 by Parsaabasi