espressif/arduino-esp32 #9017: Adding Heltec Wireless_paper to list of supported boards

Lapporatory992, 2023-12-18:
### Related area

Board Support : Heltec Wireless_paper

### Hardware specification

ESP32 Heltec Lora

### Is your feature request related to a problem?

Hi,

Would it possible to add the HeltecESP32 "wireless_paper" board to the list of supported boards  ?
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/boards.txt

Line 1406:
##############################################################
#############################################################
wireless_paper.name=wireless_paper
wireless_paper.vid.0=0x303a
wireless_paper.pid.0=0x1001

wireless_paper.bootloader.tool=esptool_py
wireless_paper.bootloader.tool.default=esptool_py

wireless_paper.upload.tool=esptool_py
wireless_paper.upload.tool.default=esptool_py
wireless_paper.upload.tool.network=esp_ota

wireless_paper.upload.maximum_size=4026368
wireless_paper.upload.maximum_data_size=327680
wireless_paper.upload.flags=
wireless_paper.upload.extra_flags=
wireless_paper.upload.use_1200bps_touch=false
wireless_paper.upload.wait_for_upload_port=false

etc.

Thanks!

### Describe the solution you'd like

Please add the wireless_paper board to Arduino library

### Describe alternatives you've considered

https://www.arduino.cc/reference/en/libraries/heltec-esp32-dev-boards/

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2023-12-19:
Hello @Lapporatory992, feel free to open a PR adding this board.

VojtechBartoska, 2023-12-21:
Hello,

as mentioned by @P-R-O-C-H-Y, you need to add the board on your own or the manufacturer of this board. We are generally not adding any third party boards on our own. \We do not know details, we do not have HW and so on. I think I do not need to explain all the reasons. :)

I am closing this issue as wontfix.

Thanks.

closed 2023-12-21 by VojtechBartoska