espressif/arduino-esp32 #9019: Spiffs on ESP32 version 2.0.14 does not work properly

VladimirVecera, 2023-12-18:
### Board

ESP32-CAM

### Device Description

-

### Hardware Configuration

-

### Version

v2.0.14

### IDE Name

Arduino IDE 2.2.1

### Operating System

Windows 10

### Flash frequency

240 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have upgraded the ESP32 to version 2.0.14 and I am finding that it is causing problems on one project where I have SPIFFS. For example, it doesn't work:
```
      File fileTISKapi = SPIFFS.open("/datapi.txt", FILE_WRITE);
      fileTISKapi.print(api);
      fileTISKapi.close();
```
Spiffs are not saved at all. On an older version of ESP32, I think I had 2.0.9 it worked without a problem. What should the new SPIFFS look like?

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-18:
If [SPIFFS_Test](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_Test/SPIFFS_Test.ino) fails, that implies a hardware issue.  Try setting your flash to DIO.

VojtechBartoska, 2024-01-31:
Hello, closing this as expired due to no answer from author, if needed you can reopen. Thanks

closed 2024-01-31 by VojtechBartoska