espressif/arduino-esp32 #9020: Serial2 breaks Serial1

terrafirma2021, 2023-12-18:
### Board

esp32-s3

### Device Description

wokwi S3
Feather-S3

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

2.2.1

### Operating System

win 10 x64

### Flash frequency

240

### PSRAM enabled

yes

### Upload speed

115200

### Description

// works fine
void sendHOverSerial1() {
  Serial1.begin(115200, SERIAL_8N1, 38, 3); // Re-initialize Serial1 with different pins
  Serial1.write("H"); 
  delay(1000); 
 Serial1.begin(115200, SERIAL_8N1, 38, 1); // Initialize Serial1 back to stock pins
}
// Causes panic 
void sendIOverSerial2() {
  Serial2.begin(115200, SERIAL_8N1, 47, 21);  // Initialize Serial2 with pin swap 
  Serial2.write("I"); 
  delay(1000); 
  // Serial2.begin(115200, SERIAL_8N1, 11, 10); // <------- causes panic
}

Also when 
void sendIOverSerial2() {

is enabled 

void sendHOverSerial1() { // breaks

### Sketch

```cpp
//https://wokwi.com/projects/384481453752614913 attached for easy viewing

void setup() {
  Serial.begin(115200);                
  Serial2.begin(115200, SERIAL_8N1, 11, 10);
  Serial1.begin(115200, SERIAL_8N1, 38, 1);
}

void loop() {
  // Check if Serial1 has received any data
  if (Serial1.available()) {
    char byteReceived1 = Serial1.read();
    Serial.print("Byte received on Serial1: ");
    Serial.println(byteReceived1);
  }

  // Check if Serial2 has received any data
  if (Serial2.available()) {
    char byteReceived2 = Serial2.read();
    Serial.print("Byte received on Serial2: ");
    Serial.println(byteReceived2);
  }

  // Call the functions to send "H" and "I"
  sendHOverSerial1();
  sendIOverSerial2(); // Stops sendHOverSerial1() working
}

void sendHOverSerial1() {
  Serial1.begin(115200, SERIAL_8N1, 38, 3); // Re-initialize Serial1 with different pins
  Serial1.write("H"); 
  delay(1000); 
 Serial1.begin(115200, SERIAL_8N1, 38, 1); // Initialize Serial1 back to stock pins
}

void sendIOverSerial2() {
  Serial2.begin(115200, SERIAL_8N1, 47, 21);  // Initialize Serial2 with pin swap 
  Serial2.write("I"); 
  delay(1000); 
  // Serial2.begin(115200, SERIAL_8N1, 11, 10); // <------- causes panic
}
```

### Debug Message

```plain
00:04.200
100%

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x370

load:0x403c9700,len:0x900

load:0x403cc700,len:0x2364

SHA-256 comparison failed:

Calculated: bea01f04c6f0e287a682f128805e3fce115955179100e98d8f412fe7697f8bdc

Expected: 08ea492e448f88de75319ed18ac319444e578d9c6bc7003c5c4807382bf389bd

Attempting to boot anyway...

entry 0x403c98ac
```

### Other Steps to Reproduce

https://wokwi.com/projects/384481453752614913

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-12-18:
I don't know about the S3, but on ESP32-WROOM32UE "Serial1" isn't available because it's used internally for the flash memory.  I always use "Serial" and "Serial2".

terrafirma2021, 2023-12-18:
@mrengineer7777 
Thanks for the reply, serial1 is working without issues, however serial2 is the issue, that only occurs with the above config,

using standard configs, both serial 1+2 work without issues

terrafirma2021, 2024-01-02:
Hi, any updates to this please :)

SuGlider, 2024-01-03:
Yes, there is some problem with ESP32-S3 UART2 indeed.
It sends a BREAK whenever it is started causing a bad reading.
It is not related to the GPIOs 10 and 11.

I'll continue to investigating it.

terrafirma2021, 2024-01-03:
Thank you for getting round to looking at this.

SuGlider, 2024-01-03:
@terrafirma2021 - Just to check, you are using Arduino Core 3.0.0-Alpha3 or some other specifc version (or actually it was checked out from master)?

I don't get any panic (in the sense of a Guru Mediation and a reset), but it seems that there is some issue with how `Serial2` is working.

Please post the error messages / output from Serial Monitor / etc.

SuGlider, 2024-01-06:
I found out something:

``` cpp
void sendHOverSerial1() {
  Serial1.begin(115200, SERIAL_8N1, 38, 3); // Re-initialize Serial1 with different pins
  Serial1.write("H"); 
  delay(1000); 
 Serial1.begin(115200, SERIAL_8N1, 38, 1); // Initialize Serial1 back to stock pins
}

void sendIOverSerial2() {
  Serial2.begin(115200, SERIAL_8N1, 47, 21);  // Initialize Serial2 with pin swap 
  Serial2.write("I"); 
  delay(1000); 
  // Serial2.begin(115200, SERIAL_8N1, 11, 10); // <------- causes panic
}
```

`sendHOverSerial1()` doesn't change the RX IO.
`sendIOverSerial2()` changes RX IO.

Changing RX IO causes a UART BREAK signal in the line. This causes an error in the HardwareSerial driver.

Other detail:
`HardwareSerial::begin()` has a problem that flushes the TX FIFO by using IDF internally. 
Therefore, when the function writes to the Serial, the byte goes to the FIFO, but when next executing `begin()`, it will flush FIFO and `loop()` won't read any data.

### I'm working on some possible patch.

terrafirma2021, 2024-01-09:
Hi, sorry for not reaching back to you sooner, I never tested on the latest core ide it was an earlier version. 

The issue still exists as noted above on the latest core ide

SuGlider, 2024-01-13:
@terrafirma2021 - #9095 shall fix it for Arduino 3.0.0.
I'll also add a patch for the Arduino 2.0.15.

terrafirma2021, 2024-01-13:
Your a star thank you very much :D

VojtechBartoska, 2024-02-20:
Closing this as solved by above mentioned PR.