espressif/arduino-esp32 #9022: Arduino Core2.0.3

savagerex, 2023-12-19:
### Board

ESP32 Wrover e

### Device Description

ESP32 Wrover e + IIS3DWB + External Flash

### Hardware Configuration

NONE

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

When i user Arduino 2.0.3 Core.
the operation as below.
BLE (Transfer data) -> MQTT (Transfer data) -> OTA(AsyncElegantOTA) ....OK!!

When i user Arduino 2.0.11 Core.
the operation as below.
BLE (Transfer data) -> MQTT (Transfer data) -> OTA(AsyncElegantOTA) ....Fail!!

### Sketch

```cpp
Because the code is too huge.

if you need , i will send to you by email.

Thanks
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-19:
How do you expect anyone to be able to help you with an incomplete report like this?
If the individual elements work, but it doesn't work when you combine them, then you likely have a memory issue. 
As has been mentioned many times, this forum is for issues with the code here, not your code.

VojtechBartoska, 2023-12-21:
Hello, closing this as it is incomplete and too generic.

I suggest to ask similar questions on Gitter channel https://app.gitter.im/#/room/#espressif_arduino-esp32:gitter.im

Thanks for understanding.

closed 2023-12-21 by VojtechBartoska