espressif/arduino-esp32 #9023: Aruduino code craashing in ESP-IDF using arduino as component

Abhay-2412, 2023-12-19:
### Board

DFRobot firebeetle 2 esp32s3

### Device Description

evalkit with ov5640 camera

### Hardware Configuration

evalkit with ov5640 camera

### Version

v2.0.6

### IDE Name

Arduino IDE and ESP-IDF using arduino as component

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

 460800

### Description

i have a working arduino project on esp32s3 version v2.0.6 due to some requirement we need to shift it to esp-idf we decided to use arduino as component same version 2.0.6 in there our program  is crashing at middle of execution. i juse need some suggestions to how to debug it and how can i compare both projects and findout what's wrong

### Sketch

```cpp
will upload if needed
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40056f60  PS      : 0x00060930  A0      : 0x820090e4  A1      : 0x3fcebf50
A2      : 0x00000000  A3      : 0x3d8009a4  A4      : 0x0004b000  A5      : 0x00000000  
A6      : 0x7a408843  A7      : 0x81438942  A8      : 0x00000000  A9      : 0x3fcebf00
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3fcb4648  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000000b  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00004aff

Backtrace: 0x40056f5d:0x3fcebf50 |<-CORRUPTED

ELF file SHA256: 805321dde2140873

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4207e182
0x4207e182: esp_pm_impl_waiti at C:/Espressif/frameworks/esp-idf-v4.4.5/components/esp_pm/pm_impl.c:832
```

### Other Steps to Reproduce

like i said it works perfectly in arduino ide when i use espressif package of version 2.0.6 and using same package in esp idf as arduino as component it crashing  just want to know how to troubleshoot it and i'm sorry if quwation looks dumb i'm complete newbie

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-12-19:
What ESP-IDF version you are using?

Abhay-2412, 2023-12-19:
Hello thanks for the response iam using ESP-IDF version v4.4.5

lbernstone, 2023-12-19:
AFAIK, DFS/light sleep is not supported in arduino-esp32.

Abhay-2412, 2023-12-19:
> AFAIK, DFS/light sleep is not supported in arduino-esp32.

sorry iam a complete newbie i did'nt understand what you meant

Abhay-2412, 2023-12-19:
> The 2.0.6 version of ESP Arduino core is based on IDF v4.4.3, so you should be using that version instead. In the ESP-IDF project are you using the same SDK config as it's used in the Arduino?

hello wehre can i find my arduino project sdkconfig.h
`C:\Users\user1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\tools\sdk\esp32s3`
i found a sdkconfig.h file here is this the sdkconfig file you mean?

lbernstone, 2023-12-19:
> > AFAIK, DFS/light sleep is not supported in arduino-esp32.
> 
> sorry iam a complete newbie i did'nt understand what you meant

Then please ask whoever wrote the code containing esp_pm_impl_waiti to open an issue.

Abhay-2412, 2023-12-19:
> esp_pm_impl_waiti

i mean is it the reason for the crash happening because i did same thing with esp32 whie shifting to esp idf there i did'nt find any issue problem or crash issue

Abhay-2412, 2023-12-19:
> The 2.0.6 version of ESP Arduino core is based on IDF v4.4.3, so you should be using that version instead. In the ESP-IDF project are you using the same SDK config as it's used in the Arduino?

hello taking your suggestion i tried it on esp idf v4.4.3 and getting the same issue with that as well

Abhay-2412, 2023-12-19:
> esp_pm_impl_waiti
/
and `esp_pm_impl_waiti ` you are talking about comes with package right since its under the framework fold?

Abhay-2412, 2023-12-20:
> > > AFAIK, DFS/light sleep is not supported in arduino-esp32.
> > 
> > 
> > sorry iam a complete newbie i did'nt understand what you meant
> 
> Then please ask whoever wrote the code containing esp_pm_impl_waiti to open an issue.

in my code i used arduino as component and copy pasted same arduino code in there i did'nt use anything called `esp_pm_impl_waiti  ` in my code if that is creating problem then help me out how to disable it

Abhay-2412, 2023-12-20:
> The 2.0.6 version of ESP Arduino core is based on IDF v4.4.3, so you should be using that version instead. In the ESP-IDF project are you using the same SDK config as it's used in the Arduino?

hello i overlooked my logs actually with this version im not getting any guru meditation error instead it got rebooted

Rebbot logs
```

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b68
0x40375b68: esp_restart_noos at C:/Espressif_4.4.3/frameworks/esp-idf-v4.4.3/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x18b0
load:0x403c9700,len:0xea4
load:0x403cc700,len:0x3398
SHA-256 comparison failed:
Calculated: 5e86d2db3e0c52ed98cab7cc186b4c1c2cfe718cb1622e9bac9968e240f036bc
Expected: a326e47a7b1bbda922e532f559df6af99730d995c121e6b55d2373085a1677c9
Attempting to boot anyway...
entry 0x403c9994
I (42) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (42) boot: compile time 14:06:47
I (43) boot: chip revision: 0
I (45) qio_mode: Enabling default flash chip QIO
I (50) boot.esp32s3: Boot SPI Speed : 80MHz
I (55) boot.esp32s3: SPI Mode       : QIO
I (59) boot.esp32s3: SPI Flash Size : 16MB
I (64) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (88) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (95) boot:  2 app0             OTA app          00 10 00010000 00300000
I (103) boot:  3 app1             OTA app          00 11 00310000 00300000
I (110) boot:  4 spiffs           Unknown data     01 82 00610000 00700000
I (118) boot:  5 model            Unknown data     01 82 00d10000 002e0000
I (126) boot:  6 coredump         Unknown data     01 03 00ff0000 00010000
I (133) boot: End of partition table
I (137) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=16c3ch ( 93244) map
I (160) esp_image: segment 1: paddr=00026c64 vaddr=3fc95ca0 size=05b58h ( 23384) load
I (164) esp_image: segment 2: paddr=0002c7c4 vaddr=40374000 size=03854h ( 14420) load
I (169) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=83bcch (539596) map
I (256) esp_image: segment 4: paddr=000b3bf4 vaddr=40377854 size=0e448h ( 58440) load
I (267) esp_image: segment 5: paddr=000c2044 vaddr=50000000 size=00010h (    16) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
E (1042) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='ec4b6bc7'
Image checksum='ffffffff'
```

Abhay-2412, 2023-12-21:
hello i identified at which particular function my code crashing i wrote a custom capture photo function which works perfectly fine in arduino IDE but esp idf it seems crashing

```
void capturePhotoSaveSCD( WiFiClient client ) {
  camera_fb_t * fb = NULL; // pointer
  bool ok = 0; // Boolean indicating if the picture has been taken correctly
  int8_t errorLEDdimmer = 0;
  int preCount = 0;
  int counter = 0;

  do {
    // Take a photo with the camera
    while (1)
    {
      if(counter >= 500)
      {
        break;
      }

      preCount = counter;
      counter = getFrameCounter();
      if (preCount < counter)
      {
        fb = esp_camera_fb_get();

        if (!fb) {
          Serial.println("Camera capture failed");
          return;
        }

        long len = fb->len;
        Serial.println("Before mallocing");
        static uint8_t* prev_buf = (uint8_t*)ps_malloc(fb->len);
        static uint8_t* prev_buf_1 = (uint8_t*)ps_malloc(fb->len);
        Serial.println("After mallocing");

        if (((counter % 20) == 1))
        {
          memcpy(prev_buf, fb->buf, fb->len);
        }
        if (((counter % 4) == 2))
        {
          memcpy(prev_buf_1, fb->buf, fb->len);
        }
        // check if file has been correctly
        ok = checkPhoto(fb);
        esp_camera_fb_return(fb);
        if (ok)
        {
          if (((counter % 4) == 2) && (counter > 20))
          {
            if (((counter % 20) == 2))
            {
              long int lTime1 = millis();

              client.setNoDelay(true);
              client.write(prev_buf, len);

              long int lTime2 = millis();
              uint32_t uAvg = lTime2 - lTime1;

            }
            else
            {
              long int lTime1 = millis();
              client.setNoDelay(true);
              client.write(prev_buf_1, len);

              long int lTime2 = millis();
              uint32_t uAvg = lTime2 - lTime1;

            }
          }
          if (((counter % 20) == 4) && (counter > 20))
          {
            long int lTime1 = millis();
            client.setNoDelay(true);
            client.write(prev_buf_1, len);

            long int lTime2 = millis();
            uint32_t uAvg = lTime2 - lTime1;

          }

        }
      }
    }
    cam_stop();

  } while ( !ok );
}
```
in above function its crashing is mallocing 
 ```
       static uint8_t* prev_buf = (uint8_t*)ps_malloc(fb->len);
        static uint8_t* prev_buf_1 = (uint8_t*)ps_malloc(fb->len);
```

can anyone help me to understand why above function works fine in arduino ide and cfrashing when i use same arduino core in esp idf .

VojtechBartoska, 2023-12-21:
Hello @Abhay-2412, please ask on this project Gitter channel: https://app.gitter.im/#/room/#espressif_arduino-esp32:gitter.im.

I am closing this ticket as it is more likely a question which have been answered multiple times, please also search across open and close issue in this repo.

Thanks for understanding.

closed 2023-12-21 by VojtechBartoska

Abhay-2412, 2023-12-21:
hello @VojtechBartoska I submitted a support ticket as I couldn't locate a definitive solution in the available repositories, both open and closed. I acknowledge there might be resources I have missed, and if you are aware of repositories that could provide insights, I would greatly appreciate your guidance.

VojtechBartoska, 2023-12-21:
You can take a look on official documentation here: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

Also there is one talk from conference DevCon22, which is covering this topic and may help you, available [here](https://www.youtube.com/watch?v=aPXvIInTi4s&t=2s&ab_channel=EspressifSystems)

Abhay-2412, 2023-12-21:
hello @VojtechBartoska thanks for the documents but i thinnk i identifed the issue i added some logs and i got below output

```
Total PSRAM size: 0
Attempting to allocate 307200 bytes from PSRAM
Failed to allocate 307200 bytes from PSRAM
Free PSRAM After Allocation: 7157023

```
similar logs i added in my arduino code and i got this

```
21:38:25.354 -> Total PSRAM size: 8385747
21:38:25.354 -> Attempting to allocate 307200 bytes from PSRAM
21:38:25.354 -> Successfully allocated 307200 bytes from PSRAM
21:38:25.354 -> Free PSRAM After Allocation: 6834103

```

i added below snippet

       ```
 Serial.print("Total PSRAM size: ");
        Serial.println(ESP.getPsramSize());
        Serial.print("Attempting to allocate ");
        Serial.print(fb->len);
        Serial.println(" bytes from PSRAM");

// Attempt to allocate memory from PSRAM
        static uint8_t* prev_buf = (uint8_t*)ps_malloc(fb->len | MALLOC_CAP_SPIRAM);

      if (!prev_buf) {
  Serial.print("Failed to allocate ");
  Serial.print(fb->len);
  Serial.println(" bytes from PSRAM");
} else {
  Serial.print("Successfully allocated ");
  Serial.print(fb->len);
  Serial.println(" bytes from PSRAM");
}

Serial.print("Free PSRAM After Allocation: ");
Serial.println(heap_caps_get_free_size(MALLOC_CAP_SPIRAM));

    if (!prev_buf) {
        Serial.println("Failed to allocate memory for prev_buf");
        // Handle the error, possibly by freeing other resources and returning
        return;
      }
```

so issue is PSRAM is not getting detected iam using esp32s3 firebeetle 2 i have enabled external spi ram support from menuconfig and changed it into OPIPSRAM is there anything else should i need to do

Abhay-2412, 2023-12-22:
hello i found out thje solution there some thing wrong with functions in `esp32-hal-psram.c` it was going to else part of the function when i comment else part of function then everything is working fine

```
#include "esp32-hal.h"

// #if CONFIG_SPIRAM_SUPPORT
#include "esp_spiram.h"
#include "soc/efuse_reg.h"
#include "esp_heap_caps.h"

static volatile bool spiramDetected = false;
static volatile bool spiramFailed = false;

bool psramInit(){
    if (spiramDetected) {
        return true;
    }
#ifndef CONFIG_SPIRAM_BOOT_INIT
    if (spiramFailed) {
        return false;
    }
    uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_PKG);
    uint32_t pkg_ver = chip_ver & 0x7;
    if (pkg_ver == EFUSE_RD_CHIP_VER_PKG_ESP32D2WDQ5 || pkg_ver == EFUSE_RD_CHIP_VER_PKG_ESP32PICOD2) {
        spiramFailed = true;
        log_w("PSRAM not supported!");
        return false;
    }
    esp_spiram_init_cache();
    if (esp_spiram_init() != ESP_OK) {
        spiramFailed = true;
        log_w("PSRAM init failed!");
        pinMatrixOutDetach(16, false, false);
        pinMatrixOutDetach(17, false, false);
        return false;
    }
    if (!esp_spiram_test()) {
        spiramFailed = true;
        log_e("PSRAM test failed!");
        return false;
    }
    if (esp_spiram_add_to_heapalloc() != ESP_OK) {
        spiramFailed = true;
        log_e("PSRAM could not be added to the heap!");
        return false;
    }
#endif
    spiramDetected = true;
    log_d("PSRAM enabled");
    return true;
}

bool IRAM_ATTR psramFound(){
    return spiramDetected;
}

void IRAM_ATTR *ps_malloc(size_t size){
    if(!spiramDetected){
        return NULL;
    }
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
}

void IRAM_ATTR *ps_calloc(size_t n, size_t size){
    if(!spiramDetected){
        return NULL;
    }
    return heap_caps_calloc(n, size, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
}

void IRAM_ATTR *ps_realloc(void *ptr, size_t size){
    if(!spiramDetected){
        return NULL;
    }
    return heap_caps_realloc(ptr, size, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
}

// #else

// bool psramInit(){
//     return false;
// }

// bool IRAM_ATTR psramFound(){
//     return false;
// }

// void IRAM_ATTR *ps_malloc(size_t size){
//     return NULL;
// }

// void IRAM_ATTR *ps_calloc(size_t n, size_t size){
//     return NULL;
// }

// void IRAM_ATTR *ps_realloc(void *ptr, size_t size){
//     return NULL;
// }

// #endif
```