espressif/arduino-esp32 #9033: TIMERG0 ESP32-S2 Not exist

dzungpv, 2023-12-22:
### Board

ESP32-S2

### Device Description

WEMOS S2 MINI V1.0.0

### Hardware Configuration

GPIO 9 and GPIO11 use for hardware Serial1

### Version

v2.0.14

### IDE Name

ESP IDF

### Operating System

Macos 13, Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I migrate from ESP32 to ESP32-S2 but could not build, it show error with TIMERG0

### Sketch

```cpp
#ifdef ESP32
void feedTheDog()
{
  TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
  TIMERG0.wdt_feed = 1;
  TIMERG0.wdt_wprotect = 0;
}
#endif
```

### Debug Message

```plain
In function 'void feedTheDog()':
/config.h:403:11: error: 'struct timg_dev_t' has no member named 'wdt_wprotect'; did you mean 'wdtwprotect'?
   TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
           ^~~~~~~~~~~~
           wdtwprotect
/config.h:404:11: error: 'struct timg_dev_t' has no member named 'wdt_feed'; did you mean 'wdtfeed'?
   TIMERG0.wdt_feed = 1;
           ^~~~~~~~
           wdtfeed
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-22:
What's the question here?  The timer groups are hardware specific, and part of the ESP-IDF.
https://github.com/espressif/esp-idf/blob/v4.4.6/components/soc/esp32s2/include/soc/timer_group_struct.h

dzungpv, 2023-12-23:
> What's the question here? The timer groups are hardware specific, and part of the ESP-IDF. https://github.com/espressif/esp-idf/blob/v4.4.6/components/soc/esp32s2/include/soc/timer_group_struct.h

I could not build it show wdt_wprotect and wdt_feed error mean it only available on the ESP32 S1? I also change the description.

me-no-dev, 2023-12-23:
There is no action for us here. You are talking about hardware registers. The chips have what they have, we can not change the chip design now, can we :) Closing as off topic.

closed 2023-12-23 by me-no-dev

wotupfoo, 2024-10-27:
@dzungpv a more friendly answer here is that the timer hardware changed from ESP32 to ESP-S2|S3 and even though some stayed the same.

@me-no-dev @lbernstone maybe get off your high horse and instead of dismissing this issue, acknowledge that you broke forward compatibility instead of dismissing this poster.

@dzungpv  here is what happened
TIMERG0 exists on both ESP32 and ESP32-S2|3 but for some unknown reason, Espressif decided to change the names of the registers. Go figure.
So for ESP32 you have

    volatile timg_wdtfeed_reg_t wdtfeed;
    volatile timg_wdtwprotect_reg_t wdtwprotect;

But for ESP32-S2|S3 you have
    uint32_t wdt_feed;                                /*Write any value will feed SWDT*/
    uint32_t wdt_wprotect;                            /*If change its value from default  then write protection is on.*/

So, the header file you're using because you changed to the WEMOS S3 (ESP32-S2) no longer has the registers of the ESP32.
[soc/esp32s2/include/soc/timer_group_struct.h](https://github.com/espressif/esp-idf/blob/v4.4.6/components/soc/esp32s2/include/soc/timer_group_struct.h)
[soc/esp32/include/soc/timer_group_struct.h](https://github.com/espressif/esp-idf/blob/v4.4.6/components/soc/esp32/include/soc/timer_group_struct.h)

EvillNooB, 2025-02-02:
@wotupfoo thank you for pointing in the right direction 😅

was having the same problem and replacing 

```
  TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
  TIMERG0.wdt_feed = 1;
  TIMERG0.wdt_wprotect = 0;
```

with 
```
 TIMERG0.wdtwprotect.wdt_wkey=1356348065;
 TIMERG0.wdtfeed.wdt_feed = 1;
 TIMERG0.wdtwprotect.wdt_wkey=0;
```
solved the issue