espressif/arduino-esp32 #9037: Wrong PIN for RGB LED on board lolin_s3_mini

mp-se, 2023-12-25:
### Board

lolin_s3_mini

### Device Description

Its not possible to control the RGB led on this board with the neopixelWrite() function since the PIN number is wrong in the pins_arduino.h

The pin definition for this board is
``#define RGB_BUILTIN LED_BUILTIN``

In neopixelWrite the following operation is done
```
#ifdef RGB_BUILTIN
  if(pin == RGB_BUILTIN){
    _pin = RGB_BUILTIN-SOC_GPIO_PIN_COUNT;
  }
#endif
```

So the for this to work the definition for this board needs to be 
``#define RGB_BUILTIN LED_BUILTIN+SOC_GPIO_PIN_COUNT``

Not sure if the macro or the LED_BUILTIN constant should be changed so I put it as a bug report instead of PR.

If this is changed its possible to change the rgb led color on this board. 

### Hardware Configuration

Just plain board not connections

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

It should be possible to control the RGB led with neopixel on this board.

### Sketch

```cpp
#include <Arduino.h>

void setup() {
  neopixelWrite(RGB_BUILTIN, 255, 0, 0);
  delay(1000);
  neopixelWrite(RGB_BUILTIN, 0, 255, 0);
  delay(1000);
  neopixelWrite(RGB_BUILTIN, 0, 0, 255);
}

void loop() {
}
```

### Debug Message

```plain
E (240) rmt: rmt_set_gpio(526): RMT GPIO ERROR
E (240) rmt: rmt_config(686): set gpio for RMT driver failed
[   331][D][esp32-hal-rmt.c:615] rmtInit():  -- TX RMT - CH 0 - 1 RAM Blocks - pin 254
[   337][E][esp32-hal-rmt.c:631] rmtInit(): RMT failed to initilize.
[   343][E][esp32-hal-rgb-led.c:18] neopixelWrite(): RGB LED driver initialization failed!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Eric-IOT, 2024-03-04:
hello, 
**SIMILIAR BEHAVIOUR**
running IDE on windows 10 :
Version: 2.3.0
Date: 2024-02-07T14:04:32.754Z
CLI Version: 0.35.2
and
Adafruit NeoPixel version 1.12.0

i get similar warnings when loading code, but the code is running

In the pin definition for LOLIN S3mini,  i made the proposed changes of @mp-se and the warning have disapeared.

mp-se, 2024-03-04:
The change has already been implemented for the esp32s3 board here https://github.com/espressif/arduino-esp32/blob/master/variants/esp32s3/pins_arduino.h. So it has partly been implemented for some of the s3 boards but this should apply for all boards based on the s3 variant.

einglis, 2024-03-09:
I believe the colour order is different too?  The example sketch `BlinkRGB` should go:
* White, Red, Green, Blue, ...
but actually goes:
* White, **Green**, **Red**, Blue, ...

SuGlider, 2024-03-15:
This has been fixed for the Arduino Core 3.0.0
https://github.com/espressif/arduino-esp32/blob/master/variants/lolin_s3_mini/pins_arduino.h

SuGlider, 2024-03-15:
We are close to release Arduino 2.0.15. 
Therefore, we may submit a PR to `release/v2.x` branch in order to apply changes for all the board.

SuGlider, 2024-03-16:
> We are close to release Arduino 2.0.15. Therefore, we may submit a PR to `release/v2.x` branch in order to apply changes for all the board.

PR #9381 will add/backport all changes to 2.0.15

SuGlider, 2024-03-21:
Fixed by #9381

closed 2024-03-21 by SuGlider

profharris, 2024-10-31:
Trying to fix the NeoPixel GRB color output instead of the expected RGB, 
I tried adding the line recommended above:
     "#define RGB_BUILTIN LED_BUILTIN+SOC_GPIO_PIN_COUNT"

That line locked up my LOLIN S3 Mini Pro.  I had to do the Hold down BOOT (Left Button),
Press and release the [Reset] Button, release the [BOOT] button... to put my S3 Mini Pro
into “Bootloader Mode”. Then remove the line, and re-upload my sketch.  

Perhaps it wasn't locked up, maybe the line directed RGB_BUILTIN to somewhere 
other than Pin 8, so the output was being sent who knows where. -profharris

SuGlider, 2024-11-14:
RGB_BUILTIN shall go in the pins_arduino.h file in the variants folder into the board definition.

It doesn't work if added to the sketch code.

PR #9381 (2024-03-18): backport: variants from 3.0.0 + fixes
## Description of Change

Backports changes from 3.0.0 to `variants` and fixes the "unphone8" wrong board definition.
Only changes boards that were already in 2.0.14. No new board is added.

RGB_LED is now fixed for all boards with NEO LED. 
The same for BUILTIN_LED. Both will work fine with `digitalWrite()` and `neo_pixel_write()`
Applied fixes done by the board makers in 3.0.0 based on PR #8726 

## Tests scenarios
Quick random testing with some boards.

## Related links

Closes #9037