espressif/arduino-esp32 #9039: Update from Filesystem File Assert Error Crash

hicham-vv, 2023-12-26:
### Board

az-delivery-devkit-v4

### Device Description

az-delivery-devkit-v4

### Hardware Configuration

Nothing

### Version

v2.0.8

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

40 MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello hope you are doing fine, 
i am facing a problem with my esp32 , 
the system crash time to time when it tries to Starting update " OTA " from a binary file located in the filesystem "SPIFFS" 

Error 
assert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))

### Sketch

```cpp
void updateFromFS()
{

  // Open the file for reading
  File file = SPIFFS.open("/update.bin");

  // Check if the file is available
  if (!file) {
    Serial.println("Failed to open file for reading");
    return;
  }

  Serial.println("Starting update...");

  // Get the file size
  size_t fileSize = file.size();
  // Start the update process
  if (!Update.begin(fileSize)) {
    Serial.println("Cannot do the update: Not enough space or other start error");
    file.close(); // Close the file early to free resources
    return;
  }

  // Perform the actual update
  size_t written = Update.writeStream(file);

  // Check if the written size is less than expected
  if (written != fileSize) {
    Serial.println("Written less data than expected");
    // Abort the update to avoid a possible corrupt firmware
    Update.abort();
  }

  // If the update was successful, the microcontroller will reset
  if (Update.end()) {
    Serial.println("Successful update");
  } else {
    Serial.println("Error Occurred: " + String(Update.getError()));
    Update.abort(); // Abort the update
    file.close(); // Close the file before returning
    return;
  }

  // Close the file after the update
  file.close();

  // Notify the user that the device will reset
  Serial.println("Reset in 4 seconds...");
  delay(4000);

  // Restart the microcontroller
  ESP.restart();

}
```

### Debug Message

```plain
SPIFFS Total Bytes:1318001
SPIFFS Used Bytes:277857
Enough Space SPIFFS 
0.00%
4.00%
8.00%
12.00%
16.00%
20.00%
24.00%
28.00%
32.00%
36.00%
40.00%
44.00%
48.00%
52.00%
56.00%
60.00%
64.00%
68.00%
72.00%
76.00%
80.00%
84.00%
88.00%
92.00%
96.00%
100.00%
Time spent downloading firmware =471262
Stopped client
382432
382432

Starting update...

assert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))

Backtrace: 0x40083bed:0x3ffbf0bc |<-CORRUPTED

ELF file SHA256: 09c1fcc05ec632bb
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-26:
It looks like you need to dereference the pointer in writeStream.  You want the data, not the file handle.

hicham-vv, 2023-12-27:
> It looks like you need to dereference the pointer in writeStream. You want the data, not the file handle.

the function writeStream takes a reference to a Stream object as its parameter. When i call writeStream with my File object (file), i think i'm  already doing it correctly, as long as File is derived from Stream or is compatible with it. In the context of Arduino and similar environments, File usually inherits from Stream, so i should be good to do that.

so i think i don't need to pass a pointer or a data string directly to writeStream. The function expects a Stream reference, and the File object  represents a stream of bytes from the file /update.bin. This means the function will read the bytes from this file stream

Unless I’m mistaken ...

lbernstone, 2023-12-27:
The semaphore error might indicate a timeout, but you would not get a memory corruption from a timeout (you would get a useable backtrace). If you are having timing issues, you may need to chunk the data yourself so that you can put a breather in there every few seconds.
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/SD_Update/SD_Update.ino#L22

closed 2025-01-15 by Parsaabasi