espressif/arduino-esp32 #9040: Analog Read sometimes breaks u8g2 library in 3.0.0 alpha dev

ForrestFire0, 2023-12-26:
### Board

LOLIN S2 Mini (ESP32 S2)

### Device Description

Hardware is esp32 lolin s2 mini board attached to a cheap typical joystick module and to a oled screen. The pinout is in the code. The individual components work separately (and worked together on previous versions of arduino-esp32 framework) to I don't think it's the wiring, happy to share.

### Hardware Configuration

used for u8g2 SPI connection
#define SPI_CS 35
#define SPI_DC 37
#define SPI_RST 33

// the analog read in question.
#define X_PIN 5
#define Y_PIN 9

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11 home

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am trying to connect my joystick (analogRead single shot) to my u8g2 screen.

Both components work independently.

See the sketch. When the function "joyReadCalibrationSample" is commented out, the correct print out appears. When I uncomment "joyReadCalibrationSample" and upload, the screen does not show anything.

### Sketch

```cpp
#include <Arduino.h>
#include <U8g2lib.h>

#include <SPI.h>

#define SPI_CS 35
#define SPI_DC 37
#define SPI_RST 33

// Green wire
#define X_PIN 5
// Orange Wire
#define Y_PIN 9

U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, SPI_CS, SPI_DC, SPI_RST); //ESP32S2 board with SPI

uint32_t x_sum;
uint32_t y_sum;
uint32_t count;
uint16_t x_center, y_center;
void joyReadCalibrationSample() {
    x_sum += analogRead(X_PIN);
    y_sum += analogRead(Y_PIN);
    count++;
    x_center = x_sum / count;
    y_center = y_sum / count;
}

void setup(void) {
    u8g2.begin();
}

void loop(void) {
    u8g2.clearBuffer();					
    u8g2.setFont(u8g2_font_ncenB08_tr);
    char buff[100];
    int val = analogRead(5);
    sprintf(buff, "val: %d\n", val);
//    joyReadCalibrationSample(); // comment me out, working! leave me in, broken!
    Serial.println(x_sum); // 0s when calibration is not active.
    Serial.println(y_sum);
    u8g2.drawStr(0, 10, buff);	
    u8g2.sendBuffer();
}
```

### Debug Message

```plain
I set the core debug level to verbose and did not get any output on the serial monitor other than the prints in my sketch. I think I am doing the debug incorrectly or maybe there is an issue with the esp32s2s ports.
```

### Other Steps to Reproduce

Was working on the stable version of arduino-esp32, however I need esp idf 5.1 functionality so need to use the dev version. I don't see any of my needed functionality as stuff that is being messed with.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ForrestFire0, 2023-12-26:
**UPDATE**

Of course it had nothing to do with the function.

analogRead on pin 5 works great, on 9 it breaks the screen.

```c
void loop(void) {
    u8g2.clearBuffer();					
    u8g2.setFont(u8g2_font_ncenB08_tr);
    char buff[100];
    int val = analogRead(5); // WORKS, 9 DOES NOT WORK!!!
    sprintf(buff, "val: %d\n", val);
    u8g2.drawStr(0, 10, buff);	
    u8g2.sendBuffer();
}
```

ForrestFire0, 2023-12-27:
I found a work for my project by using pins 3 and 5 for my joystick.

SuGlider, 2024-01-03:
@ForrestFire0 - I just checked the ESP32-S2 Lolin Mini board about its pin definiton.
I found that GPIO 9 is used as MISO. 
Therefore, given that u8g2 Library uses SPI, this GPIO will be busy as MISO in the SPI function, not INPUT.

https://github.com/espressif/arduino-esp32/blob/ab6a25ed8db8431f6d0464b560445ed9e0e62be6/variants/lolin_s2_mini/pins_arduino.h#L31-L34

SuGlider, 2024-01-03:
Mysterious is solved! Have fun with your project!

ForrestFire0, 2024-01-03:
Wonderful. Purely for my own understanding, why would configuring the pin with pin mode cause the SPI to break? Does it disable the SPI interface?

SuGlider, 2024-01-03:
> Wonderful. Purely for my own understanding, why would configuring the pin with pin mode cause the SPI to break? Does it disable the SPI interface?

In Arduino 3.0.0 there is a new feature called Peripheral Manager.
In the specific case of SPI, when any of its 4 pins (MOSI, MISO, CLK and CS) are used for any other function, for instance, with `pinMode()`, the whole SPI is detached. Therefore, SPI stops working, being necessary to restart it with `SPI.begin()`.

I think that your example brings a good questioning because most displays will only care about using MOSI and not MISO.
I could also use CS pin  for something else and only set it as CS whenever it is necessary.

Therefore, I would say that we may review it in order to allow using SPI pins partly.

ForrestFire0, 2024-01-03:
I’m happy to close this issue, it seems like at the moment this is an intended (maybe undocumented?) feature. Thanks to everyone doing all this great work, I have been an ESP-Arduino appreciator for a few years now.

closed 2024-01-03 by ForrestFire0