espressif/arduino-esp32 #9041: WROVER-B PSRAM fails to initialize on newer versions of arduino-esp32

marchingband, 2023-12-27:
### Board

ESP32 WROVER-B

### Device Description

ESP32 WROVER-B on a custom PCB (based on this open-source design https://www.sparkfun.com/products/21307)

### Hardware Configuration

DAC, eMMC, opto isolator

### Version

v2.0.14

### IDE Name

Arduino

### Operating System

macos

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have an application which I have been using for many years without issue, however I received a board from a user who made his own with what looks like an older version of the ESP32 WROVER-B, and the PSRAM fails.
If I upload the application from my older laptop, which is running 2-year old versions of the Arduino-esp32, the program works as expected. With the newer versions, it fails. Using newer version of ESP32 WROVER-B, the new library works fine, it is just this one (seemingly older) chip on which is fails.

### Sketch

```cpp
https://github.com/marchingband/wvr/examples/wvr_basic/wvr_basic.ino
```

### Debug Message

```plain
rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
E (367) esp_core_dump_fla␞␆�: No core dump partition found!
E (367) esp_core_dump_flash: No core dump partition found!
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (257) psram: PSRAM ID read error: 0x7aa9ae06
[    14][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
Total heap: 312908
Free heap: 285172
Total PSRAM: 0
Free PSRAM: 0
CPU freq: 240
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

marchingband, 2023-12-27:
[image]

I don't know if this will be helpful, but here is a photo of the part in question.

Jason2866, 2023-12-27:
A quick look in the project you linked shows that many IDF calls and functions are used.
core 2.0.14 is based on IDF 4.4.6. There are changes in IDF which you have to port.

marchingband, 2023-12-27:
@Jason2866 sorry maybe I was not clear enough.
This project works well on most WROVER-B chips, using any version including 2.0.14.
There is only a tiny subset of chips, like the one in the photo, for which the project fails to initialize PSRAM, and this failure only occurs on 2.x.x, but not 1.x.x.

Jason2866, 2023-12-27:
Do you have added the psram fixes? Probably a ESP32 rev.0 or 1 which does need

Jason2866, 2023-12-27:
Sorry, never used ArduinoIDE. Doing everything with Platformio

marchingband, 2023-12-27:
@Jason2866 I also use PIO a lot, and I have tried the same thing with PIO and this line:
```
build_flags = 
    -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue
    -DCORE_DEBUG_LEVEL=3
```
But it does not fix the issue.

marchingband, 2023-12-27:
This is the exact part that fails:
https://www.lcsc.com/product-detail/WiFi-Modules_Espressif-Systems-ESP32-WROVER-B-16MB_C529591.html

Jason2866, 2023-12-27:
Really weird. Curious if Tasmota will crash too, when trying to init PSRAM.
For testing could you flash the Tasmota32 Webcam Build? (not having a cam device is not a problem) this build includes PSRAM use and the fix. Install is easy. Just via the WebInstaller from https://tasmota.github.io/install/

marchingband, 2023-12-27:
it fails

```
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12760
load:0x40080400,len:2908
entry 0x400805c4

00:00:00.003 HDW: ESP32-D0WD v1.0 (PSRAM)
00:00:00.022 UFS: FlashFS mounted with 12600 kB free
00:00:00.036 TFS: File '.settings.lkg' not found
00:00:00.037 CFG: Use defaults
00:00:00.104 QPC: Count 1
00:00:00.400 BRY: Trying to expand FS by 12288 kB
00:00:00.643 BRY: Successfully resized FS, restarting
00:00:00.670 BRY: Berry initialized, RAM used 4333 bytes
00:00:00.686 Project tasmota - Tasmota Version 13.3.0(webcam)-2_0_14(2023-12-12T14:31:23)
00:00:00.723 CAM: InitErr 0x105 try 1
00:00:00.757 CAM: InitErr 0x105 try 2
00:00:00.791 CAM: InitErr 0x105 try 3
00:00:00.792 CAM: InitErr 0x105
00:00:00.100 WIF: WifiManager active for 3 minutes
00:00:01.718 HTP: Web server active on tasmota-2A7FCC-8140 with IP address 192.168.4.1
00:00:02.637 APP: Restarting
ets Jun  8 2016 00:22:57
```

Jason2866, 2023-12-27:
@marchingband  No it does work. If not connected within 3 minutes it restarts.
Search for AP tasmota-2A7FCC-8140 and connect to. Open address 192.168.4.1

marchingband, 2023-12-27:
So then what is the fix for Arduino or for PIO?

lbernstone, 2023-12-27:
Are you using the psram pins (16&17)?

marchingband, 2023-12-27:
@lbernstone I am not. As I said the application and board works perfectly with most WROVER-B chips.

Jason2866, 2023-12-28:
> @Jason2866 So then what is the fix for Arduino or for PIO?

@marchingband  Nothing. Just the application code is different.

marchingband, 2023-12-28:
@lbernstone  @Jason2866 
Using PIO, with 
`platform  = https://github.com/tasmota/platform-espressif32/releases/download/2023.12.00/platform-espressif32.zip`
the problem is gone.
This should be enough information for the Arduino team to fix the bug, do you agree?

marchingband, 2023-12-28:
playing around with PIO versions here is my results:

WITH PIO and `build_flags =  -mfix-esp32-psram-cache-issue`:
`platform = espressif32@6.2.0` -> fails
`platform = espressif32` -> works

WITH ARDUINO IDE using board "esp32 by espressif"
version 2.0.2 -> works
version 2.0.14 -> fails
version 3.0.0-alpha3 -> wont compile ESPAsyncWebServer

marchingband, 2023-12-28:
@Jason2866 I came here to report a bug, is there anything else I need to do? Are you able to triage it?

marchingband, 2023-12-29:
I did some regression testing here are my results, using PIO
```
platform = espressif32@4.2.0   ; esp-arduino 2.0.2 ; ESP_IDF 4.4-beta1 -> PSRAM WORKS
platform = espressif32@4.3.0   ; esp-arduino 2.0.3 ; ESP-IDF 4.4.1     -> PSRAM WORKS
platform = espressif32@5.1.0   ; esp-arduino 2.0.4 ; ESP-IDF 4.4.1     -> PSRAM FAILS
platform = espressif32@5.2.0   ; esp-arduino 2.0.5 ; ESP-IDF 4.4.2     -> PSRAM FAILS
platform = espressif32@6.2.0   ; esp-arduino 2.0.8 ; ESP-IDF 4.4.4     -> PSRAM FAILS
```

mrfaptastic, 2024-01-02:
This also impacts ESP32-S3-DevKitC-1U-N8R8. Enabing PSRAM in Arduino causes a bootloop.

```ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
ESP-ROM:esp32s3-20210327

etc...
```

[image]

maho24at, 2024-01-02:
and also the ESP32 S3 WROOM 1 DEV KIT [N8R2]

Arduino IDE ESP32 2.0.11

`00:00:48.935 -> ESP-ROM:esp32s3-20210327
00:00:48.935 -> Build:Mar 27 2021
00:00:48.935 -> rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:00:48.935 -> Saved PC:0x403743c3
00:00:48.935 -> SPIWP:0xee
00:00:48.935 -> mode:DIO, clock div:1
00:00:48.935 -> load:0x3fce3808,len:0x44c
00:00:48.935 -> load:0x403c9700,len:0xbe4
00:00:48.967 -> load:0x403cc700,len:0x2a68
00:00:48.967 -> entry 0x403c98d4`

marchingband, 2024-01-03:
@mrfaptastic @maho24at 
I doubt very much it is related. On S3 you must set the PSRAM type very carefully or it will fail DIO/QIO and

maho24at, 2024-01-03:
in my case, I can upload normal sketches, like the Blink sketch, and it will execute normally, if I upload a program that uses the PSRAM intensely (which BTW runs on other ESP32 perfect fine) my N8R2 as well the N32R8 start to boot loop.. 

so you think that is another issue? 

[image]

[image]

TD-er, 2024-01-03:
For ESP32-S3 this is indeed a completely different issue as the S3 is very strict on what SPI settings you use for both flash and PSRAM.
I tried to make an overview here on what SPI settings you need (!!!) for the S3 or else it will bootloop.
See: https://espeasy.readthedocs.io/en/latest/ESPEasy/ESPchips.html#quad-octal-spi-mode

About this issue, I think it has to do with the "silicon bug" in earlier revisions of the ESP32-WROVER.
Detection of PSRAM was not working well and the Tasmota platform packages does indeed have a fix for this included.

mrfaptastic, 2024-01-05:
It's a bit of a showstopper bug with the latest esp32-arduino idf then, as no matter what PSRAM settings used,  the device won't boot at all. 

I'll raise a seperate issue.

mrfaptastic, 2024-01-11:
> @mrfaptastic @maho24at I doubt very much it is related to my bug report. On S3 you must set the PSRAM type very carefully or it will fail DIO/QIO etc must be set right for the module or it will fail to init.

You are correct. I have deleted my comment. The issue I experienced was due to accidentally using GPIO38 which is allocated to the Octal-PSRAM. Once I did not use that GPIO (and any of the others) the issue went away.

https://api.riot-os.org/group__cpu__esp32__esp32s3.html

TD-er, 2024-01-11:
@mrfaptastic 
Thanks for that link, now that I see it I remember that's where I got that table info from.
So I will add it as a source to my table.

Jason2866, 2024-01-11:
https://api.riot-os.org/group__cpu__esp32__esp32s3.html
for the s3 a very complete and correct comprehension. On the other sind infos for the c3 are very wrong.

Mbouk, 2024-06-09:
i have the same problem with v3.0.x ; esp32 WROVER Embedded PSRAM fails to initialize !!!
but with version 2.0.15 it works;

[     6][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (369) quad_psram: PSRAM ID read error: 0x7ca9ae06, PSRAM chip not found or not supported
[    15][W][esp32-hal-psram.c:65] psramInit(): PSRAM init failed!
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.44
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   303900 B ( 296.8 KB)
  Free Bytes        :   271828 B ( 265.5 KB)
  Allocated Bytes   :    24672 B (  24.1 KB)
  Minimum Free Bytes:   265948 B ( 259.7 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jun  9 2024 13:19:57
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.4-51-g442a798083-dirty
  Arduino Version   : 3.0.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : {build.fqbn}
============ Before Setup End ============
[   537][I][tecgnss.ino:5125] setup(): [GNSS] CPU Rev : 300
[   542][I][tecgnss.ino:5126] setup(): [GNSS] SDK Rev : v5.1.4-51-g442a798083-dirty
[   550][I][tecgnss.ino:5127] setup(): [GNSS] CPU Frequency : 240 Mhz
[   556][I][tecgnss.ino:5128] setup(): [GNSS] Program started on core 1
[   562][I][tecgnss.ino:5129] setup(): [GNSS] Program build date : Jun  9 2024 13:20:36
Total PSRAM: 0
Free PSRAM: 0

///////////////////  MY CODE
void setup() {
 
  Serial.begin(115200); // For debug

  ESP_LOGI(myTAG, "CPU Rev : %d", ESP.getChipRevision());
  ESP_LOGI(myTAG, "SDK Rev : %s", ESP.getSdkVersion());
  ESP_LOGI(myTAG, "CPU Frequency : %d Mhz", ESP.getCpuFreqMHz());
  ESP_LOGI(myTAG, "Program started on core %d", xPortGetCoreID());
  ESP_LOGI(myTAG, "Program build date : %s %s", __DATE__, __TIME__);

  Serial.printf("Total PSRAM: %li\n", ESP.getPsramSize());
  Serial.printf("Free PSRAM: %li\n", ESP.getFreePsram())
}

me-no-dev, 2024-06-10:
@Mbouk did you compile with PlatformIO?

Mbouk, 2024-06-10:
@me-no-dev i compile with visual micro; 
but it's also the same problem with PlatformIO;

I use this config :
[env:esp32]
platform = espressif32@6.7.0
board = esp32dev
framework = arduino

;upload_protocol = esptool
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -DCORE_DEBUG_LEVEL=4

platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip

me-no-dev, 2024-06-10:
How about with Arduino IDE (1 or 2)

Mbouk, 2024-06-10:
@me-no-dev  is the same!
[image]

me-no-dev, 2024-06-10:
Can you get a close up image of the module, so that we can read what's on it? All modules we have here work fine on 3.0.1, so you must have one from a batch with different PSRAM.
@igrr any clues? Regression in IDF's supported PSRAM chips?

Mbouk, 2024-06-10:
[image]

Jason2866, 2024-06-10:
@Mbouk I would be interested if it does fail with my fork build too. Can you test with Platformio using this platform?
```
https://github.com/tasmota/platform-espressif32/releases/download/2024.06.10/platform-espressif32.zip
```
The debug sketch is removed in my buikd. Just try with your sample sketch.

Mbouk, 2024-06-10:
@Jason2866 it does not work,
Did I configure correctly?
[image]

me-no-dev, 2024-06-10:
We will try to find the same module and give it a shot. It might take a few weeks though (depending if we have it in the EU office or not). @P-R-O-C-H-Y please check in the office when you are back next week.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-15 by Parsaabasi

marchingband, 2025-01-15:
It seems like an important issue to me ...

Jason2866, 2025-01-15:
@marchingband  Still fails with actual Arduino core 3.1.1?

marchingband, 2025-01-15:
I havnt tested.

Jason2866, 2025-01-15:
No, since the issue was most probably not in Arduino code. 99% for this IDF is responsible.
Since the underlying IDF has changed to v5.3.2 there is the possibility the issue is gone as it appeared. There will be no fix for not anymore supported Arduino core 2.x.
Please check if the issue with actual core 3.x still exist, if yes better to open a new issue
(this issue can be referenced).

marchingband, 2025-01-15:
i will try but I am very busy ATM so I don't know when.