espressif/arduino-esp32 #9044: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::partition' [-Wmissing-field-initializers]

tbnobody, 2023-12-28:
### Board

ESP32 Dev Module

### Device Description

Plain Module

### Hardware Configuration

Nothing Attached

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

When compiling with more strict gcc configurations (`-Wall -Wextra -Wunused -Wmisleading-indentation -Wduplicated-cond -Wlogical-op -Wnull-dereference`) I get the following warnings (or errors when compiling with -Werror)

It seems that the used IDF extended the `esp_vfs_littlefs_conf_t` type which is now not  fully initialized.

```
C:/Users/Thomas/.platformio/packages/framework-arduinoespressif32/libraries/LittleFS/src/LittleFS.cpp: In member function 'bool fs::LittleFSFS::begin(bool, const char*, uint8_t, const char*)':
C:/Users/Thomas/.platformio/packages/framework-arduinoespressif32/libraries/LittleFS/src/LittleFS.cpp:86:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::partition' [-Wmissing-field-initializers]      
     };
     ^
C:/Users/Thomas/.platformio/packages/framework-arduinoespressif32/libraries/LittleFS/src/LittleFS.cpp:86:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::read_only' [-Wmissing-field-initializers]      
C:/Users/Thomas/.platformio/packages/framework-arduinoespressif32/libraries/LittleFS/src/LittleFS.cpp:86:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::grow_on_mount' [-Wmissing-field-initializers]
```

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

N/A

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-12-28:
Can you test that PR to make sure it cleans up the warnings?

Jason2866, 2023-12-28:
@lbernstone Tested, working.

tbnobody, 2023-12-28:
Can also confirm that this PR fixes the warnings!

PR #9046 (2023-12-28): Added missing identifiers in esp_vfs_littlefs_conf_t
-----------
## Description of Change
Added missing identifiers to match new version of littlefs component.
 
## Tests scenarios
LittleFS_Test.ino

## Related links
Fixes #9044
closed 2023-12-28 by me-no-dev