espressif/arduino-esp32 #9048: Can I use my Esp32-C6 in PlatformIO with the 3.0.0-alpha? if yes, how?

leorossi2005, 2023-12-29:
### Related area

Board Support

### Hardware specification

Esp32-C6 on PlatformIO

### Is your feature request related to a problem?

yes

### Describe the solution you'd like

How can I do this

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-12-29:
I think that it is possible by using Tasmota PlatformIO packages.
@Jason2866 - Any hint about it?

Jason2866, 2023-12-30:
Correct not possible with official Platformio. I forked and did a version for project Tasmota.
The fork is not 100% compatible. Some changed sdkconfig settings and some library removed (SPIFFS). You can try and use with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.12.10/platform-espressif32.zip
```

Jason2866, 2024-01-15:
Using this open PR, makes it possible to use original espressif sources and libs
https://github.com/platformio/platform-espressif32/pull/1281

leorossi2005, 2024-01-15:
> Using this open PR, makes it possible to use original espressif sources and libs [platformio/platform-espressif32#1281](https://github.com/platformio/platform-espressif32/pull/1281)

Thanks now I can use the 3.0.0 version, but for some reason it says that the ESP32-C6 is not supported anyway

Jason2866, 2024-01-15:
There is no valid C6 boards.json for Arduino. you have to add a custom boards.json yourself

leorossi2005, 2024-01-16:
How can I do this?

mcuw, 2024-04-25:
Hi @leorossi2005,
have a look at this platformio project template: https://github.com/mcuw/ESP32-ghbuild-template
It includes some boards configurations, see https://github.com/mcuw/ESP32-ghbuild-template/blob/main/boards/
The latest arduino framework version is defined in the platform.ini, see https://github.com/mcuw/ESP32-ghbuild-template/blob/d3b2f578cdd3e89b41589b63f97c2a51268fdef3/platformio.ini#L57

mcuw, 2024-06-04:
idf v3.0.0 is released. You can use the latest version so I think you can close this issue.

closed 2024-06-05 by leorossi2005