espressif/arduino-esp32 #9051: mDNS stops working on android after a while

Dante1349, 2023-12-30:
### Board

esp32-s3-devkitc-1

### Device Description

its only the plain board atm because i was moving fromd1 mini to the esp32 board.

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

platformio

### Operating System

android

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

11520

### Description

my issue here is familiar with: https://github.com/espressif/arduino-esp32/issues/4406. But it keeps working for my laptops and desktops, but my phone is not able to resolve the mDNS adress after some time.

### Sketch

```cpp
if (!MDNS.begin(hostname)) {
        Serial.println("Error setting up MDNS responder!");
    } else {
        Serial.println("mDNS responder started. Address: " + hostname + ".local");
    }

    MDNS.addService("http", "tcp", 80);
```

### Debug Message

```plain
no error message or warning.
```

### Other Steps to Reproduce

start the server, wait a while and try to connect via mdns resolution

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-02:
@Dante1349 - MDNS is supported by IDF.
ESP32 Arduino sets `# CONFIG_MDNS_STRICT_MODE is not set` in the `sdkconfig` file when building its static libraries.
You may want to try building your application using `CONFIG_MDNS_STRICT_MODE=y` in Arduino as IDF component in order to test if this solves the Android mDNS issue.

As far as I know PlatformIO allows building Arduino Applications in `Arduino as IDF component` mode.

Dante1349, 2024-01-07:
@SuGlider thank you for you input. but isn't there any easier option to have mdns running? I tried it now as it is and its only my android device that keeps losing the domain. My pc can resolve the url all the time like on the esp8266.

me-no-dev, 2024-01-08:
@Dante1349 the mdns server in ESP-IDF is fully compliant and even adds some extra features. The reason why Android is losing connection is probably somewhere inside the Android mDNS implementation. Our server has passed certification.

Dante1349, 2024-01-09:
@me-no-dev thank you very much for your support! I'm not sure if its the android implementation, that causes the issues with mDNS, i saw some logs related to bind errors when i ran the server over night. Would be nice to find the issue, but i  think i give the espidf mdns a chance now. I found an example on how to combine both frameworks with platformio. I will give it a try and tell you what my results were. This is the example i found, if anyone else is interested: https://github.com/platformio/platform-espressif32/blob/master/examples/espidf-arduino-blink/platformio.ini

zekageri, 2024-01-17:
What i have experienced with my android ( Samsung s22 ) is that if i clear the browser cache it's sometimes can't resolve mdns for my esp32. It goes for like 5-10 minutes and it's just pops back in. Meanwhile it does not happen with other platforms. I think the problem is with the Android mdns implementation. It is fairly new and has some bugs. In the same time it can resolve my raspberry, so it's strange.

Dante1349, 2024-01-18:
I dind't hat the time yet to try out the idf, but i will asap. For the android issue it is strange, i don't have problems on my D1 minis with ESP8266 on android. There i can find it all the time, that why i don't think its any android issue, but I don't know.