espressif/arduino-esp32 #9056: Arduino has no wifiLowLevelInit linked

tablatronix, 2024-01-01:
### Related area

esp32

### Hardware specification

esp32 - *

### Is your feature request related to a problem?

How can we initwifi to read configs ?
I do not want wifi on, I want lowlevelinit

wifiLowLevelInit not found in scope.

local method exists in wifigeneric.cpp

### Describe the solution you'd like

How can we read wificonfigs without turning on wifi?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-01-01:
Is this something in IDF that you want exposed, or a completely new functionality?  You want to know what the saved STA and AP ssid/pw are in nvs?

tablatronix, 2024-01-01:
Yes I am essentially trying to check NVS for persistent confs.

There is no way to check `lowLevelInitDone`

I can check wifimode, but NULL and OFF are the same thing ( I thought null and off would be different if wifi was not init vs wifi not started )

I may have found another way to get this.
ARDUINO_EVENT_WIFI_READY might be a way to know when wifilowlevelinit was invoked, So I could use that to check, 
But I am not seeing any way to do this leaving WIFI_OFF

lbernstone, 2024-01-01:
I'll give you the nvs keys next year. :champagne:

closed 2025-01-15 by Parsaabasi