espressif/arduino-esp32 #9057: 从analogWrite()切换到digitalWrite()后，analogWrite channel依旧被占用

lztmfx, 2024-01-01:
### Board

ESP32C3

### Device Description

# Airm2m_core_esp32c3

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

160Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

我通过8个GPIO控制4个电机驱动器，我知道esp32c3只支持同时输出6个PWM，所以我需要在8个GPIO中同一时刻使用4个GPIO输出PWM和另外4个GPIO输出高低电平。然后在后面某一时刻，我需要交换它们的功能，analogWrite切换为digitalWrite,此时之前输出PWM的GPIO切换为digitalWrite后无法输出高低电平，之前输出高低电平的GPIO切换后也无法输出PWM。bug就是切换模式后没有释放原来的analogwrite.

### Sketch

```cpp
//first
digitalWrite(PIN_LF1, 0);
digitalWrite(PIN_LB2, 0);
digitalWrite(PIN_RF1, 0);
digitalWrite(PIN_RB2, 0);
analogWrite(PIN_LF2, 150);
analogWrite(PIN_LB1, 150);
analogWrite(PIN_RF2, 150);
analogWrite(PIN_RB1, 150);
delay(1000);
//second
digitalWrite(PIN_LF2, 0);
digitalWrite(PIN_LB1, 0);
digitalWrite(PIN_RF2, 0);
digitalWrite(PIN_RB1, 0);
analogWrite(PIN_LF1, 150);
analogWrite(PIN_LB2, 150);
analogWrite(PIN_RF1, 150);//[1167607][E][esp32-hal-ledc.c:222] analogWrite(): No more analogWrite channels available! You can have maximum 6
analogWrite(PIN_RB2, 150);//[1167607][E][esp32-hal-ledc.c:222] analogWrite(): No more analogWrite channels available! You can have maximum 6
delay(1000);
```

### Debug Message

```plain
[  5187][E][esp32-hal-ledc.c:222] analogWrite(): No more analogWrite channels available! You can have maximum 6
[  5187][E][esp32-hal-ledc.c:222] analogWrite(): No more analogWrite channels available! You can have maximum 6
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-02:
@lztmfx  - Please use English to make it easier for others to help you. I have transated your issue:

```
I control 4 motor drivers through 8 GPIOs. I know that esp32c3 only supports outputting 6 PWMs at the same time, so I need to use 4 GPIOs to output PWM and the other 4 GPIOs to output high and low levels at the same time among the 8 GPIOs. Then at some point later, I need to swap their functions. AnalogWrite switches to digitalWrite. At this time, the GPIO that previously outputs PWM cannot output high and low levels after switching to digitalWrite. The GPIO that previously outputs high and low levels cannot output PWM after switching. The bug is that the original analogwrite is not released after switching modes.
```

SuGlider, 2024-01-02:
@lztmfx - I think that the current PlatformIO package is based on Arduino Core 2.0.13. 
Indeed, there is a such issue in the Arduino Core code, but it has been fixed in Arduino Core 3.0.0 (master branch).
You may want to try using Arduino Core 3.0.0-Alpha3 (latest release as of today). It is available in the Arduino IDE, but unfortunately PlatformIO has not released a new package for it.

Based on the current PlatformIO arduino core version, as an alternative workaround, the sketch can use LEDC functions to control PWM, instead of `analogWrite()`.

LEDC functions can be used for selecting directly the desired channels and then attaching the channel to a specific pin.
``` cpp
uint32_t    ledcSetup(uint8_t channel, uint32_t freq, uint8_t resolution_bits);
void        ledcWrite(uint8_t channel, uint32_t duty);
void        ledcAttachPin(uint8_t pin, uint8_t channel);
void        ledcDetachPin(uint8_t pin);
uint32_t    ledcChangeFrequency(uint8_t channel, uint32_t freq, uint8_t resolution_bits);
```
In such case, it would be necessary to write more code in order to set LEDC for each case.

In Arduino Core 2.0.14:
https://github.com/espressif/arduino-esp32/blob/release/v2.x/cores/esp32/esp32-hal-ledc.h

lztmfx, 2024-01-03:
Thank you very much. I will use Arduino Core 3.0.0 to resolve this issue

closed 2024-01-03 by lztmfx