espressif/arduino-esp32 #9058: esp32-hal-ledc.h library need to update for PWM Functionality

Tejas2017610, 2024-01-02:
### Board

Esp32 Lolin 

### Device Description

On Breadboard connected with servo

### Hardware Configuration

Gpio 18 Attached with servo Signal pin

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Linux ubuntu

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-ledc.h:42:5: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
   42 |     SemaphoreHandle_t lock;        //xSemaphoreCreateBinary
      |     ^~~~~~~~~~~~~~~~~
      |     xSemaphoreHandle

### Sketch

```cpp
/* Sweep
 by BARRAGAN <http://barraganstudio.com>
 This example code is in the public domain.

 modified 8 Nov 2013
 by Scott Fitzgerald

 modified for the ESP32 on March 2017
 by John Bennett

 see http://www.arduino.cc/en/Tutorial/Sweep for a description of the original code

 * Different servos require different pulse widths to vary servo angle, but the range is 
 * an approximately 500-2500 microsecond pulse every 20ms (50Hz). In general, hobbyist servos
 * sweep 180 degrees, so the lowest number in the published range for a particular servo
 * represents an angle of 0 degrees, the middle of the range represents 90 degrees, and the top
 * of the range represents 180 degrees. So for example, if the range is 1000us to 2000us,
 * 1000us would equal an angle of 0, 1500us would equal 90 degrees, and 2000us would equal 1800
 * degrees.
 * 
 * Circuit: (using an ESP32 Thing from Sparkfun)
 * Servo motors have three wires: power, ground, and signal. The power wire is typically red,
 * the ground wire is typically black or brown, and the signal wire is typically yellow,
 * orange or white. Since the ESP32 can supply limited current at only 3.3V, and servos draw
 * considerable power, we will connect servo power to the VBat pin of the ESP32 (located
 * near the USB connector). THIS IS ONLY APPROPRIATE FOR SMALL SERVOS. 
 * 
 * We could also connect servo power to a separate external
 * power source (as long as we connect all of the grounds (ESP32, servo, and external power).
 * In this example, we just connect ESP32 ground to servo ground. The servo signal pins
 * connect to any available GPIO pins on the ESP32 (in this example, we use pin 18.
 * 
 * In this example, we assume a Tower Pro MG995 large servo connected to an external power source.
 * The published min and max for this servo is 1000 and 2000, respectively, so the defaults are fine.
 * These values actually drive the servos a little past 0 and 180, so
 * if you are particular, adjust the min and max values to match your needs.
 */

#include <ESP32Servo.h>

Servo myservo;  // create servo object to control a servo
// 16 servo objects can be created on the ESP32

int pos = 0;    // variable to store the servo position
// Recommended PWM GPIO pins on the ESP32 include 2,4,12-19,21-23,25-27,32-33 
// Possible PWM GPIO pins on the ESP32-S2: 0(used by on-board button),1-17,18(used by on-board LED),19-21,26,33-42
// Possible PWM GPIO pins on the ESP32-S3: 0(used by on-board button),1-21,35-45,47,48(used by on-board LED)
// Possible PWM GPIO pins on the ESP32-C3: 0(used by on-board button),1-7,8(used by on-board LED),9-10,18-21
#if defined(CONFIG_IDF_TARGET_ESP32S2) || defined(CONFIG_IDF_TARGET_ESP32S3)
int servoPin = 17;
#elif defined(CONFIG_IDF_TARGET_ESP32C3)
int servoPin = 7;
#else
int servoPin = 18;
#endif

void setup() {
	// Allow allocation of all timers
	ESP32PWM::allocateTimer(0);
	ESP32PWM::allocateTimer(1);
	ESP32PWM::allocateTimer(2);
	ESP32PWM::allocateTimer(3);
	myservo.setPeriodHertz(50);    // standard 50 hz servo
	myservo.attach(servoPin, 1000, 2000); // attaches the servo on pin 18 to the servo object
	// using default min/max of 1000us and 2000us
	// different servos may require different min/max settings
	// for an accurate 0 to 180 sweep
}

void loop() {

	for (pos = 0; pos <= 180; pos += 1) { // goes from 0 degrees to 180 degrees
		// in steps of 1 degree
		myservo.write(pos);    // tell servo to go to position in variable 'pos'
		delay(15);             // waits 15ms for the servo to reach the position
	}
	for (pos = 180; pos >= 0; pos -= 1) { // goes from 180 degrees to 0 degrees
		myservo.write(pos);    // tell servo to go to position in variable 'pos'
		delay(15);             // waits 15ms for the servo to reach the position
	}
}
```

### Debug Message

```plain
Error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-02:
@Tejas2017610 - something may be wrong with your installed Arduino IDE environment.
`xSemaphoreHandle` is deprecated.
Currently ESP32 FreeRTOS uses `SemaphoreHandle_t`

You may try to remove all files in `/.arduino15/packages/esp32/`  and reinstall Arduino Core, using the Arduino IDE.
Let me know if, even after reinstalling it, the issue persists. It may be something related to the Linux installation.

Tejas2017610, 2024-01-03:
I Tried on Windows & do installed still there. But in the library /.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-ledc.h if I put 
#include <freertos/FreeRTOS.h>   // Include the base FreeRTOS definitions.
#include <freertos/semphr.h>     // Include the semaphore definitions.
The compilation passes that error

SuGlider, 2024-01-03:
Could you please confirm which OS generates the error message? 
Is it with Windows and Linux, or only with Linux?

Tejas2017610, 2024-01-03:
both in linux and windows

P-R-O-C-H-Y, 2024-01-08:
Hi @Tejas2017610, the ESP32Servo library is not updated for the 3.0.0 version of Arduino core. 
Please open an issue on the ESP32Servo library to update their code for the new 3.0.0 version.

closed 2024-01-08 by P-R-O-C-H-Y