espressif/arduino-esp32 #9059: Issue with esp_camera and GC2145 on ESP32S3 N16 R8

0x0fe, 2024-01-02:
### Board

ESP32S3N16R8

### Device Description

custom esp32s3 camera board

### Hardware Configuration

```cpp
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM     10
#define SIOD_GPIO_NUM     40
#define SIOC_GPIO_NUM     39

#define Y9_GPIO_NUM       48
#define Y8_GPIO_NUM       11
#define Y7_GPIO_NUM       12
#define Y6_GPIO_NUM       14
#define Y5_GPIO_NUM       16
#define Y4_GPIO_NUM       18
#define Y3_GPIO_NUM       17
#define Y2_GPIO_NUM       15
#define VSYNC_GPIO_NUM    38
#define HREF_GPIO_NUM     47
#define PCLK_GPIO_NUM     13
```

### Version

v2.0.11

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

The same firmware works fine with ESP32D0WDR2, but fails to display video with ESP32S3N16R8. IT only shows one screenshot. There is also a RAM allocation error.

### Sketch

```cpp
standard camera webserver example configured to ESP32S3_EYE

#define CAMERA_MODEL_ESP32S3_EYE
```

### Debug Message

```plain
6:27:40.708 -> E (156417) ledc: ledc_set_duty(720): LEDC is not initialized
16:27:40.708 -> E (156418) ledc: ledc_update_duty(648): LEDC is not initialized
16:27:40.846 -> E (170165) ledc: ledc_set_duty(720): LEDC is not initialized
16:27:40.847 -> E (170165) ledc: ledc_update_duty(648): LEDC is not initialized
16:27:43.408 -> E (172745) ledc: ledc_set_duty(720): LEDC is not initialized
16:27:43.408 -> E (172745) ledc: ledc_update_duty(648): LEDC is not initialized
16:27:43.445 -> E (172745) to_jpg: JPG buffer malloc failed
16:27:43.445 -> E (172749) ledc: ledc_set_duty(720): LEDC is not initialized
16:27:43.445 -> E (172754) ledc: ledc_update_duty(648): LEDC is not initialized
```

### Other Steps to Reproduce

load the stadard esp32 camera webserver example with board selection CAMERA_MODEL_ESP32S3_EYE, video fails to display.
THe same example loaded with ESP32D0WDR2 and board definition CAMERA_MODEL_ESP32_CAM_BOARD, video works normally.

The pinmap and GPIOs havebeen verified multiple times, everythign matches. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-01-04:
have you set PSRAM to OPI?

0x0fe, 2024-01-06:
@me-no-dev 
Are you talking of the flash mode setting here?

0x0fe, 2024-01-06:
@me-no-dev 
i can confirm that it works correctly when we set the PSRAM to OPI mode

[image]

0x0fe, 2024-01-07:
So, is the internal PSRAM on S3 actually wired in octal mode?

lbernstone, 2024-01-07:
https://docs.espressif.com/projects/arduino-esp32/en/latest/troubleshooting.html#how-to-determine-the-module-version

0x0fe, 2024-01-07:
> https://docs.espressif.com/projects/arduino-esp32/en/latest/troubleshooting.html#how-to-determine-the-module-version TL;DR 8MB psram is octal

Thanks for the confirmation.

closed 2024-01-07 by 0x0fe