espressif/arduino-esp32 #9061: esp32-c3 & BLE: E (4212) BLE_INIT: Invalid controller task prioriy or stack size

bmedici, 2024-01-02:
### Board

ESP32-C3

### Device Description

ESP32-C3 dev kit like this one
https://fr.aliexpress.com/item/1005005780121305.html

### Hardware Configuration

Connected to nothing except USBC port for power

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

MacOS

### Flash frequency

base config

### PSRAM enabled

yes

### Upload speed

115200

### Description

framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
ESP32 BLE Arduino @ 2.0.0

When testing a BLE slave device basic sketch, I'm unable to init BLE and get this error:
`E (4212) BLE_INIT: Invalid controller task prioriy or stack size`

Caused by the init() line, I guess `BLEDevice::init("bleslave");`

### Sketch

```cpp
#include <Arduino.h>
#include "helpers.h"

#ifndef CONFIG_BT_BLE_50_FEATURES_SUPPORTED
  #warning "BLE: not compatible hardware"
#endif

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
// See the following for generating UUIDs:
// https://www.uuidgenerator.net/
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string value = pCharacteristic->getValue();
      if (value.length() > 0) {
        Serial.println("*********");
        Serial.print("New value: ");
        for (int i = 0; i < value.length(); i++)
          Serial.print(value[i]);
        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  common_setup();
  delay(4000);

  log("setup BLE");

  BLEDevice::init("bleslave");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new MyCallbacks());
  pCharacteristic->setValue("Hello World");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  Serial.print("Server address:");
  Serial.println(BLEDevice::getAddress().toString().c_str());

  log("setup done");
}

void loop() {
  common_loop();
}
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-03:
@bmedici - Please post your `Platformio.ini` file.

Please update your plaformio package to Arduino 2.0.14 - PlatformIO release 6.5.0
https://github.com/platcantformio/platform-espressif32/releases

Do not include any external BLE library to your project. 
Use the API already in the Arduino Core 2.0.14 package.

An equivalent example to the sketch that you have posted is in
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/Server/Server.ino

I can't compile your example due to the use of `#include "helpers.h"` and calls to `common_setup();`, `common_loop();` and `log()`.
But when I comment out all of it, it compiles and runs fine, using Arduino IDE and ESP32 Arduino Core 2.0.14.
I get no error messages and I can see/connect to the C3 BLE Server using Android BLE tools.

#### Serial Monitor Output:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710

```

Android APP screenshot with the sever and its charcteristic (all fine).
[image]

bmedici, 2024-01-03:
@SuGlider thanks a lot for your time and suggestions
Indeed, I forgot to cleanup the code and share my `platformio.ini` file.
Meanwhile I found something: this `build_flags` was causing the error: `-D 'CONFIG_NEWLIB_NANO_FORMAT'`

Removing the optimimsation led to a working BLE stack:
```
ble-slave | setup BLE
Server address:48:31:b7:93:51:ca
ble-slave | setup done
ble-slave | loop
ble-slave | loop
ble-slave | loop
```

Thanks again for your time & happy new year to all.

SuGlider, 2024-01-03:
>Thanks again for your time & happy new year to all.

Thanks! An excellent 2024 for you too.
I'll close the issue, feel free to open a new one.

closed 2024-01-03 by SuGlider