espressif/arduino-esp32 #9064: ESP restarts due to invalid SSL record

caipifrosch, 2024-01-03:
### Board

custom

### Device Description

ESP32-WROOM module

### Hardware Configuration

custom hardware with ESP32-WROOM

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

ESP32 connects to an mqtt service and will restart from time to time. Sometimes it runs for days and sometimes it crashes after an hour or even less. Seems to be related to secure client/ssl. Always shows exact same decoded backtrace sequence starting from my custom mqtt file. Double heap or mqtt buffer did not help and error is not related to particular mqtt message. See logs. 

### Sketch

```cpp
too complex to strip down
```

### Debug Message

```plain
13:38:12.241 > SYSTEM: Loop Task - Stack 3332/24576 - 135/sec
13:39:05.313 > [113686][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
13:39:05.321 > [113686][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-29184) SSL - An invalid SSL record was received
13:39:05.333 > CORRUPT HEAP: Bad tail at 0x3ffe2ae0. Expected 0xbaad5678 got 0x3ffe29b8
13:39:05.338 >
13:39:05.338 > assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)
13:39:05.347 >
13:39:05.347 >
13:39:05.347 > Backtrace: 0x4008415d:0x3ffbd410 0x40095fc5:0x3ffbd430 0x4009be95:0x3ffbd450 0x4009ba8d:0x3ffbd580 0x400846cd:0x3ffbd5a0 0x4008dc9d:0x3ffbd5c0 0x40199122:0x3ffbd5e0 0x40189d02:0x3ffbd600 0x40123be3:0x3ffbd620 0x40122805:0x3ffbd640 0x40122962:0x3ffbd660 0x401228a5:0x3ffbd680 0x401eb636:0x3ffbd6a0 0x401e9c9d:0x3ffbd6d0 0x400e80c5:0x3ffbd6f0 0x400f8770:0x3ffbd710 0x40162073:0x3ffbd730
13:39:06.024 >
13:39:06.024 >   #0  0x4008415d:0x3ffbd410 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
13:39:06.024 >   #1  0x40095fc5:0x3ffbd430 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
13:39:06.024 >   #2  0x4009be95:0x3ffbd450 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
13:39:06.024 >   #3  0x4009ba8d:0x3ffbd580 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:259 (discriminator 1)
13:39:06.024 >   #4  0x400846cd:0x3ffbd5a0 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:382
13:39:06.024 >   #5  0x4008dc9d:0x3ffbd5c0 in esp_mbedtls_mem_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/port/esp_mem.c:46
13:39:06.024 >   #6  0x40199122:0x3ffbd5e0 in mbedtls_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/platform.c:66
13:39:06.024 >   #7  0x40189d02:0x3ffbd600 in mbedtls_ssl_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_tls.c:6786
13:39:06.024 >   #8  0x40123be3:0x3ffbd620 in stop_ssl_socket(sslclient_context*, char const*, char const*, char const*) at C:/Users/andre/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:336
13:39:06.024 >   #9  0x40122805:0x3ffbd640 in WiFiClientSecure::stop() at C:/Users/andre/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:98   
13:39:06.024 >   #10 0x40122962:0x3ffbd660 in WiFiClientSecure::available() at C:/Users/andre/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:248
13:39:06.024 >   #11 0x401228a5:0x3ffbd680 in WiFiClientSecure::read(unsigned char*, unsigned int) at C:/Users/andre/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:213
13:39:06.024 >   #12 0x401eb636:0x3ffbd6a0 in WiFiClientSecure::connected() at C:/Users/andre/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:257
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-03:
@caipifrosch - Not enough information. Unable to reproduce the issue.
Make sure that all PlatformIO packages are updated.

This is a symptom of a buffer overflow  or lack of memory (HEAP), somewhere in the code. 
Make sure that all the calls have their result tested, this is very important regarding any `malloc()`
The issue may also be in the MQTT library.

IB-AST, 2024-01-04:
Thanks a lot for your response and sugestions. 

Doubling heap and buffer did not seem to make a difference and error occurs randomly. Memory lack problems should happen at similar runtime figures, as processes do not vary randomly. 

From looking at decoded backtrace the secure client / ssl is stopped and memory is freed. Due to a lack of knowledge it is unclear to me when secure client would close a ssl connection and how to prevent this in my code.

caipifrosch, 2024-01-04:
Thanks a lot for your response and sugestions.

Doubling heap and buffer did not seem to make a difference and error occurs randomly. Memory lack problems should happen at similar runtime figures, as processes do not vary randomly.

From looking at decoded backtrace the secure client / ssl is stopped and memory is freed. Due to a lack of knowledge it is unclear to me when secure client would close a ssl connection and how to prevent this in my code.

SuGlider, 2024-01-04:
Maybe some community member may help you.

lbernstone, 2024-01-04:
How did you double heap memory?
ssl connections will need at least 32k of contiguous memory to run.  If you don't have enough memory to build the needed buffers, ssl connections will fail.
Code too complex for you to strip down is too complex for us to guess at the cause of your problem.  Make a minimal example with the error and we can help you.

VojtechBartoska, 2024-09-10:
Hello, due to no answer I am closing this ticket as expired. If needed, you can reopen it. Thank you for understanding.

closed 2024-09-10 by VojtechBartoska