espressif/arduino-esp32 #9065: Analog Continuous Read Not Supported

Hs-Cheng, 2024-01-03:
### Board

ESP32-WROOM

### Device Description

FireBeetle-ESP32

### Hardware Configuration

None

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I read the following page talking about the ADC Continuous mode of ESP32. However, it seems that all codes related to the continuous mode is not included in 2.0.14 release of the code. I can use codes of OneShot mode with no problem, however, all functions related to the continuous mode would generate an "undefined" error message in the compiler.

Or, am I get something wrong and I need to include some files in my sketch to activate the continuous mode of ADCs?

### Sketch

```cpp
analogContinuousSetAtten(ADC_11db);
```

### Debug Message

```plain
Compilation error: 'analogContinuousSetAtten' was not declared in this scope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-03:
>I read the following page talking about the ADC Continuous mode of ESP32 (https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/adc.html). 

@Hs-Cheng - This is only available within ESP32 Arduino Core 3.0.0-Alpha3. Please update your Arduino Core in order to use it.
There is an example avaiable in the Arduino IDE File Menu, under  `Examples/ESP32/AnalogReadContinuous/`

Hs-Cheng, 2024-01-03:
Hello @SuGlider 
In my Arduino IDE, I cannot see 3.0.0-Alpha3. The latest version is 2.0.14.
Could you advise what I should do?

SuGlider, 2024-01-03:
> Hello @SuGlider In my Arduino IDE, I cannot see 3.0.0-Alpha3. The latest version is 2.0.14. Could you advise what I should do?

Please go to the IDE Menu --> File --> Preferences. It will open a new window.
Look for `Additional Boards Manager URL:`
Beside it, there is a Text Field and after that, there is an icon.
Press that icon to open another window with all the URLs that the IDE will use to search for Boards.

Add, at the end of this list of URLs, the URL for ESP32 Development this URL:
`https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json`
Click OK and then OK again.

Go to the IDE Menu --> Tools --> Your Selected Board --> Boards Manager and search for ESP32 to select 
Arduino Core 3.0.0-Alpha3

SuGlider, 2024-01-03:
> Hello @SuGlider I just checked some other documents. It seems that the ADC continuous mode can only be used on ADC1 analog pins, right? I am using both ADC1 and ADC2 analog pins, so maybe this ADC continuous mode is not applicable to me, if my above assumption is true. Please let me know if I can use only ADC1 pins for the continuous mode. Thank you a lot for your assistance.

Yes, ONLY ADC1 pins are supported.

Hs-Cheng, 2024-01-03:
Thank you very much for the information!

closed 2024-01-03 by Hs-Cheng