espressif/arduino-esp32 #9066: Change SPI to allow detaching some pins, like MISO or CS

SuGlider, 2024-01-03:
> Wonderful. Purely for my own understanding, why would configuring the pin with pin mode cause the SPI to break? Does it disable the SPI interface?

In Arduino 3.0.0 there is a new feature called Peripheral Manager.
In the specific case of SPI, when any of its 4 pins (MOSI, MISO, CLK and CS) are used for any other function, for instance, with `pinMode()`, the whole SPI is detached. Therefore, SPI stops working, being necessary to restart it with `SPI.begin()`.

I think that your example brings a good questioning because most displays will only care about using MOSI and not MISO.
I could also use CS pin  for something else and only set it as CS whenever it is necessary.

Therefore, I would say that we may review it in order to allow using SPI pins partly.

_Originally posted by @SuGlider in https://github.com/espressif/arduino-esp32/issues/9040#issuecomment-1876012166_

ForrestFire0, 2024-01-03:
As a user, it also would have been sufficient to note in the documentation that pinMode-ing any of the pins used for SPI would stop SPI. But it of course would have also been nice if it just worked 😄

ForrestFire0, 2024-01-03:
I will also note that this is a breaking change from 2.x to 3.0, but it is not mentioned in the migration guide.

SuGlider, 2024-01-04:
Yes, I agree. The documentation must be enhanced.

PR #9117 (2024-01-19): SPI - Allow detaching of some SPI pins without stopping SPI
## Description of Change
This PR adds a feature to allow detach of SS, MISO or MOSI without stopping the SPI.
This PR fixes issue, when spiStopBus() was called twice (by SPI.end and spiDetachBus), printing an error. 

Also removed pin parameter from detaching functions, as the pins are now stored in the SPI structure. 
Without this change, peripheral manager was able to set to INIT pin, which was already detached from SPI.

## Tests scenarios
Tested on esp32-c3.

## Related links
Closes #9036 
Closes #9066
closed 2024-01-19 by me-no-dev