espressif/arduino-esp32 #9067: check version

AnakinSpain, 2024-01-03:
### Board

esp32

### Device Description

How i could tell to arduino esp32, if filenamelbin is newer than you have, please install it. What i mean is:  esp32 coneect to http, then it reads the filename like firm_02.bin How it is neaer that I have then uodate esp32

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

arduino ide 1.8.19

### Operating System

windows

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

[image]
[image]
 Second is when it chrash

### Sketch

```cpp
void firmwareUpdate(void) {
    Serial.println("paro server");
    UPDATE = true;
    WiFiClientSecure client;
    client.setCACert(rootCACertificate);
  //  WiFiClient client;
    Serial.println("FIRMWARE UPDATING");

    // Add optional callback notifiers
    httpUpdate.onStart(update_started);
    httpUpdate.onEnd(update_finished);
    httpUpdate.onProgress(update_progress);
    httpUpdate.onError(update_error);
    
    t_httpUpdate_return ret = httpUpdate.update(client, URL_fw_Bin);
    
    Serial.println("1");
    switch (ret) {
    case HTTP_UPDATE_FAILED:
    Serial.println("2");
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

    case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

    case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK vvv");
        break;
    }
}
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-05:
This forum is for issues with the code here.  Please ask tutorial questions at https://esp32.com

SuGlider, 2024-01-06:
>How i could tell to arduino esp32, if filenamelbin is newer than you have, please install it. What i mean is: esp32 coneect to http, then it reads the filename like firm_02.bin How it is neaer that I have then uodate esp32

what about using versioning in the file name?
`firm_v1.0.0-0000.bin`

SuGlider, 2024-01-06:
Then the firmware can save this version information to some NVS (persistent information) record and then compare both information, from the internal NVS and from the filename with version information in its name.

AnakinSpain, 2024-01-06:
> Then the firmware can save this version information to some NVS (persistent information) record and then compare both information, from the internal NVS and from the filename with version information in its name.

Yes. It is what I need but I don´t found any example code to understand how do that.

SuGlider, 2024-01-06:
@AnakinSpain - Check this library: https://github.com/chrisjoyce911/esp32FOTA/
It seems to deal with Firmware Version and also check it from the server JSON information.

AnakinSpain, 2024-01-06:
Thanks for this information, but when it isdownloading it stopped at 800000bytes aprox. Never download full file firm.bin

This is my esp32:

Development board model: ESP32-DevKitC-32
Module model: ESP32-WROOM-32
Main control chip: ESP32-DOWDQ6-V3 dual-core 32bit MCU integrated WiFi, Bluetooth,
Integrated 520-kBSRAM, 448-kBROM16-kBSRAMinRTC
External storage: 4MB
USB driver chip: CH340C, with good system compatibility, higher download speed, and more stability. Support VIN external wide voltage input 5-12V power supply (battery version maximum 5.5V input) Support USB power supply, external 3.3V power supply, and VIN power supply three kinds of power supply Mode development board size: 52*28mm/weight about 9.5g
Support ArduinoIDE mixly, mind+, Python and other programming software

VojtechBartoska, 2024-09-10:
Hello, I am closing this as it was mainly answered ann no the ticket is without any activity for a long time.

If needed, you can reopen it. Thanks for understanding.

closed 2024-09-10 by VojtechBartoska