espressif/arduino-esp32 #9068: HTTPS_OTA_Update doesn't work with LTE connection

Lextan276, 2024-01-04:
### Board

LiliGO T-A7608X

### Device Description

Just a board

### Hardware Configuration

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 921600
monitor_speed = 115200

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Ubuntu 22.04

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

OTA works fine when using a Wi-Fi connection but doesn't work with LTE. For LTE connection using TinyGSM library with modem TINY_GSM_MODEM_SIM7600

### Sketch

```cpp
#pragma once
#include <internet_conn.h>
#include <utilities.h>
#include "HttpsOTA.h"
#include "secrets.h"

static HttpsOTA_t otastatus;

void HttpEvent(HttpEvent_t *event) {
#if VERBOSE == WITHOUT_LOG
  switch (event->event_id) {
    case HTTP_EVENT_ERROR:
      Serial.println("Http Events Error");
      break;
    case HTTP_EVENT_ON_CONNECTED:
      Serial.println("Http Event On Connected");
      break;
    case HTTP_EVENT_HEADER_SENT:
      Serial.println("Http Event Header Sent");
      break;
    case HTTP_EVENT_ON_HEADER:
      Serial.printf("Http Event On Header, key=%s, value=%s\n", event->header_key, event->header_value);
      break;
    case HTTP_EVENT_ON_DATA:
      break;
    case HTTP_EVENT_ON_FINISH:
      Serial.println("Http Event On Finish");
      break;
    case HTTP_EVENT_DISCONNECTED:
      Serial.println("Http Event Disconnected");
      break;
  }
#endif
}

void subUpdateTopic(void *pvParameters) {
  while (true) {
    esp_task_wdt_reset();
    if (structOTA.availableOTA) {
      EspHttpsOTA.onHttpCb(HttpEvent);
#if VERBOSE == WITHOUT_LOG
      Serial.println("Starting OTA");
#endif
      vTaskDelete(processingHandle);
      vTaskDelete(netHandle);
      vTaskDelete(provHandle);
      vTaskDelay(512 / portTICK_RATE_MS);
      StaticJsonDocument<64> data;
      data["OTA_started"] = true;
      String buffer;
      serializeJson(data, buffer);
      publish_MQTT_client(rspOTA, buffer);
      data.clear();
      OTAStarted = true;
    }
    vTaskDelay(10000 / portTICK_RATE_MS);
  }
}

void OTAUpdate(void *pvParameters) {
  while (true) {
    esp_task_wdt_reset();
    if (OTAStarted) {
      if (structOTA.availableOTA) {
        vTaskDelete(updHandle);
        EspHttpsOTA.beginConnection(structOTA.fwUrl.c_str(), certCA, true);
#if VERBOSE == WITHOUT_LOG
        Serial.println("Please Wait it takes some time ...");
#endif
        structOTA.availableOTA = false;
      }
#if VERBOSE == WITHOUT_LOG
      printMillis(); Serial.println("OTA in progress...");
#endif
      otastatus = EspHttpsOTA.statusOTA();
      if (otastatus == HTTPS_OTA_SUCCESS) {
#if VERBOSE == WITHOUT_LOG
        Serial.println("Firmware written successfully. Device will reboot after 5 seconds...");
#endif
        StaticJsonDocument<128> data;
        data["success"] = true;
        String buffer;
        serializeJson(data, buffer);
        publish_MQTT_client(rspOTA, buffer);
        data.clear();
        delay(5000);
        ESP.restart();
      } else if (otastatus == HTTPS_OTA_FAIL) {
#if VERBOSE == WITHOUT_LOG
        Serial.println("Firmware Upgrade Fail. Device will reboot after 5 seconds...");
#endif
        StaticJsonDocument<128> data;
        data["success"] = false;
        String buffer;
        serializeJson(data, buffer);
        publish_MQTT_client(rspOTA, buffer);
        delay(5000);
        ESP.restart();
      }
    }
    vTaskDelay(10000 / portTICK_RATE_MS);
  }
}
```

### Debug Message

```plain
When connected through Wi-Fi:
Starting OTA
Please Wait it takes some time ...
[26.26s]MQTT client publish to: 
stream/mr/MR0089/rsp_ota
{"OTA_started":true}
[26.27s]MQTT publish success
Http Event On Connected
Http Event Header Sent
Http Event On Header, key=x-amz-id-2, value=zTHpD5DZgtxfxeTLM6QUlt0Hpb6MWLHCBf8FrbEWuzahjHvh8wt66hD0RgFg31zOex8QLxrG+bZi7DnmklwbcQ==
Http Event On Header, key=x-amz-request-id, value=0F72Z23TDGNCXM4H
Http Event On Header, key=Date, value=Mon, 18 Dec 2023 10:50:42 GMT
Http Event On Header, key=Last-Modified, value=Thu, 09 Nov 2023 15:45:35 GMT
Http Event On Header, key=ETag, value="91a43d00c6702fbaa3b8e6e67f236371"
Http Event On Header, key=x-amz-server-side-encryption, value=AES256
Http Event On Header, key=Content-Disposition, value=inline
Http Event On Header, key=Accept-Ranges, value=bytes
Http Event On Header, key=Content-Type, value=application/octet-stream
Http Event On Header, key=Server, value=AmazonS3
Http Event On Header, key=Content-Length, value=1000320
[36.28s]OTA in progress...
[46.34s]OTA in progress...
Http Event Disconnected
Http Event Disconnected
[56.56s]OTA in progress...
[66.56s]OTA in progress...
Firmware written successfully. Device will reboot after 5 seconds...
[66.56s]MQTT client publish to: 
stream/mr/MR0089/rsp_ota
{"success":true}
[66.57s]MQTT publish success

When connected through LTE:
Starting OTA
[36.78s]MQTT client publish to: 
stream/mr/MR0089/rsp_ota
{"OTA_started":true}
[36.81s]MQTT publish success
Please Wait it takes some time ...
[46.27s]OTA in progress...
E (57008) esp-tls: couldn't get hostname for :mrfirmware.s3.eu-west-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
E (57011) esp-tls: Failed to open new connection
E (57015) TRANSPORT_BASE: Failed to open a new connection
E (57020) HTTP_CLIENT: Connection failed, sock < 0
Http Events Error
E (57035) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (57036) esp_https_ota: Failed to establish HTTP connection
Http Event Disconnected
Http Event Disconnected
[56.27s]OTA in progress...
Firmware Upgrade Fail. Device will reboot after 5 seconds...
[56.27s]MQTT client publish to: 
stream/mr/MR0089/rsp_ota
{"success":false}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-04:
It seems to be a TLS issue related to the 7600 Modem + TinyGSM Library.
Not an Arduino Core issue. 

Maybe someone in the community can help.

Have you tried to connect to any other HTTPS server and got it working?

felmue, 2024-01-06:
Hello @Lextan276 

the error message says `couldn't get hostname for` - maybe an DNS issue?

Or it could be hardware issue like mentioned in this [post](https://github.com/espressif/esp-idf/issues/11153#issuecomment-1516549818).

Thanks
Felix

closed 2024-09-10 by VojtechBartoska