espressif/arduino-esp32 #9069: softAPConfig fails with "DHCPS Set Netmask Failed! 0x5001"

TD-er, 2024-01-05:
### Board

ESP32-S2

### Device Description

-

### Hardware Configuration

-

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I call `WiFi.softAPConfig` with `192.168.4.1` for both IP and gateway and `255.255.255.0` as subnet mask.

This is during the setup process, so there is no STA interface active, only the AP interface.

The error is `ESP_ERR_ESP_NETIF_INVALID_PARAMS`

As far as I can see it is generated here (???)
https://github.com/espressif/esp-idf/blob/5524b692ee5d04d7a1000eb0c41640746fc67f3c/components/esp_netif/lwip/esp_netif_lwip.c#L2260-L2279

The comment suggests you need to set the IP of the interface first, but it is set in the same function. (`set_esp_interface_ip`)

I found this PR which might be related?
https://github.com/espressif/arduino-esp32/pull/6731/files

Apparently this has been an issue which went unnoticed for quite a while but a user reported it was not working anymore when I switched to `ESP32 SDK 4.4.4.20230310`

Using `ESP32 SDK 4.4.4.20230208` was apparently still working.

Might be broken in this commit in esp-idf???
https://github.com/espressif/esp-idf/commit/a71fa82177214657a62dda77d148631227f68e9e

### Sketch

```cpp
-
```

### Debug Message

```plain
[ 11104][D][WiFiGeneric.cpp:1043] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 11108][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Started
[ 11109][D][WiFiGeneric.cpp:1043] _eventCallback(): Arduino Event: 10 - AP_START
[ 11216][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[ 11218][V][WiFiGeneric.cpp:146] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[ 11230][V][WiFiGeneric.cpp:193] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
[ 11239][E][WiFiGeneric.cpp:201] set_esp_interface_ip(): DHCPS Set Netmask Failed! 0x5001
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-01-05:
Provide code pls

TD-er, 2024-01-05:
I got a very minimal piece of code here:

```c++
#include <Arduino.h>

#include <IPAddress.h>
#include <WiFiGeneric.h>
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  IPAddress apIP(192,168,4,1);
  IPAddress gwIP(192,168,4,1);
  IPAddress subnet(255,255,255,0);

  WiFi.softAPConfig(apIP, gwIP, subnet);
}

void loop() {
  delay(1000);
}
```

With output:
```

SHA-256 comparison failed:
Calculated: d8cd69debb03de2830a5005cce52ebafc161b0c89fecb61aaf893684e90318e6
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004b108
=> 0x4004b108: ?? ??:0
[   857][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x40083c70
=> 0x40083c70: _uartDetachBus_RX at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:139
[   869][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x40083c8c
=> 0x40083c8c: _uartDetachBus_TX at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:147
[   882][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x40083ca8
=> 0x40083ca8: _uartDetachBus_CTS at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:156
[   896][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x40083cc4
=> 0x40083cc4: _uartDetachBus_RTS at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:164
[   909][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x40083c70
=> 0x40083c70: _uartDetachBus_RX at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:139
[   923][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x40083c8c
=> 0x40083c8c: _uartDetachBus_TX at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:147
[   936][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x40083ca8
=> 0x40083ca8: _uartDetachBus_CTS at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:156
[   950][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x40083cc4
=> 0x40083cc4: _uartDetachBus_RTS at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:164
[   963][D][esp32-hal-tinyusb.c:711] tinyusb_enable_interface2(): Interface CDC enabled
[   974][D][esp32-hal-tinyusb.c:589] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
[   986][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type USB_DM (42) successfully set to 0x40081e30
=> 0x40081e30: usb_otg_deinit at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-tinyusb.c:62
[   999][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type USB_DP (43) successfully set to 0x40081e30
=> 0x40081e30: usb_otg_deinit at C:/Users/gijsn/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-tinyusb.c:62
[  1013][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 19 successfully set to type USB_DM (42) with bus 0x3ffcd060
[  1026][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 20 successfully set to type USB_DP (43) with bus 0x3ffcd060
[  1249][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled
[  1261][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3ffbf540
[  1272][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3ffbf540
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S2
  Package           : 0
  Revision          : 100
  Cores             : 1
  Frequency         : 240 MHz
  Embedded Flash    : No
[... 68 lines cut ...]
  Allocated Bytes   :    54656 B (  53.4 KB)
  Minimum Free Bytes:   170520 B ( 166.5 KB)
  Largest Free Block:   155636 B ( 152.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  2097152 B (2048.0 KB)
  Free Bytes        :  2079840 B (2031.1 KB)
  Allocated Bytes   :    14928 B (  14.6 KB)
  Minimum Free Bytes:  2079604 B (2030.9 KB)
  Largest Free Block:  2064372 B (2016.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------
    19 : USB_DM
    20 : USB_DP
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
```

Including a complete PlatformIO build env:
https://github.com/TD-er/MCVE_ESPxx/tree/bug/ESP32_softAP_subnetmask

TD-er, 2024-01-06:
OK, added PIO envs for C2/C3/C6/S2/S3 to the MCVE repo.
Just tested with C2/C3/C6 and all of these show the same result.

SuGlider, 2024-01-06:
@TD-er - What happens when the AP IP address is diferent from the GW IP address, but still both are in the same subnetwork?

``` cpp
  IPAddress gwIP(192,168,4,1);
  IPAddress apIP(192,168,4,2);   // DHCP range will be 192.168.4.3 to 192.168.4.12
  IPAddress subnet(255,255,255,0);
```

SuGlider, 2024-01-06:
I just run the example code with the C3. It works fine:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
[   156][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   189][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   190][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[   192][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[   203][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[   216][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
```

SuGlider, 2024-01-06:
I change it to use Arduino Core 3.0.0-Alpha3 with the Arduino IDE. No errors, using the same AP_IP and GW_IP (192.168.4.1)

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x458
load:0x403cc710,len:0x814
load:0x403ce710,len:0x2880
entry 0x403cc710
[   165][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005b9e
[   177][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005b62
[   188][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005b26
[   199][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005aea
[   211][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42005b9e
[   222][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42005b62
[   233][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42005b26
[   245][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42005aea
[   263][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 20 successfully set to type UART_RX (2) with bus 0x3fc8df10
[   273][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 21 successfully set to type UART_TX (3) with bus 0x3fc8df10
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C3
  Package           : 0
  Revision          : 3
  Cores             : 1
  Frequency         : 160 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   306872 B ( 299.7 KB)
  Free Bytes        :   275028 B ( 268.6 KB)
  Allocated Bytes   :    28036 B (  27.4 KB)
  Minimum Free Bytes:   274888 B ( 268.4 KB)
  Largest Free Block:   139252 B ( 136.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan  5 2024 22:58:46
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.2-185-g3662303f31-dirty
  Arduino Version   : 3.0.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C3_DEV
  Arduino Variant   : esp32c3
  Arduino FQBN      : esp32:esp32:esp32c3:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
[   595][V][esp32-hal-uart.c:409] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(20) txPin(21)
[   612][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   662][V][WiFiGeneric.cpp:394] _arduino_event_cb(): AP Started
[   669][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 10 - AP_START
[   677][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[   689][V][WiFiGeneric.cpp:145] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[   702][V][WiFiGeneric.cpp:192] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   306872 B ( 299.7 KB)
  Free Bytes        :   230616 B ( 225.2 KB)
  Allocated Bytes   :    69920 B (  68.3 KB)
  Minimum Free Bytes:   230272 B ( 224.9 KB)
  Largest Free Block:   114676 B ( 112.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    20 : UART_RX[0]
    21 : UART_TX[0]
============ After Setup End =============

```

SuGlider, 2024-01-06:
@TD-er  - I can't reproduce the error. Could you please also try it using the Arduino IDE? Thanks.

TD-er, 2024-01-06:
I will and I will also contact @Jason2866 about this.
Is it possible he might be using a different LWIP version compared to the regular Arduino platform package?

Jason2866, 2024-01-06:
@TD-er Using https://github.com/espressif/esp-lwip/tree/542ba2997f6cc14fe9c3d781bf9b0d890cd70bb1 which is same in espressif branch IDF 5.1

TD-er, 2024-01-06:
Using PlatformIO, I'm not (yet) able to use the latest Arduino code and it seems to be using 2.0.14
```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
```

Any idea how I can use the arduino 3.0.0 code without using the packages Jason made, so we can be sure about where the issue may come from?

Jason2866, 2024-01-06:
@TD-er Which core you are talking? 2.0.14 or alpha 3.0.0  or latest master?
You can't easily compare the official with our fork. Most of the time the fork is more actual.
You can't use the Git branch master with Platformio. 
A build is currently running including the new commits from IDF branch 5.1
Imho jumping between cores in an issue makes no sense. To much has changed everywhere.

TD-er, 2024-01-06:
I see this issue with platform packages 1877 you made.
2.0.14 doesn't seem to have this issue.
Tested with the Espressif code for 2.0.14 also and there it is working.

TD-er, 2024-01-06:
It seems like the Arduino repo is slightly behind with the ESP-IDF code compared to what @Jason2866 uses for his platform builds.
So I guess it will be a future issue for this repo :)

Purely based on the comment here in the LWIP code:
https://github.com/espressif/esp-idf/blob/5524b692ee5d04d7a1000eb0c41640746fc67f3c/components/esp_netif/lwip/esp_netif_lwip.c#L2260-L2279
So I removed the part of the code in WiFiGeneric.cpp where it tries to set the netmask.
These lines:
https://github.com/espressif/arduino-esp32/blob/b2e7338a5e8529b5ede0a8fc7d95703074be093c/libraries/WiFi/src/WiFiGeneric.cpp#L194-L203
And now I can get an IP from the softAP where it failed before.

Maybe this code can be wrapped in some `#ifndef DISABLE_DHCPS_SET_NETMASK` ?
This way it is easy to switch whenever others also hit this issue?

TD-er, 2024-01-06:
I also tested with 2.0.14 by removing the linked lines in WiFiGeneric.cpp and it still works fine to setup DHCP on the softAP.
So I propose to remove these lines as they are no longer needed (or not even working for a while???) and will cause issues soon when you move to newer ESP-IDF code.

SuGlider, 2024-01-06:
This is something to check in the next release, for sure. Thanks @TD-er!
@me-no-dev - please take a look. It may reflect into 3.0.0, in the near future.

me-no-dev, 2024-01-08:
@SuGlider  @TD-er noted! Only DHCPS netmask option needs removal? All else works as expected? Same for 2.0.x?

TD-er, 2024-01-08:
Yep, only the lines I mentioned.
Please test also with the current "3.0.0" code in the Arduino branch (released version of ESP-IDF 5.1.2) as I only tested with the platform IO packages made by @Jason2866 
But the commit message where the LWIP code was changed mentioned something like "finally fixing something that's not working" or something like that, so I guess there is little chance it will break anything with the IDF code used in the Arduino repo.
And setting the subnetmask is already done earlier in the function.

me-no-dev, 2024-01-08:
> And setting the subnetmask is already done earlier in the function.

yes. the lines in question set the DHCP Server option, not the netmask of the interface. Through those options we set what is advertised to the clients (what is the gateway address, DNS, if we have NTP server, etc.)

TD-er, 2024-01-08:
Just to be sure we're talking about the same line (we probably are)

I was talking about this line where the netmask is being set:

https://github.com/espressif/arduino-esp32/blob/b2e7338a5e8529b5ede0a8fc7d95703074be093c/libraries/WiFi/src/WiFiGeneric.cpp#L137

TD-er, 2024-01-08:
N.B. as you can see in the GitHub view, there is some indentation mess, so it looks like there has been some moving around of blocks in the past in that function?

me-no-dev, 2024-01-08:
it's about to be rewritten altogether, so all lines will be checked :)

VojtechBartoska, 2024-01-08:
This will be backported to 2.X branch too, fixes in 3.0.0 version will be covered by networking refactoring.

PR #9016 (2024-01-15): This is a continuation on the topic of adding IPv6 Support to ESP32 Arduino
Original: https://github.com/espressif/arduino-esp32/pull/8907
Discussion: https://github.com/espressif/arduino-esp32/discussions/9009
Closes: #9069
Closes: #5624
Closes: #9085 
Closes: #7065
closed 2024-01-15 by me-no-dev