espressif/arduino-esp32 #9070: ESP32-WIFI Select Channel - Set Channel on WiFi.softAP(ssid, password)

CWGwaltney, 2024-01-06:
### Related area

ESP-32 WIFT

### Hardware specification

ESP32x3

### Is your feature request related to a problem?

I am facing the issue missing the function
WiFi.softAP(ssid, password) establishes a AP on Channel 1 which conflicts with other major network on channel 1

### Describe the solution you'd like

Establish a parameter in the  INIT to set channel

### Describe alternatives you've considered

I searched the internet for information regarding this and did not find an existing solution, maybe a solution already exists

### Additional context

no Command Reference found to discrib this feature

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2024-01-06:
This is already implemented in the SoftAP Construcutor:
``` cpp
/**
 * Set up an access point
 * @param ssid              Pointer to the SSID (max 63 char).
 * @param passphrase        (for WPA2 min 8 char, for open use NULL)
 * @param channel           WiFi channel number, 1 - 13.
 * @param ssid_hidden       Network cloaking (0 = broadcast SSID, 1 = hide SSID)
 * @param max_connection    Max simultaneous connected clients, 1 - 4.
*/
bool WiFiAPClass::softAP(const char* ssid, const char* passphrase, int channel, int ssid_hidden, int max_connection, bool ftm_responder)

```

SuGlider, 2024-01-06:
The defaults are in the header file declaration, which says that the default Channel is 1. You can change it in the code.

``` cpp
    bool softAP(const char* ssid, const char* passphrase = NULL, int channel = 1, int ssid_hidden = 0, int max_connection = 4, bool ftm_responder = false);
```

VojtechBartoska, 2024-02-12:
@CWGwaltney can I close this as solved?

closed 2024-02-20 by VojtechBartoska