espressif/arduino-esp32 #9071: Calculating and updating the Application binary file CRC8 and SHA256 externally (patching)

0x0fe, 2024-01-06:
### Board

ESP32-S3FH4R2

### Device Description

ESP32S3 module

### Hardware Configuration

DNA

### Version

latest master (checkout manually)

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

For a production tool i need to patch few bytes inside the test firmware binary (flashed at 0x10000), on the fly, because these bytes are calculated for each unit.
This is done inside a C# software, which ultimately will flash the binary at 0x10000.

My issue is with the checksum byte.

[image]

I am unsure on how i should calculate this checksum in C#,
is it a CRC8 of total length - sha256 - crc8?
Is it a xor sum of total length -256 - 1 where each bytes is xored with 0xEF? 
is it a standard checksum of total length - sha256 where  last byte (crc position) is set to 0xEF?

here is how i do the checksum of total length - sha256 - crc.

Could anyone point me to the right way of calculating this?

### Sketch

```cpp
#
            var fileByteArray = ConvertToByteArray(filePath); //application.bin
            byte crc;
            for (int i = 0; i < fileByteArray.Length - 32 - 1; i++) //sha256 length : 32, crc8 length  : 1
            {
                crc ^= fileByteArray[i];
            }
```

### Debug Message

```plain
a wrong crc
```

### Other Steps to Reproduce

DNA

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-06:
I know that this checksum is calculated by ESPTOOL, which is a Python tool.
https://github.com/espressif/esptool/blob/master/esptool/loader.py#L176
https://github.com/espressif/esptool/blob/master/esptool/loader.py#L241
https://github.com/espressif/esptool/blob/master/esptool/loader.py#L393

``` python
    # Initial state for the checksum routine
    ESP_CHECKSUM_MAGIC = 0xEF

    @staticmethod
    def checksum(data, state=ESP_CHECKSUM_MAGIC):
        """Calculate checksum of a blob, as it is defined by the ROM"""
        for b in data:
            state ^= b

        return state

```

0x0fe, 2024-01-06:
Thanks! i see, it seems identical to what i do.
I must be checksuming the wrong data length, i will dig into the py code.

SuGlider, 2024-01-06:
@radimkarnis - may you help? Or suggest someone who can?

Phetdanai, 2024-01-06:
> ### Board
> ESP32-S3FH4R2
> 
> ### Device Description
> ESP32S3 module
> 
> ### Hardware Configuration
> DNA
> 
> ### Version
> latest master (checkout manually)
> 
> ### IDE Name
> arduino IDE
> 
> ### Operating System
> windows 10
> 
> ### Flash frequency
> 80M
> 
> ### PSRAM enabled
> yes
> 
> ### Upload speed
> 921600
> 
> ### Description
> For a production tool i need to patch few bytes inside the test firmware binary (flashed at 0x10000), on the fly, because these bytes are calculated for each unit. This is done inside a C# software, which ultimately will flash the binary at 0x10000.
> 
> My issue is with the checksum byte.
> 
> ![image](https://private-user-images.githubusercontent.com/89607149/294639188-7f1b0af2-1624-4abc-8a9c-38db49e3b440.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MjM1NjksIm5iZiI6MTcwNDUyMzI2OSwicGF0aCI6Ii84OTYwNzE0OS8yOTQ2MzkxODgtN2YxYjBhZjItMTYyNC00YWJjLThhOWMtMzhkYjQ5ZTNiNDQwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDA2NDEwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxZDdlYWMxMWRjMzIzNjkzMDE5ZWE2NDNiMzIwNjVhOWY1YWRiZDljNjc3MTlhOGY0ZmQwMjkyMzVhM2RmNjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KJDd7vHSCVbu2x8GxmwDJV7aOWJ-2TpRfbNPxZ3bNVo)
> 
> I am unsure on how i should calculate this checksum in C#,
> 
> here is how i do the checksum of total length - sha256 . and i replace the last byte (where CRC goes) with 0. But i do not get the correct value still.
> 
> Could anyone tell how to calculate this properly?
> 
> ### Sketch
> ``` cpp
>             var fileByteArray = ConvertToByteArray(filePath);
>             byte crc=0xEF;
>             fileByteArray[fileByteArray.Length - 32 - 1] = 0x00;
>             for (int i = 0; i < fileByteArray.Length - 32; i++)
>             {
>                 crc ^= fileByteArray[i];
>             }
>             MessageBox.Show(crc.ToString("X2"));
> ```
> 
> ### Debug Message
> ```
> a wrong crc
> ```
> 
> ### Other Steps to Reproduce
> DNA
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Phetdanai, 2024-01-06:
> > ### Board
> > ESP32-S3FH4R2
> > ### Device Description
> > ESP32S3 module
> > ### Hardware Configuration
> > DNA
> > ### Version
> > latest master (checkout manually)
> > ### IDE Name
> > arduino IDE
> > ### Operating System
> > windows 10
> > ### Flash frequency
> > 80M
> > ### PSRAM enabled
> > yes
> > ### Upload speed
> > 921600
> > ### Description
> > For a production tool i need to patch few bytes inside the test firmware binary (flashed at 0x10000), on the fly, because these bytes are calculated for each unit. This is done inside a C# software, which ultimately will flash the binary at 0x10000.
> > My issue is with the checksum byte.
> > ![image](https://private-user-images.githubusercontent.com/89607149/294639188-7f1b0af2-1624-4abc-8a9c-38db49e3b440.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MjM1NjksIm5iZiI6MTcwNDUyMzI2OSwicGF0aCI6Ii84OTYwNzE0OS8yOTQ2MzkxODgtN2YxYjBhZjItMTYyNC00YWJjLThhOWMtMzhkYjQ5ZTNiNDQwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA2VDA2NDEwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxZDdlYWMxMWRjMzIzNjkzMDE5ZWE2NDNiMzIwNjVhOWY1YWRiZDljNjc3MTlhOGY0ZmQwMjkyMzVhM2RmNjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KJDd7vHSCVbu2x8GxmwDJV7aOWJ-2TpRfbNPxZ3bNVo)
> > I am unsure on how i should calculate this checksum in C#,
> > here is how i do the checksum of total length - sha256 . and i replace the last byte (where CRC goes) with 0. But i do not get the correct value still.
> > Could anyone tell how to calculate this properly?
> > ### Sketch
> > ```js
> >             var fileByteArray = ConvertToByteArray(filePath);
> >             byte crc=0xEF;
> >             fileByteArray[fileByteArray.Length - 32 - 1] = 0x00;
> >             for (int i = 0; i < fileByteArray.Length - 32; i++)
> >             {
> >                 crc ^= fileByteArray[i];
> >             }
> >             MessageBox.Show(crc.ToString("X2"));
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ### Debug Message
> > ```
> > a wrong crc
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ### Other Steps to Reproduce
> > DNA
> > ### I have checked existing issues, online documentation and the Troubleshooting Guide
> > 
> > * [x]   I confirm I have checked existing issues, online documentation and Troubleshooting guide.

0x0fe, 2024-01-06:
@Phetdanai 
Looks like you forgot to write something

So far i tried : 

from 0 to CRC (excluded) :wrong CRC
from 8 to CRC (excluded) : wrong CRC
from 24 to CRC (excluded) : wrong CRC
from 0 to CRC included (set to 0) : wrong CRC
from 8 to CRC included (set to 0) : wrong CRC
from 24 to CRC included (set to 0) : wrong CRC

0x0fe, 2024-01-07:
maybe i have to read each segment and perform the checksum **only** on the data area? 

[image]

0x0fe, 2024-01-12:
hello @radimkarnis can you help with this? what i try to do is to modify a pair of byte in the already patched firmare bin and recalculate the checksum and sha256 so that the image is valid.

0x0fe, 2024-01-12:
so, since i ran out of time to figure this out i used another approach, i suspected that the coredump area was probably not integrity checked, and i tried it, indeed i could flash my 4 butes at it start address with esptool and then in the firmware i used the rom flash functions to read these 4 bytes and do the check, it works perfectly so no need to bother with the firmware image;

closed 2024-01-12 by 0x0fe