espressif/arduino-esp32 #9072: Using two RFID readers (PN532) on esp32.

thaianramalho, 2024-01-06:
### Board

ESP32 DOIT DEVKIT V1

### Device Description

My esp32 is connected to a relay, an Ethernet adapter, Buzzer and PN532 RFID Reader.

### Hardware Configuration

The Pn532 reader is connected to the HSU.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

9600

### Description

I need to connect two RFID readers to my esp32. Currently one reader is using the HSU input. When trying to configure a second reader to I2C, the readers conflict and only one works. Generally, I2C. I would like to know how I can make this work.

### Sketch

```cpp
#include <Wire.h>
#include <Adafruit_PN532.h>

#define SDA_PIN 21
#define SCL_PIN 22

#define SDA_HSU_PIN 17
#define SCL_HSU_PIN 16

Adafruit_PN532 nfcI2C(SDA_PIN, SCL_PIN);

Adafruit_PN532 nfcHSU(SDA_HSU_PIN, SCL_HSU_PIN);

void setup(void) {
  Serial.begin(115200);
  delay(1000);

  Serial.println("Configurando leitor I2C...");
  nfcI2C.begin();
  nfcI2C.SAMConfig();

  Serial.println("Configurando leitor HSU...");
  nfcHSU.begin();
  nfcHSU.SAMConfig();
}

void loop(void) {
  uint8_t success;
  uint8_t uid[] = { 0, 0, 0, 0, 0, 0, 0 };
  uint8_t uidLength;

  success = nfcI2C.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);

  if (success) {
    Serial.println("Tag found at I2C!");
    printUID(uid, uidLength);
  }

  success = nfcHSU.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);

  if (success) {
    Serial.println("Tag found at HSU!");
    printUID(uid, uidLength);
  }

  delay(1000);
}

void printUID(uint8_t *uid, uint8_t uidLength) {
  Serial.print("UID: ");
  for (uint8_t i = 0; i < uidLength; i++) {
    Serial.print(" 0x");
    if (uid[i] < 0x10) {
      Serial.print("0");
    }
    Serial.print(uid[i], HEX);
  }
  Serial.println("");
}
```

### Debug Message

```plain
Dont have.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-06:
@thaianramalho - Are both RFID readers exactly the same hardware? 
If these 2 devices are the same product, each one must have a different I2C Address in order to have both in the same I2C bus.

The PN532 supports both I2C and SPI.
I would suggest to use SPI instead of I2C to communicate and select one or the other using the CS pin.

https://learn.adafruit.com/adafruit-pn532-rfid-nfc/breakout-wiring#wiring-the-breakout-for-spi-1556257
https://github.com/adafruit/Adafruit-PN532/blob/master/examples/ntag2xx_read/ntag2xx_read.ino

thaianramalho, 2024-01-09:
Hey @SuGlider - Yes, they are exactly the same hardware. Both PN532. I'll try with SPI, following these links, thank you very much for the comment.

VojtechBartoska, 2024-09-10:
Hello, can this be closed @thaianramalho?

There is no activity for some time already.

thaianramalho, 2024-09-10:
> Hello, can this be closed @thaianramalho?
> 
> There is no activity for some time already.

Hey! Yes!