espressif/arduino-esp32 #9074: EspClass::deepSleep only accepts 32bit Integer instead of 64bit

juepi, 2024-01-06:
### Board

WEMOS S2 Mini (occurs probably on any ESP32)

### Device Description

no other hardware involved

### Hardware Configuration

Nothing special, just testing DeepSleep

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm really not sure if this is by intention as it's a so obvious bug to me that i really wonder why no one else has stumbled over it, but
according to file `Esp.cpp`:

```
void EspClass::deepSleep(uint32_t time_us)
{
    esp_deep_sleep(time_us);
}
```
Where `esp_deep_sleep` expects a uint64_t value (which makes much more sense, as the 32bit value limits sleep time to ~4300 seconds). 

### Sketch

```cpp
uint32_t SleepTime = (7200 * 1000 * 1000); //µs -> overflow!
ESP.deepSleep(SleepTime); // -> actual sleep time ~2900sec
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-01-06:
Thanks @juepi !!

SuGlider, 2024-01-06:
@juepi - Just make sure you declare the time correctly for the compiler:
``` cpp
uint64_t time_us1 = 7200000000ull;  // works fine
uint64_t time_us2 = (uint64_t) 7200 * 1000 * 1000;  // works fine

//warning: integer overflow in expression of type 'int' results in '-1389934592' [-Woverflow]
uint64_t SleepTime = (7200 * 1000 * 1000); // <<<<===== this will not work. Compiler will try to use utin32_t in the right side.
// result is actually SleepTime = 2,905,032,704 in 32 bit ===> 18,446,744,072,319,617,024 in 64 bit representation
```

juepi, 2024-01-06:
> @juepi - Just make sure you declare the time correctly for the compiler:
> 
> ```c++
> uint64_t time_us1 = 7200000000ull;  // works fine
> uint64_t time_us2 = (uint64_t) 7200 * 1000 * 1000;  // works fine
> 
> //warning: integer overflow in expression of type 'int' results in '-1389934592' [-Woverflow]
> uint64_t SleepTime = (7200 * 1000 * 1000); // <<<<===== this will not work. Compiler will try to use utin32_t in the right side.
> // result is actually SleepTime = 2,905,032,704 in 32 bit ===> 18,446,744,072,319,617,024 in 64 bit representation
> ```

Thanks for the hint @SuGlider, to be honest i managed to step into exactly the described problem when i was calculating my sleep time out of 2 timt_t values as soon as i've switched to the uint64_t datatype 😆 
But sleeping >2hrs  works perfectly well now when calling `esp_deep_sleep` directly.

yours,
Juergen

PR #9077 (2024-01-08): Fixes EspClass::deepSleep(64 bits)
## Description of Change
`esp_deep_sleep(uint64_t time_in_us)` takes a 64 bit argument, but `void EspClass::deepSleep(uint32_t time_us)` allows only 32 bit.

This fixes it, for a sleep time over 2,147 seconds (over 36 minutes).

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#_CPPv414esp_deep_sleep8uint64_t

## Tests scenarios
Just CI.

## Related links
Fixes #9074
closed 2024-01-08 by me-no-dev